riscvarchive/riscv-v-spec #418: Introduce vlmt (vl multiplicative threshold) / VLMT Vector LiMiT

David-Horner, 2020-04-14:
This proposal expands register groups into 8 physical registers for all LMUL<8 , including fractional.

LMUL=8 already uses 8 physical registers and continues to use all 8.

Recapping LMUL>=1 register groups:

Register groups addressed by register numbers that 
	are a multiple of 8 have 8 physical registers,
	are a multiple of 4 but not 8 have 4 physical registers,
	are a multiple of 2 but not 4 have 2 physical registers,
	are not a multiple of 2 have a single physical registers.

These are named reggroup type 8, 4, 2 and  1 respectively

As with LMUL>1 the register number determines the maximum physical registers that the underlying register group can provide.  *** 
These are equivalent to previous constraints of LMUL 8, 4, 2 and 1 respectively.

If vl exceeds any source or destination reggroup type size an illegal instruction exception is raised.

**Register Group processing:**
For LMUL<=1 the addressed reggroup is processed as if successive consecutively numbered physical register are appended to the first (up to the reggroup type).
This results in linear registers with effective VLEN up to 8 time the original length.

For LMUL=8 no extension is possible. The register groups of type 8 are not affected.

For LMUL=2 and LMUL=4  the regroup is processed as if each successive register group is concatenated as register pair or quads to the first group. The effective VLEN is up to 4 times for LMUL=2 and twice WLEN for LMUL=4.

VLMT , with values 1 through 8 specifies the maximum number of physical registers that vl can address. 
When LMUL>1 then VLMT must be a power of two with LMUL <= VLMT.

VLMT is set by the vlmt 3-bit field in vtype. 
When vlmt = 0, VLMT is set to the max(1,LMUL) [the max function is required if LMUL with fractional values is defined, otherwise LMUL is the setting]. Thus a hardcoded value of zero disables the facility. All other values of vlmt set VLMT to 2 through 8 according to vlmt encoding.

On execution of vsetvl[i] if LMUL>1 and LMUL >= VLMT then vill is set (and the rest of the bits of vtype set to 0].

The derived value VLMAX  is now defined in terms of VLMT, and is equal to  VLMT*VLEN/SEW and  represents the maximum number of elements that can be operated on with a single vector instruction given the current SEW and VLMT settings.

**Possible encodings of vlmt with these objectives:**
	1) vl is directly affected by vlmt and so placing it into the lower 11 bits allows the vsetvli instruction to directly control it. These bits are also competing for other important configuration fields, specifically: sew,lmul and ediv. There are more fields pending, including expanding lmul.

2) In that vlmt is used only to determine the value of vl (allowing it to span physical registers), the value of vlmt does not need to be persistent. When re-establishing vl after a context switch it suffices to set vlmt to the representation of VLMT=8 in the saved vtype and execute vsetvl as usual with the save vl as AVL (rs1).

	3) The values 3, 5, 6 and 7 can reasonably be expected to occur less frequently. Splitting the field into two sections with 1 bit to select between powers of 2 and the other values.

Some alternatives:
1) standard bit representation of 3 consecutive bits in lower 12 bits.
2) standard bit representation of 3 consecutive bits in upper 20 bits[31,12].
3) split fields with
	vlmt.h one bit in upper 20 bits [31,12]
	vlmt.l two bits in lower 11 bits.
	When vlmt.h = 0, VLMT = 2 ** vlmt.l	
	When vlmt.h = 1, VLMT = 3, 5, 6 and 7 when vlmt.l = 0, 1, 2, 3 respectively,
4) same as (3) with vlmt.h not set/reset by vsetvli.

**Mask mappings:**

A significant aspect of the expanded register groups is the mapping of the mask bits to the elements.
LMUL>1 has already established a mapping appropriate to the fixed grouping (powers of 2) inherent in its structure.
LMUL=8 is completely established and cannot be expanded.

Mapping for LMUL<=1.

Within the mask MLEN=SEW/8. Each mask position corresponding to the SEW element is therefore divided into 8 equal segments. As previously, only the least significant bit in the segment is the mask bits. On write the other upper bits are cleared and on read only the low order bit is checked.
The fill order establishes even levels (physical register offsets 0,2,4,6) in the low segments of the SEW length field. All odd levels (physical register offsets 1,3,5,7) in the higher segments.
The set order is specifically from most significant segment to least:
7,3,5,1,6,2,4,0  
fill table
phys reg\ bit number|7|6|5|4|3|2|1|0
--|--|--|--|--|--|--|--|--|
fill order
0||||||||x
1||||x
2||||||x
3||x
4|||||||x
5|||x
6|||||x
7|x
--|--|--|--|--|--|--|--|--|
set order
7|x
3||x
5|||x
1||||x
6|||||x
2||||||x
4|||||||x
0||||||||x
 

LMUL=2 is fully compatible with LMUL=1, and the fill order for the first two physical registers matches.

For LMUL=4 the current MLEN fields are sub divided and the upper halves of each are used for the allocation of 5 through 7 physical registers. The fill order could go either of two ways. To match the 5 through 7 of LMUL=1 and 2, or match the lower halves. 

With the current strawman model, fractional LMUL (1/2,1/4,1/8) halves the effective mask length at each lower level, matching the cluster size. As the mask is SEW length based, applying the masking approach is identical within the corresponding mask each element. The same gap between clusters is applicable to the mask and within the physical registers as well.

David-Horner, 2020-04-27:
With Version 0.9-draft-20200424:
All LMUL levels share the same (relative) element mapping within a physical register for a given SEW. Relative because the absolute element address varies for each physical register, but the allocation pattern is the same.
In other word SEW determines the positions of elements within a physical register independent of and LMUL setting.  

Thus LMUL>1 no longer provides internal format information, but only number of physical registers within a register group, and for LMUL<1 only provides limit of vl for use on single physical registers. This number of physical registers can be checked at instruction decode (adjusting for vsetvl dynamic instruction hazard) to ensure compliance with register address alignment ( e.g. with LMUL=4 instructions can only address registers whose address is a multiple of 4). However, there is no need for advance checking, runtime is sufficient as the exception is illegal instruction.
 

When LMUL=8 the resultant vl (vector length) established by vsetvli can vary across all 8 physical registers. Each register is filled consecutively (vx+0, vx +1 ... vx+7). It thus becomes apparent that LMUL=4 and 2  can be established by limiting vl respectively to 1/2 and 1/4 of what LMUL=8 allows. This happens at vsetvli time. Further, physical register sizes of 3,5,6 and 7 are also possible by specifying explicitly the number in the vsetvli instruction.

I thus propose:
 LMUL<1 be replaced programmatically by limiting AVL in vsetvli .

LMUL>1 be replaced by providing the physical register count to the vsetvli instruction.
The count thus provided does not need to be retained with in vtype, as the resultant vl value is sufficient to determine by vtype vsew what the minimum physical register count must have been.
An exception or interrupt handler can reestablish vl by restoring vtype (which no longer has a vlmul field) and specifying the maximum (8) physical registers in the vsetvl instruction.

Compliance with register alignment on register groups of size 8, size 4 and size 2 is performed pre-execute but not likely at decode time as was easily possible previously.
 
And this is the potentially contentious/controversial recommendation:

I propose the number of physical registers in vsetvli be specified by the low 3 bits in the rs2 register that provided the AVL value. 

Why not use 3 bits from the immediate field?
- The immediate fields are prime real estate.
- The bits do not need to be retained, thus allocated a location in vtype, nor in immediate.
- The effect does not directly provide 
- Any static information useful for scheduling, etc. can be obtained from the rs2 field.

Why can we use rs1 for physical register count encoding?
- Only one register is required to establish the count.
- 32 registers are available.( RV32E could theoretically support RVV but is a niche target at best.)
- Each count value has sets of 4 registers from which to choose 
	except for the zero 3-bit value, as r0 cannot generally be used.
- The underlying base RVI architecture only restricts r0 from use.
- Conventions and RVC implicit register use reduces the freedom to use r1 (ra) and r2 (sp). 
- To a lesser extent RVC specifically encoded register subset r8 through r15 discourages their use. 
- All other register usage are interchangeable.
- Registers r24 through r31 are prime candidates to use, however there are at least 2 good alternatives available for each. 

 Thus the constraint on the program to use one of at least 3 various alternative registers and the minimal register pressure thus caused are a reasonable and legitimate trade-off to provide the compressed encoding in a 32bit space for the envisioned 64bit Vector ops.

Note: The top 3 bits of rs1 might be better to use as it combines zero, r1and r2 to a single encoding that can be mapped to a low use multiplier (say 5 or 7). Further, the high register space, which has no RVC encodings and less convention requirements, would be reserved for the high use 1,2,4 and 8 multipliers.  
I mention this as a note because low order bits are easier to initially get ones head around, and it highlights the potential RVC and convention conflicts .

hanna-kruppe, 2020-04-27:
>  LMUL<1 be replaced programmatically by limiting AVL in vsetvli .

It's not clear to me how this would work, can you give an example? Let's take a very simple "load SEW=32 vector, widen to SEW=64, store result to memory" loop as example and suppose we want "LMUL = 1" behavior for the SEW=64 vectors. Currently that would look something like this (number of elements in a0, source pointer in a1, destination pointer in a2, using only `vle` and vtype changes to match current spec):

```
vsetvli t0, a0, e32, mf2
vle.v v0, (a1)
vwcvt.x.x.v v0, v0
vsetvli x0, x0, e64, m1
vse.v v0, (a2)
sub a0, a0, t0
slli t0, t0, 2
add a1, a1, t0
slli t0, t0, 1
add a2, a2, t0
```

If I understand correctly, you propose to remove the `mf2` feature used here, and replace that `vsetvli` with `vsetvli t0, t1, e32, m1` where t1 is somehow derived from a0. But *how* would one modify this example to compute a `t1` such that the behavior of this loop is preserved (i.e. vl is at most VLEN/64 and the same for both element widths)?

David-Horner, 2020-04-27:
Yes, I propose to remove mf2, especially as a "persistent" setting in vtype.
We could consider a further parameter to vsetvli, but for now this will do.

```
loop:
mv tmpl1,a0     # tmpl1 encodes LMUL=1 in the register number
vsetvli t0, tmpl1, e64, m1
vsetvli t0, t0, e32, m1      # the t0 value constrains to a single physical register so any VLMT is OK
      # t0 in unchanged by current vl derivation                        
vle.v v0, (a1)
vwcvt.x.x.v v0, v0
vsetvli t0, t0, e64, m1          #  x0,x0 behaviour should be revisited  
vse.v v0, (a2)
... # update a0, a1, a2
bnez a0, loop

```

David-Horner, 2020-04-27:
Obviously the initial vsetvli and its tmpl1 assignment can be hoisted out of the loop:
```
mv tmpl1,a0     # tmpl1 encodes LMUL=1 in the register number
vsetvli tmpl1, tmpl1, e64, m1  
loop:
mv t0,tmpl1
blt t0,a0,setusingt0   # I may have comparison backwards (which is vs2?)
mv t0,a0     # sub optimal vl when a0 < 2 * MAXVL but avoids additional vsetvli
setusingt0:
vsetvli t0, t0, e32, m1      # the t0 value constrains to a single physical register so any VLMT is OK
      # t0 in unchanged by current vl derivation                        
vle.v v0, (a1)
vwcvt.x.x.v v0, v0
vsetvli t0, t0, e64, m1          #  x0,x0 behaviour should be revisited  
vse.v v0, (a2)
... # update a0, a1, a2
bnez a0, loop        # yeah. I try to use eq/ne compares so register specification order doesn't matter

```

A Note about the title. I decided not to change it as the parameters can be considered virtual and may be useful in descriptions.

hanna-kruppe, 2020-04-27:
Thanks for clarifying. This seems like a significant amount of extra instructions (dynamic, for short vectors/small trip counts, and static for compact loops). It's also concerning that trying to hoist the parts that ought to be loop-invariant apparently introduces other overheads and complications. This is obviously just an artificial example loop, but I don't really see how it's is going to get any better in more realistic examples. If anything, needing more scalar registers (e.g. a0 *and* tmpl1 here, even more if you juggle more than two element widths) will be *more* of a problem in bigger kernels.

(By the way, I also believe the second example would have to update *both* a0 and tmpl1 in the part of the loop body that I omitted, this makes it look even worse in comparison.)

At a glance, it seems to me that most of these problems are easily avoided if we just leave the "mf2" etc. capability of `vsetvli` intact: remove them from vtype, but still using it in the calculation of vl from AVL. Not only does that remove the need for "dummy" `vsetvli`s just to get the intended vl, it also suggests to move the "LMUL-equivalent" bits back to the immediate (stealing *four* bits from rs2 is even less practical than stealing three), which also removes other instructions *and* (most?) of the extra scalar register pressure.

knightsifive, 2020-04-27:
I'm still a bit unclear on the original goal: is it to remove LMUL from vtype? If so, I'm skeptical that this benefit outweighs the issues Hanna described.

kasanovic, 2020-04-28:
I agree with @hanna-kruppe here.  If we were to go down this path, it would use the current  bits in immediate and the example loop's assembly code would be unchanged.

We still use LMUL to format the mask registers, and I can't see a clean way around that, beyond making masks always be a fixed width (e..g, single bit).

Also, not keeping LMUL in state doesn't allow the effective EW version of loads and stores. Error checking is a little more complex/longer circuit latency as have to inspect more bits of vl value versus register number in instruction to detect error.

Dropping LMUL also means losing track of the tail.  The number of vector registers touched by an operation is given by vl, so tail-undisturbed is straightforward to implement.

There was also a request to allow ELEN>VLEN for larger lmul - I believe this could be supported.

kasanovic, 2020-04-30:
This proposal boils down to removing LMUL in the vtype CSR, and instead only specifying in vsetvli instruction such that AVL is set correctly.  The dynamic value of vl would be used to constrain the vector register used to start a group (e.g., if vl is > VLEN/SEW, then LMUL>1). 

There are several issues with this proposal:

1) Some current instructions use VLMAX=LMUL*VLEN/SEW to describe the input operand, e.g., vrgather and vslides.  In particular, vrgather uses LMUL to describe size of lookup table, which is independent of vl.  These instructions would need to changed.

2)  A certain class of code errors would not be caught, and would appear at different times on different implementations.   e.g., 

```
vsetvli t0, a0, e16,m8 # LMUL=8
vadd.vv v1, v2, v3
```

This code will run without error if AVL in a0 <= VLEN/SEW, but will cause an illegal instruction exception if AVL >VLEN/SEW.  When the errors appear depends on both the microarchitecture (VLEN) and the input data (AVL), which is disconcerting for software QA. While an assembler could help catch most of these errors, some might escape or be difficult to analyze if code has more complex control flow.

3) While the data layout does not now depend on LMUL, the mask register layout does.  We would have to change the mask register layout, e.g., to have single bit per element at all times.  This is quite a disruptive change.  There is a benefit to current scheme where mask register layout matches data layout (MLEN=SEW) when LMUL <=1, as LSB of data value can be used as predicate without additional compare (and vice versa, using mask as data input directly).

4) The dynamic error checking logic becomes somewhat more complex, as have to consider the vl value shifted by the SEW value.

5) The benefit of dropping LMUL in vtype is small, beyond aesthetics (it is admittedly attractive conceptually).  The vsetvli instruction still has to use immediate encoding bits to specify LMUL to set vl correctly, so we do not reclaim any of these bits.  The three flops needed to encode LMUL is not a big expense.

6) A later ILEN=64 encoding would not need this, so only introduces disadvantages there.

Given that dropping LMUL has these issues and would be a very disruptive change (change mask encoding, rework gather and slide instructions), I propose we drop consideration of this idea.

David-Horner, 2020-04-30:
I included (conflated?) 6 ideas in this issue

1) fractional lmul can be emulated in software (thus only 2 bits needed for 1,2,4,8)
2) lmul is not need to convey structure information (only number of physical registers in a register group).
3) lmul persistence is not needed

4) lmul does not need to be stored in the low 11 bits affected by vsetvli
5) lmul encoding can instead be provided in rs1 register
6) lmul can also support multipliers 3,5,6 and 7 (and potentially more).

Of these: 

3) persistence is definitely needed for VLMAX(vrgather/vslide and register group alignment checking)
1) software emulation of fractional lmul –
		I agree it is comparatively expensive (@hanna-kruppe above) 
		(when I expect the net to only be one more bit from vsetvli immediate)
2) structure information encoded – Krste: not for data but valuable for mask when LMUL<=1
		I intend to respond to this in issue #435, obviously a lot of clarification need there.
	The validity of this may affect issues 6.

But for the remaining sub-issues, 4, 5 and 6, I believe they are still open .
I’m willing to close this issue, and open one for each of 4,5 and 6 if this would facilitate/clarify the issues.
However, 4 and 5 are very closely related and may be better presented together.

Once again thank you each for the consideration of my concern that these issues are clearly identified and addressed.

kasanovic, 2020-05-02:
Closing this issue as @David-Horner will be creating new single concept issues.

closed 2020-05-02 by kasanovic