riscvarchive/riscv-v-spec #420: vfwcvt /vfncvt  allow [u]i16 conversion without fp16 support

chihminchao, 2020-04-15:
Some software programmers discuss the following case if there is no fp16 support
#SEW =16 :  i16 -> f32
vfwcvt.f.xu.v vd, vs2, vm    
vfwcvt.f.x.v  vd, vs2, vm 

valid:
   Although the instructions are floating conversion, It doesn't handle the fp16 but only i16 and  fp32
invalid:
    They belong to floating point instructions and the spec require it support fp16 when sew is set to e16
    [ref](https://github.com/riscv/riscv-v-spec/blob/master/v-spec.adoc#vector-floating-point-instructions)
    "The standard vector floating-point instructions treat 16-bit, 32-bit, 64-bit, and 128-bit elements as IEEE-754/2008-compatible values. If the current SEW does not correspond to a supported IEEE floating-point type, an illegal instruction exception is raised."

aswaterman, 2020-04-15:
My reading of the spec is that these instructions should be supported.  The constraint in the spec only applies to the floating-point types, not the integer types.  Similarly, if the system supports float32 and int64, then float32 to int64 widening conversions should be supported.

kasanovic, 2020-06-28:
I agree with @aswaterman 
The spec needs clarification on supported FP types.

kasanovic, 2020-07-03:
Cleaned up by: https://github.com/riscv/riscv-v-spec/commit/50fab57354c3d6d136de5b844f843e59acf2846e#diff-4fa320b4e415eee7270590bb06607389

closed 2020-07-03 by kasanovic