riscvarchive/riscv-v-spec #424: No size fits all : There is no one true SEW level

David-Horner, 2020-04-19:
The fixed byte/half/word load/store issue has raised the awareness/mindfulness of conflicting requirement trade-offs.

Code size  vs  hardware simplicity  vs  vector model clarity vs process cycles vs effective encoding via vtype/setvl/vcsr and more...

This 32 bit encoding is seen/described as a compromised compressed form of the ideal 64 bit encoding.

However, 64 bit encoding will certainly be a loss wrt code size, hence instruction cache impacts. In many cases the 32bit encoding will out perform. Unless a novel multi-op encoding at 64bit arises, and I am confident some are already considering "VLW" encoding in 64bit.

The vtype provides a persistent prefix for the base vector instructions providing "Standard Element Width" and register group level (defining depth and structure), with pending Tail and Mask Fill directive and a planned Element Division Extension.

**"Standard Element Width" is found to be inadequate**, and that was apparent from the outset with

 (# 1) demand for (typically) smaller data width elements (specifically byte,half and word) to be readily incorporated into the instruction flow, especially at load/store, and now increasingly float data type and fixed-point.

 (# 2) widened data types (2 * SEW) to contain full results of operations (such as multiply) and iterations of operations. And also quad operations, with a recent emphasis to address the demands of point # 1.

Krste's insight of separating data expansion from the fixed size encoding in the load/store is helpful in highlighting this SEW inadequacy, and further that temporary over-ride of the vtype SEW is beneficial to 1) reduction of code and instructions, that 2) augments conceptual instruction mode flow.

To wit: Establish the correct SEW for the majority of the operations and handle the exceptions with a one-off mechanism. (Granted in this form it is only for load/store, but effectively so for many cases).

With an apparently viable approach to address issue # 1, we are now trying to assess the correct SEW level to provide, by judicial use of the widening (and to a lesser extent the narrowing) operations, the other SEW inadequacy (see  # 2).

I suspect that this is a loosing game/ a goose chase.

To understand this I believe it is instructive to realize alternatives to the " SEW to 2 * SEW" widening are possible.

The draft states this in 11.3 in the context of narrowing operations but it is equal valid for widening:

> An alternative design decision would have been to treat LMUL as defining the size of the source vector register group. The choice  here is motivated by the belief the chosen approach will require fewer LMUL changes.

And this is applicable to SEW changes a well.

With 2 * SEW widening, the conditioning on the input data (including mask set operations) can be performed effectively on SEW data and then widened without a (vsetvli) change of LMUL or SEW.

Thus a good choice.

**Consider quad widening:**

The draft current proposal has only Quad-Widening Integer Multiply Add , a necessity to support any reasonable degree of accumulation.

To support compact byte/half/word loads "useful" quad and octal widening operations based on the current "2 * SEW" are anticipated.

One expectation is that quad will widen from SEW to 4 * SEW, octal from SEW to 8 * SEW.

However, 1/2 SEW to 2 * SEW is a reasonable alternative, as is 1/4 SEW to SEW. The latter representing in the classic sign/unsign extend operation to SEW.

For Octal there are 4 variations, all with some inherent merit. And for simplicity it is appealing to have symmetry between quad and octal, even though that may be sub-optimal when the typical use cases are finally analyzed.

And here is where I see this as a loosing game.

(## 1) Minimally, to support the load equivalent byte/half/word to SEW (and only to SEW of double)  we need to support integer and float quad and octal widening.

 Can we reasonably provide the many instruction ops necessary to mirror in quad/octal each of the current double widening?

 Are there "important" ones that we will support at quad and octal widening, and others that are not?

 If there are such "unimportant" at quad/octal, why are they sufficiently important to be include at double?

 At quad and octal the "opportunity" to further mix data sizes exists. Are any of these "important"? e.g.  2SEW = (1/4 SEW * 1/2 SEW) + 2SEW.

 Can we reasonably make the right choices for quad and octal of variants to include?

(## 2) Can we choose the right SEW level for these alternatives.

 the example: 2SEW = (1/4 SEW * 1/2 SEW) + 2SEW.

 could just as easily be 8SEW = ( SEW * 2SEW) + 8SEW.

   or SEW = (1/8SEW * 1/4SEW) + SEW.

  Each of which is legitimate and optimizes the other single-SEW operations in the code.

My perception is that optimal SEW for a code segment is highly dependent upon adjacent/local/context operations (complicated by control flow operations).

KEY POINT: A local optimal SEW:LMUL with exceptions is I believe possible.

  Scaled load/store goes a long way to achieve that by providing such exceptions in a key activity.

  However, I believe we may need another exception mechanism.

REQUEST: What I am hoping is that we can get consensus on the above KEY POINT. Then, move on to an efficient exception mechanism.

David-Horner, 2020-04-19:
within #423  I propose just such an exception mechanism. The vmodinstr as part of the prefix provides a SEW and LMUL scaling field that does the one time adjust to allow the operation to be performed at the resultant Effective SEW and Effective LMUL level.

Further variations could apply to each input to provide 1/4SEW to vs1 and 1/8SEW to vs2 with vd still at SEW. The 64bit encoding (or at least a further subset) in a two instruction packet.

kasanovic, 2020-04-19:
please see #425 for my solution

commit a526fb9c (2020-05-01): Add EEW loads/stores.  Update instruction encoding and code examples.
Closes #425, closes #424.
closed 2020-05-01 by kasanovic