riscvarchive/riscv-v-spec #427: Make unsigned integer in widening instructions 2 * SEW

David-Horner, 2020-04-21:
This suggestion weighs the benefits of increased scalar range of 2 * SEW unsigned rs1 (X register input) with 
- possible requirement in RV code to clear upper SEW bits
- gate changes to support 2 * SEW input from previous SEW length

**First the increased scalar range.**

a) Biasing, preparing and tailoring wide accumulator values.   Not all accumulations will start from zero, or SEW size values.
Especially when SEW is 8 or 16 the range of values to prime an accumulator is insufficient in many use cases. #287 opened by Zpedro would avoid a SEW/LMUL switch to condition its accumulators with this change.

b) Extended operating range for foundation operation.
e.g. in SEW=8 double widening multiply allows rs1 X[14– n:0]bits  * vs2[n:0]bits operation without overflow. (and quad widening vqmaccsu.vx will allow full range of 2 * SEW rs1 by SEW vs2)

c) Such enhanced operations are available in code flow without changing SEW.
Enhanced operations would be:
vwaddu.vx
vwaddu.wx
vwsubu.vx
vwsubu.wx
vwmulu.vx
vwmulsu.vx	   
vwmaccu.vx
vwmaccsu.vx (or vwmaccus.vx see #426) 	
vqmaccu.vx
vqmaccsu.vx (or vwmaccus.vx) 	

Note: once SEW reaches XLEN there is no benefit for this enhancement.
Nevertheless three enhanced modes for RVV64 is substantive.
	
**Second, the hardship to condition X register values for 2 * SEW.**
There may be an additional RV base instruction (and X register) needed to clear top bits of the X register to simulate current behaviour.
   For SEW=8     andi x’FF’ would suffice
   For SEW=16   andi x’3FF’ may suffice, but lui; andi combo will always.
   For SEW=32   lui; andi is required

In other areas, it appears the general feeling is that trading a few RV base operations for enhanced RVV functionality is a good tradeoff.

**Third, how disruptive to current micro-architecture designs is this change.**
   Given vwopwv operands already provide 2 * SEW input along with a SEW input, the circuitry is already there. 
   The opcode decoding is also minimal.
   The gating of the extended X range is present for current designs to support SEW=XLEN.
   The multiple ops are probably the most affected and then only for minimal implementations using  multi-step calculations.
  
Overall I see this as a win with little hardship. However, I definitely need to have hardware gurus’ input.

**What about 2 * SEW for signed rs1 input?**

The trade-off for scalar are more substantial at SEW=8 and 16. RV64 has addw to sign extend at SEW = 32. Thus sign extension in RVV widening is more valuable to avoid sll;sra combination. 
I don’t plan to advocate 2 * SEW for signed rs1, but I don’t think I can reasonably oppose as the trade-offs are close.

David-Horner, 2020-06-28:
This needs to be resolved for V1.0
It cannot be easily retrofitted to a V1.0 model.

kasanovic, 2020-07-20:
Concretely, this proposal is to change the widening integer vector-scalar operations to treat their x register input (rs1) as 2*SEW not SEW, first for unsigned, then possibly also for signed.

I think the biggest benefit would be in the widening multiply instructions, but that also implies extra hardware (larger multiplier array) over what is needed for current case.  I can see there is also some benefit to short computations into a wider accumulator with a large initial value, but also see that some cases will need more scalar instructions.  There is non-zero hardware cost even just for adds to handle the new case, with possibly some software overhead in other cases (e.g., when using an XLEN-wide scalar load to bring in packed 8b elements into one register, then feeding these one at a time into a widening operation - the current spec provides the mask).  Also, this would be a large change at a late stage in the proposal process.

I will bring up again in next TG meeting.

David-Horner, 2020-07-21:
Thank you very much for the synopsis.

x input is naturally limited by XLEN. Once that is exceeded zero extend (or sign extend) takes over (thus xlen precision) 

> the biggest benefit would be in the widening multiply instructions

As Krste alludes, this form of the multiply is not otherwise available.

Even for widening multiply, EEW has to be a supported SEW width.
As a result multiply circuitry is already in the design.
 i.e. RVV64:  vmulu.vx at SEW=64 has sufficient multiplier circuitry for the expanded vwmulu.vx at SEW=32.
Simple  micro-architectures are likely to leverage the same circuity for vmulu as vwmulu. 
High end performance machines may duplicating circuity in difference execution units and so widening units may pay a price for the increased functionality/performance.

>feeding these one at a time into a widening operation - the current spec provides the mask

The original proposal discusses this - a single andi is sufficient for byte.

```
<digression detailing scalar affects vector>
The "feeding one at a time" would entail a shift likely, so an andi for each of them 
(except the last if logical shift)
This would simplistically be mixed vector/scalar code, with scalar feeding vector -
 with possible interlock delays between scalar and vector processors.
Alternatively,  four x registers could each have their byte values established before
 the set of vector instructions that consume the values. 
This could relax the interface constraints. 
But again a single addi for each byte is not arduous.
</digression detailing scalar affects vector>
```
>this would be a large change at a late stage in the proposal process.

Agreed. However, it cannot be retrofitted. The benefits are there, the timing was not excellent.

kasanovic, 2020-07-24:
We discussed proposal to widen scalar input to widening operations to
2*SEW.  For widening multiplies, this would double the size of
multiplier arrays required.  The group discussed using a
microarchitectural check on scalar width to select a narrower
multiplier, but group consensus was that this information should be
supplied through a different opcode.  Given that multiplies would
provide the larger benefit, and that adds would then have a
non-uniform format, the descision was made to stay with the PoR.

closed 2020-07-24 by kasanovic

David-Horner, 2020-07-30:
Comments on resolution and request to re-frame 

>We discussed proposal to widen scalar input to widening operations to
2 * SEW. 

>For widening multiplies, this would double the size of
multiplier arrays required. 

When SEW < XLEN, I  noted that the double size multiplier arrays (conceptually)
1) already exist for the next (SEW) level up non-widening multiplies, and
2) the widened output would be compatible with that next up multiply.
Note as well that only vector operator needs to be distributed to the appropriate wider multiplier as the scalar value is “constant” across all multiply operations.

This approach is quite appropriate for many micro-archs;  those Uarchs that internally have SLEN=VLEN (= channel width)  , and of these, especially those that are register write limited. 
In these Uarch the VLEN register 0 would be written in one cycle (or process set) with register 1 in the next (and if LMUL>1 register 2, 3, etc. with each subsequent cycle (or process set). The throughput would be 1/2 of discrete multiplier units per vector operand but as the register write would be saturated there is no actual loss.

This approach does not work well for SLEN<VLEN (and perhaps multiple active channels) that might distribute both vector source from register groups to multiplier units,  and double width results to distant register ports. Possibly further complicated by renamed register segments. 
These Uarch would rather have dedicated SEWxSEW multiply units (potentially sharing segments of the same multipliers for the next (2 * SEW) level up, extended to provide a double width result. 
The benefit of such a configuration is full hardware throughput, that would be tailored to “normal” vector register file read port rate.  In that a channel (likely SLEN) width slice would be generating double width element in the same physical register (but potentially to renamed segments of the register) the advantage seen in the simpler SLEN=VLEN design (of consecutively writing full VLEN registers) is not present.  

There is further impetus to optimizer the SEW=8 case. Both in the vector x vector and vector x scalar use are expected to be a common use case,. But further, 8 bit is the extreme situation for number of   elements, source operand distribution and/or widened result distribution. And lastly, the 8x8 multiplier array is relatively small, so the investment in gates pays substantial dividends on the smaller bit sizes.

> The group discussed using a microarchitectural check on scalar width to select a narrower
multiplier, but ...  [keep in mind that this dynamic selection is for SLEN<VLEN type Uarchs)

With the scalar 2 * SEW introduction, dynamically selecting between the two approaches would require reading the scalar value , and determining if the upper half (SEW bits) of it were zeros (or all ones for signed), in which case it could use the optimized approach. If the high SEW were not just the sign, then  the fall back to using the 2 * SEW multipliers approach would be used. This dynamic re-configuring was rightly trounced. Evaluating the high SEW bits would occur much to late in the process introducing stalls or complex read ahead X register circuitry that is not needed anywhere else and would likely impact cycle timing. Dead on arrival.

>group consensus was that this information should be
supplied through a different opcode. 

>Given that multiplies would
provide the larger benefit, and that adds would then have a
non-uniform format, the decisions was made to stay with the PoR.

I believe this narrative  correctly reflects the reasoning.

I fully agree with this final conclusion that uniformity persuades to handling all potential integer 2 * SEW the same way.

However, I believe I must take blame for framing the issue as a duality. Either leverage the next level (2 * SEW) multiplier or optimize with a narrower widening multiplier circuit. An the latter would require a dynamic “macro” selection between them.

I did not present alternatives that change the narrative and basis for decision.

Firstly, a full double width multiplier is not necessary (but certainly sufficient) for the integer SEWx(2 * SEW) case. By definition, the high SEW bits of the vector operand are zero and do not participate in the (2 * SEW)x(2 * SEW) circuitry. Further, only SEW bits of the product of the vector with the high SEW bits of the scalar are retained, and thus need only be generated and summed. 

Especially when ELEN > XLEN, but even with lower SEW, the widening multiply (and even non-widening) will likely be implemented as temporal iterations of sum of partial products, in some cases this will be driven by the desire to keep cycle time constrained. This temporal circuity  could be utilized for conditionally summing the High-SEW-Scalar-bits  with the SEW-vector on the narrow multiply. Thus zero/sign high bits of scalar are not a selection between LARGE/narrow but rather an optimization of the narrow process.

The optimization of narrow multiplies can be incorporated independently for various sizes of SEW. The cost to add the upper X-register  SEW-bits is nominal at 8bits and still small at 16 bits. For a RV32 these are the only two integer widening of concern for 2 * SEW scalar. For RV64 the only other integer widening 2 * SEW is 32 bit. 
A tradeoff between 
1) half throughput (use next level up full multiplier),  or 
2) (as above) conditional temporal, or
3) parallel partial-product generation and fast sum hardware
can be chosen independent of upper 32 bits of X register.

Re-framing the proposal in these terms changes the question from a dichotomy to a continuum of design options that can be effectively  implemented (and as efficient as possible) on simple Uarch designs without hobbling performant designs.

The question then becomes one of worth for complexity at V1.0.

In this context I believe it is worthy,  especially as Krste remarked for the expanded multiple.

David-Horner, 2020-08-04:
I posted a comment to the closed #
Not everyone subscribes to GitHub, so I post it below,

I am requesting  this proposal be reconsidered/re-evaluated for V1.0 
inclusion in light of the posting:

Some additional comments to the post.

Increased overhead.

    An extra SEW bits need to be distributed to the execution units,
      which on a large VLEN machine could be multiple and physically
    dispersed on the chip.
    More lines to toggle.

Yes, there is extra power, however only once, the scalar values remain 
resident through all successive iterations on different channels.

There is not additional distribution circuitry, the sew=XLEN case will 
have to be wired in and
     is thus available for the sew=XLEN/2 case (which has EEW of XLEN 
for the rs1).

The additional power/complexity/transfer is self limiting, once 
sew>=XLEN no extra SEW bits are transferred.

Potential Usage:

It is not to save hardware (much can be reused), but to increase 
functionality.
We have instructions

# Widening unsigned integer add/subtract, 2*SEW = 2*SEW +/- SEW
vwaddu.wv  vd, vs2,  vs1, vm # vector-vector
vwaddu.wx  vd, vs2,  rs1, vm # vector-scalar

of the form:

VWADDU.WV:    vd(at 2*sew)[i] := vs2.(at 2*sew) [i] + zext((to 2*sew) 
vs1(at sew width) [i])
VWADDU.WX:    vd(at 2*sew)[i] := vs2.(at 2*sew) [i] + zext((to 2*sew) 
narrow((to sew bits) rs1))

the WX form would become:

VWADDU.WV:    vd(at 2*sew)[i] := vs2.(at 2*sew) [i] + narrow((to sew 
bits) rs1)

It effectively becomes a 2sew add scalar and replaces the sequence:

vsetvli 0,0, sew2n
VADDU.VX
vsetvli 0,0, sewn

In general using 2sew of rs1 allows a scalar input range commensurate 
with the target rather than the source in the vector widening operation.

Github Posting: 
https://github.com/riscv/riscv-v-spec/issues/427#issuecomment-666487664

Comments on resolution and request to re-frame

    We discussed proposal to widen scalar input to widening operations to
    2 * SEW.

    For widening multiplies, this would double the size of
    multiplier arrays required.

When SEW < XLEN, I noted that the double size multiplier arrays 
(conceptually)
[... 85 lines cut ...]
with the SEW-vector on the narrow multiply. Thus zero/sign high bits of 
scalar are not a selection between LARGE/narrow but rather an 
optimization of the narrow process.

The optimization of narrow multiplies can be incorporated independently 
for various sizes of SEW. The cost to add the upper X-register SEW-bits 
is nominal at 8bits and still small at 16 bits. For a RV32 these are the 
only two integer widening of concern for 2 * SEW scalar. For RV64 the 
only other integer widening 2 * SEW is 32 bit.
A tradeoff between

 1. half throughput (use next level up full multiplier), or
 2. (as above) conditional temporal, or
 3. parallel partial-product generation and fast sum hardware
    can be chosen independent of upper 32 bits of X register.

Re-framing the proposal in these terms changes the question from a 
dichotomy to a continuum of design options that can be effectively 
implemented (and as efficient as possible) on simple Uarch designs 
without hobbling performant designs.

The question then becomes one of worth for complexity at V1.0.

In this context I believe it is worthy, especially as Krste remarked for 
the expanded multiple.