riscvarchive/riscv-v-spec #429: Clarification of legal SEW values for vfmv.f.s and vfmv.s.f

JamesKenneyImperas, 2020-04-23:
Most vector floating point operations require that the SEW is a supported floating point size in the base. Can you please confirm the expected behaviour for `vfmv.f.s` and `vfmv.s.f` in this respect? The specification for `vfmv.f.s` is worded like this:

> The vfmv.f.s instruction copies a single SEW-wide element from index 0 of the source vector register to a destination scalar floating-point register. If SEW > FLEN, vfmv.f.s substitutes an FLEN-bit canonical NaN if the element value is not correctly NaN-boxed for FLEN. If SEW < FLEN, the value is NaN-boxed (1-extended) to FLEN bits.

which implies that SEW > FLEN is legal. Similarly, for `vfmv.s.f`:

> The vfmv.s.f instruction copies the scalar floating-point register to element 0 of the destination vector register. If SEW < FLEN and the value is not correctly NaN-boxed for SEW bits, an SEW-bit canonical NaN is substituted. If SEW > FLEN, the value is NaN-boxed (1-extended) to SEW bits. The other elements in the destination vector register ( 0 < index < VLEN/SEW) are unchanged. If vstart ≥ vl, no operation is performed and the destination register is not updated.

which again implies that SEW > FLEN is legal. Is that right?

If these instructions do indeed deviate from the normal rules, then I think it would be helpful to explicitly state any constraints on minimum and maximum SEW.

(I have a feeling I might have asked something about this before but can't find the issue - apologied in advance if so.)

kasanovic, 2020-05-05:
Yes, SEW > FLEN is supported (SEW goes up to 1024 bits).   In vector instructions, any vector destination has to have an EEW of a supported floating-point type, so NaN-boxing doesn't appear except where explicitly noted.

JamesKenneyImperas, 2020-05-05:
Thanks, Krste.

I'm still getting to grips with the new EEW terminology, but if the vector element for these instructions has to have an EEW of a supported floating-point type, then under what circumstances does the statement _If SEW > FLEN, the value is NaN-boxed (1-extended) to SEW bits_ have an effect? Doesn't the vector element have to be a supported floating-point type, in which case since it must also by definition be supported in the base, it can't be > FLEN? Could you perhaps give a concrete example where this constraint applies and has an effect?

I think you are saying that these instructions are following the normal floating point instruction rules and that the vector element must be a supported floating point size in the base - if so, I think the statements I've highlighted are describing cases that cannot occur.

kasanovic, 2020-05-07:
I think the confusion is that vfmv.f.s/vfmv.s.f are considered _transfer_ instructions that do not imply that floating-point _arithmetic_ operations are supported at the widths they support.

It could be argued that the vfmv* instructions should be illegal if SEW>FLEN.  I think I'd be OK with that constraint.

JamesKenneyImperas, 2020-05-07:
Yes, I think they should either be illegal or it needs to be explicitly stated what the statements about NaN-boxing mean in these cases. For example, in an extreme case when the EEW is 1024 and FLEN is 32, does `vfmv.s.f` NaN-box the result in bits 1023:32?

There is also the other case to consider of SEW<FLEN. For example, if EEW is 8 and FLEN is 32, what does `vfmv.f.s` do, if this is legal? What does it even mean to NaN-box an 8 bit value? Again, this should be explicitly stated if it is desired behavior.

aswaterman, 2020-05-07:
My preference would be to treat vfmv, vfmerge, etc. behave the same as other FP ops and require that SEW correspond to a supported floating-point type.  (Along the same lines, I'd prefer these instructions check for a valid value in `frm`, even though they don't round.)  There's little utility in relaxing those constraints, and instruction decoding is simpler if everything in the OPFVV/OPFVF opcode spaces behave the same in these respects.

JamesKenneyImperas, 2020-05-07:
I agree with Andrew.

kasanovic, 2020-07-03:
Cleaned up by: https://github.com/riscv/riscv-v-spec/commit/50fab57354c3d6d136de5b844f843e59acf2846e#diff-4fa320b4e415eee7270590bb06607389

closed 2020-07-03 by kasanovic