riscvarchive/riscv-v-spec #431: Clarify version features

JamesKenneyImperas, 2020-04-26:
As part of this change, the specification version has been updated from 0.9-draft-20200221 to 0.9-draft-20200424. Can you please clarify which features are in which version?

Because of the fence post problem, I am never quite sure whether the version is intended to draw a line under past features or specify a baseline for new features. It would be very helpful if, as part of a change to the version, you can clearly state your intention.

In this specific case, which versions contain:
1. Removal of non-SEW loads and stores; and
2. Fractional LMUL and associated changes.

(I'm asking this because in riscvOVPsim we have a configuration parameter allowing selection of specification version. It is possible using this to select any intermediate version from 0.7.1 to the current master. Usually I have been able to work out from the context where the cut lines are, but in this case I cannot.)

kasanovic, 2020-04-27:
We previously agreed to have three levels of versioning:
* the v0.x version, substantial milestones in the spec development, tagged in history
* -draft-<date> versions, approximately tied to results of TG meetings, no more frequent than every couple of weeks, intended to reflect significant updates
* git commit hashes, individual commits

In all cases, they reflect the status for the commit on which they're added .
If you want to be very specific, use the git hashes.  The others are human-friendly tags for that point in history.  

Maybe you're asking to tag the repo on every date change version?

JamesKenneyImperas, 2020-04-27:
Hi Krste,

No, I'm not asking for any more tags: I just would like clarity on what the tags mean. For example, your latest specification version bump was done in the middle of the fraction LMUL commits. Was the intention that:
1. The tag 0.9-draft-20200424 is intended to represent the final state _before_ those commits; or:
2. The tag is intended to represent the new baseline state _including_ those commits?
From what you have said, I think it is 2, but I'm not sure.

kasanovic, 2020-04-27:
Any version cannot include future updates, so must only refer to everything up to the point of the commit that changed the version.

Krste

kasanovic, 2020-04-27:
The changes after the date change are only clarifications so far, including
the constraint on fmul. The actual spec didn't change.

Krste

| Hi Krste,
| No, I'm not asking for any more tags: I just would like clarity on what the tags mean. For example, your latest specification version bump was
| done in the middle of the fraction LMUL commits. Was the intention that:

|  1. The tag 0.9-draft-20200424 is intended to represent the final state before those commits; or:
|  2. The tag is intended to represent the new baseline state including those commits?
|     From what you have said, I think it is 2, but I'm not sure.

| —
|

closed 2020-04-30 by JamesKenneyImperas