riscvarchive/riscv-v-spec #432: vtype, vsetvl[i] and backwards-compatibility

JamesKenneyImperas, 2020-04-26:
I'm still digesting the fractional LMUL changes, but I wonder whether the proposed changes to `vtype` (and therefore `vsetvl[I]`) should be reconsidered. The proposed change makes all existing tool chains incompatible with the latest specification because fields have been moved and resized. If the indication for fractional LMUL was instead moved to bit 7, with all other fields kept in there previous locations, then existing tool chains would still work.

aswaterman, 2020-04-27:
I also think we should consider this option. Depending on the nature of the other backwards incompatible changes, this could make a material difference on how much effort it takes to migrate to v0.9.

David-Horner, 2020-04-27:
It becomes moot with #418  - vlmul goes away.

commit a5a62bd8 (2020-04-30): Reverted lmul backwards-incompatibility.
Change bit used to indicate fractional lmul to an additional non-contiguous bit in encoding to keep backwards compatibility.
Also, moved all references to EDIV (including vtype location) out of base spec and into EDIV extension section.
Closes #432.
closed 2020-04-30 by kasanovic