riscvarchive/riscv-v-spec #434: Should SLEN=VLEN be an extension?

kasanovic, 2020-04-27:
(I'm summarizing an issue that has popped up in several public and private conversations)

SLEN<VLEN introduces internal rearrangements to reduce cross-datapath wiring for wide datapaths such that bytes of different SEWs are laid out differently in register bytes versus memory bytes, whereas when SLEN=VLEN, the in-register format matches in-memory format for vectors of all SEW.

Many vector routines can be written to be agnostic to SLEN, but some routines can use the vector extension to manipulate data structures that are not simple arrays of a single-width datatype (e.g., a network packet). These routines can exploit SLEN=VLEN and hence that SEW can be changed to access different element widths within same vector register value, and many implementations will have SLEN=VLEN. 

To support these kinds of routines portably on both SLEN<VLEN and SLEN=VLEN machines, we could provide SEW "casting" operations that internally rearrange in-register representations, e.g., converting a vector of N SEW=8 bytes to N/2 SEW=16 halfwords with bytes appearing in the halfwords as they would if the vector was held in memory.  For SLEN=VLEN machines,  all cast operations are a simple copy.  However, preserving compatibility between both types of machine incurs an efficiency cost on the common SLEN=VLEN machines, and the "cast" operation is not necessarily very efficient on the SLEN<VLEN machines as it requires communication between the SLEN-wide sections,  and reloading vector from memory with different SEW might actually be more efficient depending on the microarchitecture.

Making SLEN=VLEN an extension (Zveqs?) enables software to exploit this where available, avoiding needless casts.  A downside would be that this splits the software ecosystem if code that does not need to depend on SLEN=VLEN inadvertently requires it.  However, software developers will be motivated to test for SLEN=VLEN to drop need to perform cast operations even without an extension, so this split will likely happen anyway.

A second issue either way is whether we should add "cast" operations.  They are primarily useful for the SLEN<VLEN machines though are difficult to implement efficiently there; the SLEN=VLEN implementation is just a register-register copy.  We could choose to add the cast operations as another optional extension, which is my preference at this time.

knightsifive, 2020-04-27:
On the mailing list I mentioned one such application, from cryptography: reinterpreting SEW = 32 vectors as SEW = 64 (with half VL), performing 64-bit arithmetic, then reinterpreting the result as SEW = 32. These typecasts succeed if LMUL = 1 or SLEN > 32. (The spec also comments that "in most implementations, the striping distance SLEN >= ELEN".) 

My point is that SLEN = VLEN is sufficient but not necessary for this optimization.

I do support adding a "reinterpret cast" instruction, which changes SEW and LMUL on a register group as if SLEN==VLEN. For example,

```
# SEW = 32, LMUL = 4
v_reinterpret v0, e64, m1
```

would perform "register-group fission" on `[v0, v1, v2, v3]`, concatenating (logically) adjacent pairs of 32-bit elements into 64-bit elements (up to, or perhaps ignoring VL). And we could perform the inverse operation, "register-group fusion", as follows:

```
# SEW = 64, LMUL = 1
v_reinterpret v0, e32, m4
```

It's true that this is implementable by sequences of stores and loads. However, my intuition is that the optimal sequences will be highly microarchitecture-dependent, and as a programmer I'd prefer to trust this analysis to the designer.

I also don't see harm right now in adding this to the base spec: my impression is that all extant implementations take SLEN = VLEN, so the burden right now is just adding a no-op. 

I haven't kept up with the fractional LMUL discussion so don't know how any of this would dovetail.

David-Horner, 2020-04-27:
Observations:
- single SEW operations are agnostic to underlying structure (as Krte noted in recent doc revision)
- mixed SEW operations (widening & narrowing) have substantial impact on contiguous SLEN=VLEN
- mixed SEW operations are predominantly SEW <--> 2 * SEW
- by 2 interleaving chunks in SLEN=VLEN at SEW level align well with non-interleaved at 2 * SEW

Postulate: 
That software can anticipate its need for a matching structures for widening/narrowing and memory overlay model and make a weighed choice.

I call the current interleave proposal SEW level interleave (elements are apportioned on a SEW basis amongst available SLEN chunks in a round robin fashion).

I thus propose a variant of #421    Fractional vtype field vfill – Fractional Fill order and Fractional Instruction eLement Location :

INTRLV defines 4 interleave formats:

- SLEN<VLEN (SEW level interleave)
- SLEN=VLEN (proposed as extension, essentially no interleave)
- Layout is identical to “SLEN=VLEN” even though SLEN chunks exist, but upper 1/2 of SLEN chunk is a gap (undisturbed or agnostic fill).
- Layout is identical to “SLEN=VLEN” even though SLEN chunks exist, but lower 1/2 of SLEN chunk is a gap (undisturbed or agnostic fill).

A 2bit vtype vintrlv field defines the application of these formats  to various operations, the effect is determined by what kind of operation it is:

Load/Store will depending upon mode
```
   vintrvl level = 0 --  scrample/descramble SEW level encoding 
   vintrvl level = 3 --  transfer as if SLEN=VLEN ( non-interleaved) 
   vintrvl level = 1 --  load (as if SLEN=VLEN) lower 1/2 of SLEN chunk 
				(upper undisturbed of agnostic filled)
   vintrvl level = 2 --  load (as if SLEN=VLEN) upper1/2 of SLEN chunk 
				(lower undisturbed of agnostic filled)
```

Single width operations will work on either side of SLEN for vintrvl levels 1 and 2 , but identically on all vl elements for vintrvl level s 0 and 3

Widening operations can operate on either side of the SLEN chunks providing a 2 *  SEL set of elements in an SLEN length chunk (vintrvl levels 1 and 2).
Further, Widening operations can operate with one source on one  side and the other on the other side of the SLEN chunks providing a 2 * SEL set of elements in an SLEN length chunk. ( vintrvl levels 3).

For further details please read #421 which proposed an extended version that supported quad interleave (and octal) and the previous LMUL>1 striped across physical register scheme. .

jnk0le, 2020-04-27:
How about using EDIV for the mentioned type punnig?

David-Horner, 2020-04-29:
@knightsifive 
I very much like this vertical stacking approach.
Elegant, conceptually simple and potentially efficient. 
I suggest that it is sufficient to have just a single bit select for 8 level stacking, e64 support.
All sub elements (half, word) are supported.
(as an aside, issue #418 concept can be used to allow effective doubling of vl to 16 physical registers in this mode)
If this byte width striping can be efficiently implemented on a wide range of micro-architectures it is a most excellent solution.
It is not a world view change. It can coexist with single stack data structures AND be used for all single SEW operations at all SEW sizes.
And widening/narrowing are also supported because stacked structure is built on the underlying SEW level interleave.
Mask is compatible with an "ordinal" based encoding I proposed to Krste offline. 
I wish I had seen the light first time I read through your proposal.

I am keen to have some micro-arch experts chime in.

David-Horner, 2020-04-29:
@knightsifive Update: On third reading what you proposed were specific instructions to:

>perform "register-group fission" on [v0, v1, v2, v3], concatenating (logically) adjacent pairs of 32-bit elements into 64-bit elements

and not the register group stacking I proposed.

However, the "register-grouping" was the inspiration for the idea of vertical stacking as a mode.

David-Horner, 2020-05-06:
I still have concerns about the proposed world shift that occurs when SLEN changes from less than VLEN to VLEN (and back).
The all or nothing approach has an effect comparable to changing harts from BIG to little and back.

The alternative of only providing casting instructions is itself problematic. Many approaches are possible to provide the functionality in instructions or prefixes to existing instructions. The "best" alternative is relative to the micro-architecture approaches and constraints. There are also the concerns of software portability as on SLEN=VLEN only implementations these castings or prefixes are not required and provide bloat (and confusion).

I have no perfect solution, but I suggest one that partially addresses these issues, provided our ecosystem has sufficient compiler support.

Instead of a world shift with the vslen bit change (VLEN or SLEN ) with every register (or remnant of registers) changed between internal formats, I propose only a subset of the registers be required to support both modes.
Specifically, I propose physical registers 24 through 31 be mode switchable.
These 8 registers would appear as either in-memory format, or SLEN dependent in-register format.
In general, after a vslen switch applications should not rely on the contents or residual format of the data in these 8 registers.  Specific mapping approaches may be standardized and then relied upon by applications. However, initially a presumption of invalidation should be assumed as the method used (to implement in-memory order structure from the SLEN format used elsewhere) can be implementation dependent .
For machines which support only SLEN=VLEN the vslen state is a no-op, preserved for application migration among cores.

Mask register 0 will continue to be the default in-register format.
Registers 0 through 23 will also only be in the default in-register format.
Instructions that share registers from both sets of registers, 0-23 and 24-31 will adjust there inputs accordingly and format the destination result accordingly.
In either mode whole register store/load must suffice in re-establishing the vector register state for context switch, etc.

I suggest 8 registers as
- it allows the current maximum LMUL factor to be used
- limits the registers that require any micro-architecture specific adaption.
       (e.g. extra ports, more complex structure, increased wiring)
- limits the number of registers that would be involved in mixed format transforms
       ( moves to / from the two register sets, inputs from different register sets, etc.)
- Although a significant reduction in total registers, use of these 8 registers would only be required for structures that require in-memory structure. 
- For that purpose these 8 registers should suffice. (640K?) 
- The penalty for invalidating 16 (1/2 of register set) appears too great, and 4 register appears too few to support meaningful algorithms requiring in-memory format structures.
- The compiler should avoid the use of these 8 registers when vslen<>default , weighing such use against the performance penalty estimate or known value for the class or specific machine.

I think it is possible to mandate that SLEN<VLEN architects are required to support this limited in-memory support.
If nothing else, minimal support would be emulation in software by having traps when any of the 8 registers are addressed while vslen <> default value. Analogous to what is done in RVI for unaligned memory accesses. 

I am certainly willing to rework this suggestion as a "solid" proposal structured in 5Q&A format.

David-Horner, 2020-05-13:
I am starting an analysis of the various mitigation approaches.

I hope that, at least, the itemizing of them and links to them will be 
helpful.

The remainder is
- a collection of areas of concern (not cross referenced at all to 
whomever has previously raised them)
- a start as establishing the domain of the concerns and grouping 
related issues
- a start at a taxonomy of the concerns, mitigation approaches and areas 
of impact.

I have not progressed in a presentation of the cross-referencing the 
above, especially mitigation methods with percieved benefit/impacts.
However, I wanted to get this out early as a help for others to begin to 
do so within their own minds and areas of expertice.

-----------------------------------------------------------------------

I won’t detail, here, the differences between in-register order vs 
in-memory. See** instead.

Rather I intend to compare the various approaches proposed to address 
the differences.

The alternatives I have seen proposed –

A.  require SLEN=VLEN mode for all implementations (even if achieved by 
reduced VLEN) #434
            https://github.com/riscv/riscv-v-spec/issues/434

B.  limited SLEN=VLEN setting for a subset of registers
https://github.com/riscv/riscv-v-spec/issues/434#issuecomment-624820438

C. an approach that provides modes of operation to help compensate
https://github.com/riscv/riscv-v-spec/issues/434#issuecomment-620122441

D.  restructuring/cast operations that no-op on SLEN=VLEN harts, 
mentioned at end of original post
            https://github.com/riscv/riscv-v-spec/issues/434

E.  a variation from @knightsifive (that notes SLEN>ELEN as fundimental 
to soln)
https://github.com/riscv/riscv-v-spec/issues/434#issuecomment-620083542

F. an approach that provides instruction prefixes to adjust for 
SLEN<VLEN structure
      effectively making any instruction able to transform source and 
destination.
      (discussed by me on mailing list or github comment I cannot 
currently find )

G. Define Cluster size at 64bits to retain byte alignment up to 64bit 
elements.
           https://github.com/riscv/riscv-v-spec/issues/461

H. Use fractional structure of source to widening to maintain lane 
alignment.
            https://github.com/riscv/riscv-v-spec/issues/465
        Further embelished to

I. a variation of the latter that allows the clustering (interleave 
width) to be set dynamically.
              https://github.com/riscv/riscv-v-spec/issues/465  at A5. 
As an objection to proposal.

J.  Not a proposal I have actually seen, but a default position of fall 
back to pre-v0.9-draft-20200424.

[... 89 lines cut ...]
to mitigate this disparity, all retain some risk of fragmentation.

The new SLEN was introduced to support widening operations.

This is the fundamental challenge, how to widen from a given element 
width to an effective width efficiently across multiple 
micro-architectures; avoiding element positioning skew and long wiring 
lengths.

So far vertical and horizontal approaches  to accommodate double width 
results from fully packed source registers has proved challenging. Each 
creates anomalies in the register structure.

The original vertical striping was rigidly allocated powers of 2 
register groups  as it compounded at each level of widening. The applied 
striping length partially mitigated but still allowed for in-register 
structure to mismatch in-memory on a register group and implementation 
specific SLEN basis.

See***
There is a double edged sword of awareness/transparency/clarity of 
implications ability to explain,     and communicate the need to use 
mitigation, etc.
vs counter-intuitively,  making application manage complexity may yield 
greater ease of use/simplicity

kasanovic, 2020-06-28:
Group agreed to mandate SLEN=VLEN layout (not just as an extension).

commit 7facdcc4 (2020-07-01): Made SLEN=VLEN layout mandatory.
In-register byte layout of vectors matches in-memory byte layout.
Closes #434
Closes #507
closed 2020-07-01 by kasanovic