riscvarchive/riscv-v-spec #436: VS field in xstatus

billhuffman, 2020-04-29:
Section 3.2 defines the VS field of the xstatus register.  I would have expected from how the privilege spec is written that VS would be in a vector state and reflected in XS.  I wonder if that is the better way to do this as we don't want to fill the status register with several such states, I think.  Floating-point is fundamental enough to deserve its own field, but I don't know that vector is.

If we do include the VS field as in section 3.2, I think we also have to say whether or not it is reflected in XS.  I assume it should not be, but the privilege spec says XS reflects "additional user-mode extensions" and might need to be updated slightly in that case.

Similarly, we have to say that the VS field is reflected in the xstatus.SD bit.

aswaterman, 2020-04-29:
XS is meant for custom extensions.

Packing VS into mstatus is justifiable.  We are confident there will not be a proliferation of additional fields in mstatus, because new standard extensions are likely to be overlaid on the existing scalar or vector state.

billhuffman, 2020-04-30:
That's reasonable.  But I think the vector spec needs to say VS is or is not included in XS and that it is (I assume) included in SD.

The Privilege spec might then better refer to custom extensions than to user extensions.

aswaterman, 2020-04-30:
Agreed. I’d prefer to make that update at the same time that we add VS to the priv spec.

billhuffman, 2020-04-30:
Sounds fine to me.

kasanovic, 2020-07-03:
Spec has said vs is in mstatus for some time.

closed 2020-07-03 by kasanovic