riscvarchive/riscv-v-spec #439: Clarification on overlap of input mask for comparisons under LMUL>1

rofirrim, 2020-04-30:
Hi all,

it is unclear to us if the sequence below would be valid (assume `a3` and `v0` have sensible values already) under LMUL>1

```
vsetvli	a2, a3, e64,m2
vmsle.vv	v0, v2, v4, v0.t   # Is this OK under LMUL>1?
vadd.vv	v2, v2, v4, v0.t
```

or instead something like this should be emitted

```
vsetvli	a2, a3, e64,m2
vmv1r.v	v1, v0
vmsle.vv	v1, v2, v4, v0.t
vmv1r.v	v0, v1
vadd.vv	v2, v2, v4, v0.t
```

Thank you,

aswaterman, 2020-04-30:
The mask register overlap constraint described in this section https://github.com/riscv/riscv-v-spec/blob/master/v-spec.adoc#53-vector-masking only applies to register groups.  Since the result of a comparison op is not a register group, but just a register, it's OK to overwrite the mask register.

rofirrim, 2020-04-30:
Thanks a lot @aswaterman  That explains it. Feel free to close the issue.

aswaterman, 2020-04-30:
You're welcome!

closed 2020-04-30 by aswaterman