riscvarchive/riscv-v-spec #448: A Simple Ordinal Based Mask Encoding V1a

David-Horner, 2020-05-05:
Q. What is the problem?
A. Ordinal mapping for all combinations of SEW and LMUL is absent.
Ordinal mapping is a desirable trait.
So much so that v0.8 (and 0.7 before it) advocated keeping LMUL to SEW ratio the same, in part so that it would retain ordinal mapping of mask values. 
Q. What's wrong with what's there now?
A. A change in SEW will invalidate the ordinal mapping of masks for a given LMUL register group. 
E.g. Changing SEW size by 1/2 will cause every second mask bit to be ignored. see**
Whereas this may be a useful characteristic that could be exploited by software, the every other bit  mapping varies by SLEN. see*** Making it much less attractive, and the kind of feature previously discouraged in defining the RVV design. Granted, this mapping was quite functional under the previous vertical striped LMUL design, but is unnecessarily complicated under the SEW dependent interleave format. see****
Likewise, A change in LMUL will invalidate the ordinal mapping of masks for a given SEW setting. 

see** 
Ignoring every second bit is apparent from the definition of MLEN =  SEW/LMUL (#443). 
Consider LMUL=1 and SEW=16. MLEN matches SEW of 2 bytes. Set all mask bits  to one with vseteq v0,v0,v0. Now every two bytes has lowest bit set. Now set SEW=8 (using vsetvli). The mask at this SEW level now alternates between set and clear. 

see***
 However, the alternating element bytes are consecutive only if VLEN=SLEN, for VLEN=2 * SLEN the ordinal numbers of adjacent bytes advance by 2, so every 4th element is set (by 8 for 4 * SLEN, etc.)
If we now also halve LMUL, we should be back to MLEN=2 * SEW, and we are.

see****
Note from the MLEN formula above, LMUL chunks of MLEN fit into a SEW chunk. 
The comparatively simple VLEN=SLEN mask address calculation is mask_LSB_index(__i__) =  __i__  * MLEN for the ordinal number __i__. 
(Note also for LMUL=1 that the defined SEW chunk housed all MLEN with the same SEW/LMUL ratio. This is the base configuration for SEW elements. )

It is apparent that consecutive ordinals are mapped into consecutive MLEN chunks. This previously had physical meaning as consecutive vertical striped elements were in successive registers. MLEN set (LMUL of them) contained within the base SEW element designated what physical register the corresponding element was in. That physical meaning is no longer relevant. 

The mapping to physical registers of MLEN chunks aligned to the base configuration SEW(i when LMUL=1) is dependent upon its ordinal within the base SEW, i, SLEN, VLEN, and two of MLEN, SEW and LMUL. 

Q. What is the solution you're proposing?
A. A simple Ordinal based mask encoding

Q. How is it better?
A. It is simpler. It retains ordinal mapping of mask values for all the values last set that are addressable by the new LMUL and SEW, even when the SEW/LMUL ratio changes.

Q. How is it worse?
A. i) It is a potentially substantive mind shift. 
ii) For the special case LMUL=1, the mask Least Significant Bit no longer aligns with bit zero of every SEW element. The merit of this alignment is reduced now that, with fractional LMUL, the prominent position as the lowest denominator for elements is no longer held by LMUL=1.  As of patch d7dbbd7 #443 , fractional LMUL no longer retains LSBit alignment.  Further, given that multi-precision instructions directly provide corresponding V0 mask bit to any SEW at any LMUL the need for the alignment is dubious.
( e.g. - add mask bit to vd :  vadc.vxm vd,vd,x0,v0);
. . . .  -  if mask bit 0 set vd to 0 else -1 : vsbc.vvm vd, vd, vd, v0)
    

The specific proposal:
The mask register remains a single physical register of length VLEN.

The fundamental changes from current specification are:
  
```
current spec:
 MLEN=SEW/LMUL;
mask_LSBit_index(__i__) = (__i__ % (VLEN / SLEN)) * SLEN +
                   floor(__i__ / (VLEN / SLEN)) * MLEN

proposed has 

MLEN=1; 
mask_LSBit_index(__i__) = (__i__ % (VLEN / SLEN)) * SLEN +
.                   floor(__i__ / (VLEN / SLEN) ) * 8 + __i__ % (VLEN/SEW)

When VLEN=SLEN both fomulae for mask_LSBit_index(__i__) are simplified:
Current:           __i__  *  SEW/LMUL
Proposed:        __i__ * 8    + __i__ % (VLEN/SEW)
	
```
Due to this simple ordinal mapping and MLEN=1,:

Element __i__ , regardless of  SEW, share the same mask value. 
In addition, the mapping is independent of LMUL.
Dependent upon VLEN, SLEN, SEW and __i__ (that is vl).

Sets of elements can be processed under mask, in smaller register groups (smaller LMUL) reducing register pressure.
A logical right shift of n on the mask register reduces by n the number of physical registers mapped, effectively shifting to the lower registers the mapping of the higher registers. SEW=8 allows this to be performed most simply, but in many cases a larger SEW is possible as bits shifted from byte to byte will be ignored by subsequent operations. If elements in higher physical registers than the mask are expendable, then an LMUL=1 change can also be avoided.

kasanovic, 2020-05-13:
I've been working through various mask designs and agree we have to move to a design with MLEN=1.  A big problem with current design is that to work for fractional LMUL, MLEN grows to large values  (e..g, for SEW=8, LMUL=1/8, MLEN=64b).

In SLEN=VLEN designs (i.e., with no spatial datapath sections), we can simply have mask for element i lives at bit i.  For designs with spatial datapath sections (SLEN<VLEN), we map bit I to the appropriate section (i is strided across sections). Within each section, the bits are simply packed in contiguous order for the element indices mapped to that section.  These mappings are independent of SEW or LMUL.

This does require potentially bit-level updates of the mask register, but they were needed anyway when MLEN=1.

closed 2020-05-13 by kasanovic

David-Horner, 2020-05-14:
I totally botched the formula (and my description of **it** when I included SEW in the formulation).
It was total rubbish.

I appreciate the idea was acknowledged.
However, with the correct formulation the **characteristics** that I described may be apparent.
Aspects of the description may still be useful for reference as I reply to the new MLEN=1 update.

The fundamental changes from current specification are:
  
```
...
proposed : 

MLEN=1; 
mask_LSBit_index(__i__) = (__i__ % (VLEN / SLEN)) * SLEN +
                    (floor ( __i__ * SLEN / VLEN) * 8) % SLEN + 
             floor( (floor ( __i__ * SLEN / VLEN) * 8)/  SLEN)  ) / 8

When VLEN=SLEN ... for mask_LSBit_index(__i__) [is] simplified:
...

mask_LSBit_index(__i__) = 0 +
                    (        __i__                * 8) % VLEN + 
             floor( (        __i__                * 8)/  VLEN)  ) / 8
```

commit 9a77e128 (2020-05-13): MLEN=1 update

This is a new simpler mask register layout that addresses many of the
concerns with the earlier mask encoding.

Closes #448.