riscvarchive/riscv-v-spec #449: Expand sign injection instructions to operate on integer values and any supported sew.

David-Horner, 2020-05-05:
A1. Sign injection instructions are currently limited to Float and Float SEW size.

A2. Float format is not the only signed magnitude format, nor data type that can benefit from high order bit injection.
     To restrict the instruction to only float SEW limits its use in other useful domains (e.g. biased arithmetic).

A3. Expand sign injection instructions to operate on integer values and any supported SEW.

Use new names vsgnj.vv, vsgnjn.vv, vsgnjx.vv for current vector-vector operators, 
    retaining aliases of vfsgnj.vv, vfsgnjn.vv and vfsgnjx.vv for float.
These new instructions will operate as before but also support all SEW widths.

Add three new vector-scalar integer instructions vfsgnj.vx, vfsgnjn.vx and vfsgnjx.vx.
SEW is allowed to exceed XLEN for these operations. 

Retain the current 3 float instructions vfsgnj.vf, vfsgnjn.vf and , vfsgnjx.vf.

A4. This is better as it provides support for more signed magnitude datatypes and provides a generic vector-vector high order bit operator. 

A5. Supporting the 3 vector-scalar-integer would increase bloat for a further datatype that otherwise has limited instruction set support.
       Hardware restriction on the alias “float” instructions would be removed. However, compile time checks would remain.

kasanovic, 2020-05-13:
The instructions _are not_ limited to float - the instructions do not interpret the bits as a datatype.

They _are_ currently only defined for SEW sizes that are supported for float.

While we could relax this constraint, and require they're supported for any supported SEW, this would increase implementation design/verificatiion complexity for limited utility.

kasanovic, 2020-06-28:
Spec will have blanket statement that behavior of the unsupported width/operation combinations is "reserved" to allow future expansion.

David-Horner, 2020-06-29:
As tagged, I agree with @kasanovic intent to reserve the unsupported width/operation combinations.

I appreciate the label that keeps this issue open for V1.0 consideration.
Opcode mapping consistency could be retained if implemented before V1.0. (Lost if post v1.0 as it is in the float group now)

from above:
>provides a generic vector-vector high order bit operator.

in effect also states consistency as a benefit offsetting incremental cost..

I note that RVI (and RVC) weighed heavily towards consistency at the expense of verification (e.g. x0 ubiquity). 
However, hardware implementation simplicity and opcode reduction trumped consistency (especially orthogonality).

I fully agree that there is an incremental cost.

I disagree that hardware implementation complexity is greater when supporting all SEW widths.
Perhaps it is more disruptive at this late stage to existing efforts that tied the instruction to float only.

I do agree that software support is increased nominally at assembler level and intrinsics, but only at the compiler level if a mod is made to use the functionality generally, and that will be a classic tradeoff proposition..

kasanovic, 2020-07-03:
For now, these are reserved (cause illegal instruction exceptions), which simplifies control logic.

closed 2020-07-03 by kasanovic