riscvarchive/riscv-v-spec #450: Question about sgemm.S test

WilliamWangPeng, 2020-05-05:
Hello guys
     My honor to ask question here, I'm a student in riscv vector group, nowadays I'm doing rvv-benchmark test, which passes all 6 tests except sgemm.S;
typedef unsigned long size_t;

int printf(const char *restrict format,...);

extern void sgemm_nn(size_t i, size_t j, size_t k, const float* aa, size_t lda, const float* bb, size_t ldb, float* cc, size_t ldc);

int main(void){
    printf("//sgemm,S---------------------// \n");
        size_t i=2;
        size_t j=2;
        size_t k=2;
        const float aa[4] = {1, 2, 3, 4};
        size_t lda = 4;
        const float bb[4] = {2, 2, 2, 2};
        size_t ldb = 4;
              float cc[4] = {1, 1, 1, 1};
        size_t ldc = 4;

        sgemm_nn(i, j, k, aa, lda, bb, ldb, cc, ldc);

        for(int i=0; i<4; i++)
        printf("%f, ", cc[i]);
        printf("\n");
        return 0;
}
 the error is 
/tmp/main-11c812.o: In function `main':
main.c:(.text+0x602): undefined reference to `sgemm_nn'
clang-10: error: linker command failed with exit code 1 (use -v to see invocation)
    Anyone knows what happened here and how to correct this?
Best Regards
William
https://github.com/riscv/riscv-v-spec/blob/master/example/sgemm.S

knightsifive, 2020-05-05:
It looks like you're using Clang/LLVM. I'm not familiar with Clang/LLVM support for RVV so I can't even say what you're attempting is possible. Personally, I've only used the GNU toolchain, in particular,
https://github.com/riscv/riscv-gnu-toolchain/tree/rvv-0.9.x 

Your linker error does not sound RVV-specific, perhaps you forgot to include `sgemm_nn.o` on the link line? Please share all the commands you typed to compile and link. Again, I cannot guarantee this is possible with Clang/LLVM.

Also, your definitions of `aa`, `bb`, and `cc` suggest you've set `lda`, `ldb`, and `ldc` incorrectly. The signature of `sgemm_nn` is specified as:
```
# void
# sgemm_nn(size_t n,
#          size_t m,
#          size_t k,
#          const float*a,   // m * k matrix
#          size_t lda,
#          const float*b,   // k * n matrix
#          size_t ldb,
#          float*c,         // m * n matrix
#          size_t ldc)
#
#  c += a*b (alpha=1, no transpose on input matrices)
#  matrices stored in C row-major order
```
In more detail, 
- `a` is a `m`-by-`k` submatrix of a `??`-by-`lda` matrix,
- `b` is a `k`-by-`n` submatrix of a `??`-by-`ldb` matrix, and
- `c` is a `m`-by-`n` submatrix of a `??`-by-`ldc` matrix;

because the three larger matrices ("supermatrices"?) are stored in row-major layout, 
https://en.wikipedia.org/wiki/Row-_and_column-major_order
their numbers of rows are irrelevant to indexing: this is why I wrote `??`.

knightsifive, 2020-05-05:
Also, keep in mind that the `sgemm.S` test assumes the number of rows (`m`) is a multiple of 16:
```
    # Handle end of matrix with fewer than 16 rows.
    # Can use smaller versions of above decreasing in powers-of-2 depending on code-size concerns.
end_rows:
    # Not done.
```
Your code takes `m` to be 2, so I assume it will get an incorrect answer.

WilliamWangPeng, 2020-05-22:
hi Nick Knight
    Thank you for your detailed reply, recently I use riscv64-unknown-elf-gcc to compile my main.c, spike pk main and it works.
    Also, ”clang --target=riscv64-unknown-linux-gnu -S main.c -o main “ worlks for these six assembling functions memcpy.s           strlen.s     vvaddint32.s saxpy.s      strcpy.s     strncpy.s except sgemm.S 
    my test.c only for sgemm.S is as following, could you give me a typical example or point out where I need to correct for this example?  thank you best regards

typedef unsigned long size_t;
int printf(const char *restrict format,...);
extern void sgemm(size_t i, size_t j, size_t k, const float* aa, size_t lda, const float* bb, size_t ldb, float* cc, size_t ldc);
int main(void){
        printf("\033[1m\033[45;33m rvv-benchmark \033[0m\n");

printf("//sgemm,S---------------------// \n");
        size_t i=16;
        size_t j=16;
        size_t k=16;
        const float aa[16] = {1};
        size_t lda;
        const float bb[16] = {2};
        size_t ldb;
                 float cc[16] = {3};
        size_t ldc;

        sgemm_nn(i, j, k, aa, lda, bb, ldb, cc, ldc);

        for(int i=0; i<4; i++)
        printf("%f, ", cc[i]);
        printf("\n");
        return 0;
}
               

[image]

WilliamWangPeng, 2020-05-22:
The main test function of the six assembling functions memcpy.s strlen.s vvaddint32.s saxpy.s strcpy.s strncpy.s  is here, and use command here, you can have a look when you want to make rvv-benchmark test
riscv64-unknown-elf-gcc main.c -o main  
spike pk hello  

typedef unsigned long size_t;

int printf(const char *restrict format,...);

extern char* strcpy(char *d, const char *s);
extern char* strncpy(char *d, const char *s, size_t n);
extern void vvaddint32(size_t n, const int *x, const int *y, int *z){
        for (size_t i=0; i<n; i++)
        { z[i]=x[i]+y[i];  }
}

extern void* memcpy(void* dest, const void* src, size_t n);
extern size_t strlen(const char* str);

extern void saxpy(size_t n, const float a, const float* x, float* y){
        size_t i;
        for (i=0; i<n; i++)
                y[i] = a*x[i] + y[i];
}

int main(void){

        printf("\033[1m\033[45;33m rvv-benchmark \033[0m\n");
//--------------------------//

        const char srcstr[] = "ABCDEFGH";
              char dststr[] = "abcdefgh";

        printf("//strcpy.s ----------------------// \n");
        strcpy(dststr, srcstr);
        printf(" srcstr[] = ABCDEFGH, dststr[] = abcdefgh, after calling strcpy.s     \n ");
        printf("dststr[] = ");
        printf("%s \n", dststr);

//---------------------------//

        printf("//strncpy.s---------------------// \n");
        const char srcnstr[] = "ABCDEFGH";
              char dstnstr[] = "abcdefgh";
        strncpy(dstnstr, srcnstr, 3);
        printf(" srcnstr[] = ABCDEFGH, dstnstr[] = abcdefgh, n=3,  after calling strcpy.s     \n ");
        printf("dstnstr[] = ");
        printf("%s \n", dstnstr);

//---------------------------//
        printf("//vvaddint32.s---------------------// \n");
        size_t n=5;
        const int x[5]= { 1, 2, 3, 4, 5};
        const int y[5]= { 1, 1, 1, 1, 1};
              int z[5];
        vvaddint32(n, x, y, z);
        for(int i=0; i<n; i++)
        printf("%d, ", z[i]);
        printf("\n");
//---------------------------//
        printf("//memcpy.s and strlen.s---------// \n");
        char* src="hellowilliam";
        char dest[20];
//      size_t m=5;
         size_t m= strlen(src)+1;
        memcpy(dest, src, m);
        printf("%s \n", dest);

//---------------------------//
        printf("//saxpy.s---------------------// \n");
        const float xx[7] = {7, 7, 7, 7, 7, 7, 7};
              float yy[7] = {1, 1, 1, 1, 1, 1, 1};
        const float a=2;
        size_t p=19;
        saxpy(p, a, xx, yy);
        for(int i=0; i<p; i++)
        printf("%f, ", yy[i]);
        printf("\n");
//---------------------------//
        printf("//---------------------------// \n");

        return 0;
}

WilliamWangPeng, 2020-05-22:
> Also, keep in mind that the `sgemm.S` test assumes the number of rows (`m`) is a multiple of 16:
>
> ```
>     # Handle end of matrix with fewer than 16 rows.
>     # Can use smaller versions of above [...]

> Also, keep in mind that the `sgemm.S` test assumes the number of rows (`m`) is a multiple of 16:
>
> ```
>     # Handle end of matrix with fewer than 16 rows.
>     # Can use smaller versions of above [...]

WilliamWangPeng, 2020-05-22:
u@u-virtual-machine:~/tools/test/rvv-benchmark/assemble$ 
```  
riscv64-unknown-elf-gcc test-sgemm.c -o test-sgemm  
```  
the error is 
```  
/home/u/tools/rv64/lib/gcc/riscv64-unknown-elf/10.0.0/../../../../riscv64-unknown-elf/bin/ld: /tmp/cc5zhnMQ.o: in function `.L10':
main.c:(.text+0x41a): undefined reference to `sgemm_nn'
collect2: error: ld returned 1 exit status  
```

knightsifive, 2020-05-22:
Hi @WilliamWangPeng, 

Please understand that this GitHub repository is for discussion concerning the RISC-V V-extension. I believe that your issue is not directly related to the RISC-V V-extension, or even RISC-V in general. Rather, I believe that you have some misunderstandings about the C programming language and compilation/linking with the GNU toolchain. These are difficult and important topics that can take many years of study; unfortunately they are not really relevant to this repository.

Going forward, I first recommend you try out some (non-RISC-V) examples of compiling and running C programs that consist of *multiple* source files, and do some research about how the GNU compiler and linker work. Then I suggest you take a look at the RISC-V reader: http://riscvbook.com/, and perform some Google searches for examples on compiling/linking RISC-V assembly programs and running them on RISC-V simulators (or hardware, if available to you). Finally, I suggest returning to the code examples you quoted above, and considering the following comments:

First, your `ls` screenshot doesn't show any object files (`.o` files) associated with the assembly codes (`.S` files) you've taken from the RVV spec. This indicates to me that you have not successfully compiled any of the assembly codes. This is reinforced by the suspicious definitions at the top of your `main.c`:
```
extern char* strcpy(char *d, const char *s);
extern char* strncpy(char *d, const char *s, size_t n);
extern void vvaddint32(size_t n, const int *x, const int *y, int *z){
        for (size_t i=0; i<n; i++)
        { z[i]=x[i]+y[i];  }
}

extern void* memcpy(void* dest, const void* src, size_t n);
extern size_t strlen(const char* str);

extern void saxpy(size_t n, const float a, const float* x, float* y){
        size_t i;
        for (i=0; i<n; i++)
                y[i] = a*x[i] + y[i];
}
```
Why did you define `vvaddint32` and `saxpy` but not the other four? I claim that your `main.c` is leveraging a mixture of compiler built-ins (`strcpy`, `strncpy`, `memcpy`, and `strlen`) plus your custom definitions of `vvaddint32` and `saxpy`, and that you aren't actually invoking any RVV codes at all!

Once you understand and resolve this issue, I think that you will also understand how to resolve your `sgemm_nn` issue.

Second, your `saxpy` output is displaying some very strange results, related to your choice of `p == 19`:
```
        const float xx[7] = {7, 7, 7, 7, 7, 7, 7};
              float yy[7] = {1, 1, 1, 1, 1, 1, 1};
        const float a=2;
        size_t p=19;
        saxpy(p, a, xx, yy);
        for(int i=0; i<p; i++)
```

Third, there's still a bug in your invocation of `sgemm_nn`. It has changed from the previous bug, but is still covered by my earlier comment, https://github.com/riscv/riscv-v-spec/issues/450#issuecomment-624330465 .

Lastly, you attempted to run `./main` from the command line. The program `main` was ostensibly compiled to run on a RISC-V vector machine, or a simulator like Spike. I can't tell your host platform from your screenshot, but I'm almost certain it was not a RISC-V vector machine.

Good luck!
Nick Knight

knightsifive, 2020-06-13:
A simpler matmul example was recently proposed: https://github.com/riscv/riscv-v-spec/issues/495

This example has a simpler API and features test codes showing how to allocate, initialize, and display the inputs/outputs.

kasanovic, 2020-06-28:
Please move discussion to mailing list - probably sw-dev is best fit.

closed 2020-06-28 by kasanovic