riscvarchive/riscv-v-spec #451: loss of decoupling in mask instructions

jnk0le, 2020-05-05:
Even though reduction instructions are designed to avoid decoupling loss, the `vpopc.m` and `vfirst.m` still write their results directly scalar regfile. In consequence we end up with pipeline stalls or uops in ooo.
This decoupling is especially destructive in case of deep temporal InO or where only `vmv.x.s` and `vmv.f.s`  can write to scalar without the stalls from skewed execution stages.

With current `EEW` concept those instructions can write `XLEN` sized results into `vd[0]` regardless of current `SEW`.

kasanovic, 2020-05-07:
The common use case for vpopc.m and vfirst.m is to determine scalar control flow or to produce a value to write to vl etc.  In this case, the loss of decoupling is inherent in what the code is trying to do and writing to the scalar register directly helps reduce the LoD.

 I'm not sure how useful these instructions are writing to vd[0].  Where would this be used?

The other mask instructions vms?f.m were designed to support some kinds of data-dependent loop-exit processing without the LoD (see section 6.8 of my thesis).

jnk0le, 2020-05-07:
> I'm not sure how useful these instructions are writing to vd[0]. Where would this be used?

Most notably to reduce nastyness of microarch implementations. Especially when writeback to scalar core can be scheduled few instructions later to not break tail-chain, instruction flow etc.
Figure 6.9 and 6.11 from [1] shows the 6 instruction hole between vfirst and use of its result. In RVV examples:
- strncpy: 4 (+2 possible) 1 of them is `m8` vector store
- strlen: 1 (+1 possible)
- strcpy: 4 (+1 possible ) 1 of them is `m8` vector store

>Note how all vector instructions following the first.f can be issued before the scalar unit receives the result of the first.f instruction

I hardly belive that this is possible on InO with precise (and somewhat deterministic) exceptions (assuming no stalls on/before issuing `vfirst`).

And because instructions are always used according to designers intentions. We are about to see popcnt reductions based on vpopc.m due to the lack of vpopc.vv.
In this case we could slide `vpopc.m` results (8x for `lmul`==8 dataset) within one reg group and do `vredsum` on it. Otherwise we have to do 7 scalar additions per vl chunk. This is very likely gonna fall into the "faster than Y but only on dev machine" category anyway.

I'll leave link to what we are talking about for others:
[1] http://people.eecs.berkeley.edu/~krste/thesis.pdf

jnk0le, 2020-05-07:
Just realized that we need another `vmv.x.s` with `XLEN` scaled `EEW` to avoid excess vsetvli.

aswaterman, 2020-05-07:
> I hardly belive that this is possible on InO with precise (and somewhat deterministic) exceptions (assuming no stalls on/before issuing `vfirst`).

I've already built an in-order implementation with precise exceptions that does exactly this.

jnk0le, 2020-07-28:
Ok, but if this kind of coupling is that easy/cheap/efficient and of course scalable then why not make reductions to also use that path for writeback into scalar?

I also think that this issue has to be resolved prior to 1.0. Any backwards compatible addition is pointless.

kasanovic, 2020-07-28:
The vfirst/vpopc instructions produce an integer and are primarily used to drive scalar control flow, so have to get result to x registers.

The vector unit will support types/operators that won't fit in x/f registers, and so useful to have a pattern that reduces into elements of a vector register. 

Reductions via x/f registers would add additional latency into reduction->reduction chain, also lose decoupling when just storing reduction result to memory.

jnk0le, 2021-06-21:
Closing as there is nothing useful for post 1.0 (even now it's a quite a moot)

BTW, one of the `vpopc.m` from datareg happened already here:

https://lists.riscv.org/g/tech-vector-ext/topic/sparse_matrix_vector_multiply/74057261?p=,,,20,0,0,0::recentpostdate%2Fsticky,,,20,2,40,74057261

closed 2021-06-21 by jnk0le