riscvarchive/riscv-v-spec #454: Legality for fractional LMUL when LMUL < SEW/ELEN?

kito-cheng, 2020-05-06:
> Implementations must support fractional LMUL settings for LMUL ≥ SEW/ELEN, for the ELEN value at LMUL=1, which ensures there is space to store at least one element. An attempt to set an unsupported SEW and LMUL configuration sets the vill bit in vtype.

Spec said implementations must support fractional LMUL settings for LMUL ≥ SEW/ELEN, but does it mean SEW=64 and LMUL=1/4 is always illegal configuration on ELEN=64 even VLEN is large than 256, or it's implementation dependent?

kasanovic, 2020-05-06:
The constraint is a lower bound that software can rely on.  An implementation could choose to support a greater range of fractional LMUL settings, but software cannot rely on this being supported.  The constraint was set to enable the most common use case of fractional LMUL, which is to allow use of the largest number of architectural registers with all supported types.

kito-cheng, 2020-05-07:
Thanks for clarification :)

closed 2020-05-07 by kito-cheng