riscvarchive/riscv-v-spec #457: How about vamo access to 2 identical byte offset values

weishan-wu, 2020-05-07:
Is it allowed there are more than two identical offset in vector atomic instruction?
For example: 
vs2 = [0x0, 0x0, 0x0, 0x0] that means 4 operations try to write the same address(rs1+0x0) at the same time or in some kind of order. 
But the Vector AMOs  provide no ordering guarantee between element operations.  Does that mean the result is randomized by the implementation?

aswaterman, 2020-05-07:
Correct.

However, I wonder if we should strengthen this to impose locally sequential semantics when the wd bit is set. (AMOs to disjoint addresses could still be reordered in the global memory ordering in this case.)

kasanovic, 2021-06-07:
Zvamos have been removed from v1.0 draft.  This issue being labelled as post v1.0.