riscvarchive/riscv-v-spec #458: Novel LMUL encoding in vsetvli to "future proof" immediate field

David-Horner, 2020-05-07:
A1.
There are limited immediate bits in the vsetvli instruction. 
Early use of bits will become entrenched in the design. Misuse cannot be corrected later.
 Extensions to RVV will undoubtedly wish a single mechanism to both set vtype and establish the appropriate vl. Extensions such as complex and quaternion numbers affect vl calculation, and can leverage all of integer, fixed-point and especially float data formats for add/substract, multiply, divide (reciprocals) and share conjugate and norm. 
There may be many other such data-types. 
It is fully possible that the proliferation of modes and datatypes will exhaust the currently remaining 3 bits. See**

A2.
SEW and LMUL are essential opcode modifiers. However, together they use 8 of the 11 available immediate bits in vsetvl, even though a dense encoding is used. This is undesirable. Finalizing this encoding will entrench other bits in the instruction making them unavailable for future use via innovative encoding.

A3. 
An alternate encoding specifically for LMUL is here presented. (Whereas SEW could be similarly encoded, LMUL is proposed as it appears the most constrained. See*** )

Proposal: 
1) LMUL be relocated in vtype from the lower 11 bits to within the range (XLEN-2):11
2) Remove LMUL encoding from the immediate field (thus freeing three bits there)
3) Encode LMUL in the high 3 bits of rs2 register designation.
      To be clear, NOT the bits in the register that provides the AVL value, 
                but the rs2 field that specifies that register. 
4) The 3 high bits are encoded as follows:

upper 3 bits | LMUL code |  rs2 register designates |
-|-|-|
 000 | reserved | 0,1,2,3
 001 | 1/8| 4,5,6,7
 010 | 1/4| 8,9,10,11
 011 | 1/2| 12,13,14,15
 100 | 1| 16,17,18,19
 101| 2| 20,21,22,23
 110 | 4| 24,25,26,27
 111 | 8| 28,29,30,31

Note: the reserved value includes rs2 register designations 0, 1 and 2 which are special purpose registers in RVI and RVC. Register 3 is also included.  

This is a modification of #418 (that does not include additional multiplier values 3,5,6 and 7)

A4.
- Novel rs2 encoding must occur before v1.0, this proposal therefore is timely.
- This novel encoding removes the bit pressure in immediate fields.
- The field for rs2 is used as immediate values in RVI S-format.
- This encoding in rs2 is consistent with the approach in #456, vsetvli like instructions using rs1 and rs2 for immediate encoding.
- each of the defined codes allows any one of 4 different registers to encode the LMUL value.
- only one of 32 registers needs to be specifically engaged for any given vsetvli execution.
- this design trades off RVI register scheduling for RVV flexibility/encoding space.
- the encoding in rs2 is static and can have as effective look ahead as immediate field encoding.
- although this encoding is more intrusive, RVI, RVC and RVV have similar precedents. see**** 
- the unusual dual purpose for this field it mitigated by the prominence of vsetvli. It is not a fringe instruction that has peculiar encoding, but a central one in which the unusual encoding is prominent and central to its function.

- in summary, the constraint on the program to use one of 4 various alternative registers and the minimal register pressure thus caused are a reasonable and legitimate trade-off to provide the compressed encoding in a 32bit space for the envisioned 64bit Vector ops.

A5.
Negatives are 
- unusal encoding
- dual purpose field
- additional RVI support required to potentially free a required register.

see**
A primary role for this instruction is to determine the appropriate vl stored in rd given r2 (the application vector length, AVL) and sub fields of the remaining immediate. One of the 12 bits in the base RVI I-format is consumed to identify vsetvli (from vsetvl and other possible instructions).
Of the remaining 11 bits, immediate 

see***
SEW provides Standard Element Widths of 8 to 1024. 256 through 1024 provide dubious value for the currently defined integer and fixed.point operators, and no value for float operators. As a result these encodings could be reserved for novel compression approaches. LMUL expresses a multiplicative range of 1/8 to 8, 7 discrete values in 3 bits. One value is available for escape, or special state. Granted 1/8 is also of dubious value but as it can be as considered close to free, it apparently has been. It provides limited value, but at low cost and as there is already an unused encoding why make an effort to preserve an additional encoding?

see****
The classic RVI has designation of x0 as un-writable zero value, and uses this constant to simulate various simpler instructions and hints.
RVC register subset ,and especially special casing x2 as stack pointer, with expanded offsets.
RVV trades off extra RVI instructions for RVV encoding of "Sequences to synthesize `vmsge{u}.vx` instruction". 
   
Krste asked for proposals (issues?) from me (everyone?) to be in the (implicit) answer format:
Q1. What is the problem?
Q2. What's wrong with what's there now?
Q3. What is the solution you're proposing?
Q4. How is it better?
Q5. How is it worse?
So if these posts appear in a stilted, akward , contrived format for the particular subject material, please forgive my deficiencies in working within this framework.
Further, as is often the case, problems can be identified that do not have an apparent optimal solution. These are particularly difficult to present in this format (and such bleed-through is seen in parenthesized text above.

kasanovic, 2020-05-13:
I don't like the idea of limiting rs1 register names now to leave space for more bits later.  As long as one or more immediate fields in vsetvli have a reserved encoding, we can always reclaim rs1 bits later by declaring the reserved encoding to modify the instruction format.

David-Horner, 2020-05-13:
I agree that we can defer using "rs1 encoding", we currently have bits available.

As we approach v1.0 we should evaluate if we will inevitably exceed 10 bits encoding using only immediate bits, and if so reconsider:
a) whether the recovered 6 bits of rs1 and rd encoding will be sufficient for the life time of ILEN=32
b) we want to support two distinct and competing encodings (for perhaps overlapping settings)
c) if the expanded format will in effect supersede the original encoding, and thus result in dead weight of a low use format to be supported in perpetuity.

Note, If we defer until we only have one bit available we will use that one bit for selecting a mode that we could have chosen from the beginning without that loss of bit. This will also weigh into the considerations above.

I am certainly willing to resurrect this issue on appropriate occasions, such as the rejection of a configuration option due to limited vsetvli immediate field encoding space.

closed 2020-05-13 by David-Horner

David-Horner, 2020-05-13:
the description in #460 refers to and presupposed implementation of this proposal.

Note: the 6 bits referred to above as potentially recoverable assumes #460 is also implemented.
Without #460 only 3 bits would be recovered at the expense of (potentially) one bit used in the immediate field to select the "rs1 encoding" format. Better the bit be used to enable both the rs1 and rd additional encoding thus gaining  a net of 5 bits.