riscvarchive/riscv-v-spec #462: Could we have a new type of arithmetic instruction over the element in a register and another register?

JerryShih, 2020-05-11:
Could we have a new type of arithmetic instruction to support the following formula?
`
vd[i] = vd[i] op vs2[simm5]
`

There is no type of instruction could do the op over "one-element" in a vector register and another register. We should use vslide to move the specific index of element to the first position and use vfmv/vmv to transfer the first element in register to a scalar register. Finally, we could use the OPIVX/OPFVF op for the results.

If we could just use the specific index of vector register directly, we could minimize the data transfer overhead.

There is no enough encoding space for arithmetic instruction(32bit only), so the new instruction could just like **OPIVI**. There is only 5bit index for vs2 register. We could use the last encoding space [12:14](the last **1 1 1** encoding) for this new type of instruction.

`funct6   | vm  |   vs2    |   simm5  | 1 1 1 |    vd   |1010111| OP-V (OPVIndexOP)
`

aswaterman, 2020-05-11:
You can use vrgather.vx/vi to splat any vector element across whole vector,
then use vop.vv. (Implementations are likely to optimize vrgather.vx/vi for
this purpose.)

JerryShih, 2020-05-11:
Yes, I could to the following codes:

e.g.
```
index=4
vd[i] = vd[i] * vs2[index]

vrgather.vi vData, vInput, 4 # vData[i] = vInput[4], i=0 to vl
vmul.vv vOut, vOut, vData # vOut[i] = vOut[i] * vData[i] = vOut[i] * vInput[4]
```

But this approach should use an additional temporary register "vData". And the vrgather is not a fast instruction(one element per cycle?).

aswaterman, 2020-05-11:
It's straightforward to make vrgather.vx/vrgather.vi go much faster than vrgather.vv.

JerryShih, 2020-05-21:
Thank you. It should work well before the new vop sets.

closed 2020-05-21 by JerryShih