riscvarchive/riscv-v-spec #463: how to do content switch on vtype CSR

jeremyccl, 2020-05-11:
Current RVV defines vtype CSR as RO. 
For RTOS doing content switch, the only way I know is to read vtype CSR and store it back by using setvl rd, rs1, rs2.
However, RTOS should not know AVL information, which is not stored in any CSR.
Is there any method defined by V extension to do content switch on vtype CSR?

aswaterman, 2020-05-11:
Use the vsetvl instruction to restore the old context’s vl and vtype CSRs
at the same time. (And since vsetvl resets vstart, vstart needs to be
restored last.)

jeremyccl, 2020-05-11:
Hi Andrew

As I know vl value depends on both AVL and vtype. If RTOS couldn’t get AVL from CSR or somewhere, how to make sure to restore old vl value by using setvl?

Thanks

aswaterman, 2020-05-11:
You don't need the AVL; you just use the preserved value of the vl register as the AVL in the context-switch code.  The effect on architectural state is the same as if you actually used the AVL.

jeremyccl, 2020-05-11:
I got it. Thanks.

closed 2020-05-11 by jeremyccl