riscvarchive/riscv-v-spec #464: SLEN has dractically changes its meaning, should it change its name?

David-Horner, 2020-05-11:
The doc still defines SLEN:

> The striping distance in bits, _SLEN_, which must be VLEN {ge} SLEN {ge} 32, and which must be a power of 2.

This used to mean the size (length) of a chunk of register space that needed filling before the fill would move onto the next physical vertically stacked registers.

I believe that the new meaning of for the parameter is better described as a striding distance. The distance established (minus chunk size) between filling of successive elements within the register. The element size being EEW.
Coincidentally, Striding and Striping have the same initial character. So renaming could leave the same acronym which will not differentiate, but will leave those with an understanding of prior SLEN meaning understandably retaining the previous mental image.
I suggest we choose an alternate acronym for the new meaning.
 HSLEN (for Horizontal Striding LENgth) appears acceptable to me (and differentiates from the previous Vertical Register Striping parameter that we cannot retroactively rename).

Also, the minimum HSLEN of 32 appears impractically small.
HSLEN has to be at least twice the ELEN (that is maximum element width) to be effective on all element widths. Such a value would in effect cause all the negative consequences (of in-register component order mismatching with in-memory) but not provide the benefit of vertically aligning elements of HSLEN or more). 
HSLEN is frequently compared to VLEN, its operational characteristics determined by that ratio Perhaps the fundamental parameter is that ratio. (VLEN:HSLEN as values 1,2,4, etc.). 
Then HSLEN would be a derived term that is useful for noting equality or inequality with VLEN. 
 
Note:This issue is definitely not in 5Q&A, Not amenable to the subject matter . _-Ed_

kasanovic, 2020-05-14:
I disagree that SLEN has changed meaning.  It was always intended to allow implementations to limit the width of data path that wires have to cross to support mixed-width arithmetic operations.  Values are now packed into each SLEN<VLEN section differently than before, but that doesn't change the fundamental meaning of SLEN.  The acronym could be renamed to "section" length.

David-Horner, 2020-05-14:
I don't think intent is sufficient for nomenclature, rather what it actually does; whether the effects are significantly different should determine if distinguishing names should apply. Perhaps drastically was an over statement, certainly a subjective term, but I believe we can agree that the effect on the appropriate range of values is significant.
Because the nature of the "section" has changed from bottom end  (typically many segments of striping integral and essential to widening)  to a few top end ( typically zero) "sections" that are elective, I would still say the change was substantive (if not to the level of drastic). 
Perhaps once the new order is in place the disparity/contrast is moot. I was considering the confusion possible in the interim.
Never-the-less, the minimum of 32 now appears impractically small.
I would continue to suggest that this advisory be dropped if there is no good minimum that can reasonably be suggested.
.

knightsifive, 2020-05-14:
I've never understood the motivation behind the lower bound `SLEN >= 32`. It makes sense that it divides VLEN (so is a nonnegative power of two), but I don't see what breaks if I take, say, `SLEN = 1`.

solomatnikov, 2020-05-14:
I don't think `SLEN < ELEN` makes sense

knightsifive, 2020-05-14:
None of this makes sense to me. But there's currently no (ISA-mandated) lower bound on `ELEN`, so what's wrong with using `SLEN = ELEN = 1` for my (bitvector) VPU?

solomatnikov, 2020-05-14:
Minimum of `vtype.vsew` is for 8-bit elements

knightsifive, 2020-05-14:
Here's how I think it should be: https://github.com/riscv/riscv-v-spec/pull/472

David-Horner, 2020-06-28:
Having decided that SLEN is bring dropped in favour of a non-segmented VLEN design, this question is moot.

closing.

closed 2020-06-28 by David-Horner