riscvarchive/riscv-v-spec #467: Assembly syntax for vsetvli: order between EDIV and vta, vma

kito-cheng, 2020-05-13:
Spec seems not specify the order between EDIV and vta, vma filed.

Which order is right?
```
vsetvli x0, x0, e8, m1, tu, mu, d1
```
or
```
vsetvli x0, x0, e8, m1, d1, tu, mu
```

Current binutils implementation[1] is EDIV always last field, or maybe we could support both order?

[1]  https://github.com/riscv/riscv-binutils-gdb/commit/5116641d171c9a629eb7e96535a8bfc1b1715aa8#diff-3bea72e3cf034d97089e237e60e41f85R12

kasanovic, 2020-05-13:
I have a preference to keep the datatype information (e/m/d) next to each other, with vta,  vma coming last.

I'm also wondering whether "ta, ma" or "ma, ta" is preferable.  Having tail at end might make more sense.

David-Horner, 2020-05-13:
I'm surprised the order would be enforced. Is this just a preferred order,
or required? Is it in this document that such aspects as order are
standardized? Should this not be mandated as is done within RVI, which as I
understand it has spun these considerations off to the toolchain docs?

kito-cheng, 2020-05-14:
The order is enforced in current assembler implementation, I don't have strong preference on that, it's fine to me either relax the order constraint or change the order to `vsetvli x0, x0, e8, m1, d1, mu, tu`.

David-Horner, 2020-05-14:
Each of the parameters are distinct. I assumed that was by design to allow
flexibility in order. Default parameters are already optional. So
simplicity in parsing is not paramount. I see no value to the coder to
enforce a strict order  beyond registers and immediates. Even the mask code
should be allowed flexibility in order, as I see it.

David-Horner, 2020-05-14:
@David-Horner Even the mask code should be allowed flexibility in order, as I see it.

That was an attempt to segue to a generalization of the order issue to other instructions. Not well done.

But I believe there should be a principle of relaxing expressive constraints. Clarity of code is possible for the majority of users. A convention of leaving the parameters, unchanged from one invocation to the next, grouped together helps highlight what has changed, and to do that requires flexibility in specification order.

But back to universality. Should we not define these vsetvli parameters as "base"."qualifier" as with the mask qualifier? A multiplicity of short qualifiers becomes more difficult to remember and risk aliasing.
mu, ma, m8 cf. m< pre-processor variable > (or macro label, etc.)  are just too close for comfort.

msk.u, msk.a, mul.8  cf.  msk.< pre-processor variable > and mul.< pre-processor variable > disambiguate.

Of course, what I've done is improved clarity mostly by increasing the expressiveness of the parameter "base" name.
e.g. tail.a, tail.u, el.8, div.1 are clearer but more typing.
The real benefit in the "base"."qualifier" is for the macro/pre-processor case.

As with v0.t, the syntax is powerful to differentiate.

However, as with v0.t which could have specific contractions macro def vt v0.t
"Standard" defs can be established
def e8 ewid.8
def e16 ewid.16
def ta tail.a
def tu tail.u
etc.

I consider this much preferred than the current proliferation of acrimonious [sic] of 2 characters.

kasanovic, 2021-03-19:
I believe the current status quo has been accepted by default, so closing the issue.

closed 2021-03-19 by kasanovic