riscvarchive/riscv-v-spec #470: Describe interaction of vector instructions with big-endian memory model

kasanovic, 2020-05-13:
The current specification is silent on big-endian implications.

Vector code is not generally portable between big-endian and little-endian machines.
While this is also true of scalar code, vector code may more frequently rely on memory byte order.

David-Horner, 2020-05-13:
I would expect we would leverage the RVI settings for endianness to rely on memeory byte order.

The simplest approach is to load/store in big endian order into/from the elements of a given Element Width.
Once loaded into the "little-endian" in-register format it would look like the current register structure and operations.at that SEW, certainly up to XLEN..  
I expect we require  in-memory agreement of the component bytes,word, etc.
For this we need to resolve the in-register in-memory mismatch established by SLEN<VLEN horizontal interleave.(I think we could probably use the same mechanism as for little-endian.
But as I see it, other than Element widths above XLEN,  there is no other issues we need to address.

kasanovic, 2020-06-26:
Agreed, assuming SLEN=VLEN layout would resolve this issue.

kasanovic, 2020-06-28:
SLEN=VLEN decision resolves this issue.

closed 2020-06-28 by kasanovic