riscvarchive/riscv-v-spec #474: the source operand of vcompress

jovireal, 2020-05-15:
the spec writes : The vector mask register specifies by vs1 indicates which of the first vl elements of vector register group vs2 should be extracted. 

For other permutation insts, the whole VLMAX elements are used as source for permutation, while for vcompress, only the first vl elements are used. Is there any special reason for this restriction?

kasanovic, 2020-05-15:
The effect of vl=VLMAX on input can be achieved by using vsetvli x0, x0, .. form.

If it was always VLMAX, then would cause unnecessary execution on machines with long vector registers compressing short application vectors.  (Cannot rely on micro arch tricks on mask register having zeros past end point, as will want to use tail-agnostic when truly don't care about end mask values, and hence could have arbitrary values past end of application mask vector) This issue does not show up on vrgather or vslidedown.

closed 2020-05-15 by kasanovic

jovireal, 2020-05-15:
but i think it will better if all permutation insts follow the same rules.  It is easier for users to understand. especially when vrgather uses the whole vlmax elements and it does not set 0 when index>vl, but index>VLMAX.

kasanovic, 2020-05-15:
The vl value would be otherwise unused in vcompress, and as stated above, there is a serious performance issue with long vector machines if VLMAX was the only behavior.  The effect of vl=VLMAX can be easily obtained when needed.    These instructions (vcompress, vrgather, vslidedown) are all very different ways of moving elements around so it should not be surprising they have different ways of addressing their source.