riscvarchive/riscv-v-spec #478: use 1 vtypei bit to clear/preserve upper part of vtype

jnk0le, 2020-05-17:
Concrete proposal extracted from duscussion in #410 

It's about allocation of 1 bit in `vtypei` for selecting whether we want to clear or preserve upper part of `vtype`, where a future or custom fields will reside.

It will allow code to:
- Set the whole `vtype` with `li + vsetvl` and continue changing `sew`/ `lmul` by regular `vsetvli` without touching upper part.
- Clear the upper part on initial `vsetvli` and thus avoid prologues/epilogues 

We can also place that in `imm[11]` position in case if microarchitectures can somehow recycle constant sign extension.  `vsetvl`/`vsetvli` selector have to be moved somewhere else.

aswaterman, 2020-05-18:
It is a useful simplification that `vsetvl` and `vsetvli` currently overwrite all of `vl` and `vtype`.  In particular, the renaming and bypass logic for these registers can treat these registers as a unit, rather than having different rename state and bypass control for individual fields to handle the partial writes.  (Since these are latency-sensitive scalar instructions, the concern about partial writes is greater than for vector register writes.)

The downside of maintaining this simplification is that setting all of `vtype` will typically require three instructions (`lui` + `addi` + `vsetvl`). The silver lining is that the `lui` and `addi` are loop-invariant.

kasanovic, 2020-06-28:
Vtype is viewed as window into larger CSR, so vsetvli setting is decoupled from future/custom extensions that need extra bits.
The custom bits can be placed in a different. CSR,and both the custom and vtype could be mirrored in a new CSR to allow faster context save/restore.

closed 2020-06-28 by kasanovic

David-Horner, 2020-06-29:
Currently there is no larger CSR.into which  vtype is windowed.

I agree that custom extensions could be mapped into such a CSR; 
Applications using custom extension could reference it.
However, obligating the operating system to be aware of such an extension and context switch a hypothetical/custom specific  CSR is problematic.
It causes a significant impediment to incremental enhancement.

There are substantial free bits in vtype (19+ in RVV32 and 51+ in RVV64), sufficient that custom and reserved space can be carved out without a foreseeable need for another CSR in which to window. Indeed, the new CSR will be of the same size as vtype and thus provides no benefit to an aggregated standard vtype and custom bit.

If this windowed CSR is to be provided, it needs to be in place for V1.0 and not a future option.
OS code written for V1.0 should support custom extensions (which are by nature "future").  
As a result operating systems will always context switch the "window CSR" that contains the full state.
Thus, I see no need for such a windowing CSR and no value over managing a single vtype that has custom bits designated.

I believe this issue should be reopened until either a windowing CSR is agreed upon and included, or a method such as proposed below is. 

Proposal:
1) designate custom bits in vtype..

2) vsetvli will clear custom bits.
   
3) vsetvl will update any implemented custom bits one-to-one with the corresponding value in vs2.
      This is a constraint on the custom implementation to retain compliance.
      An implementation that does not support a given custom bit will set vill if vsetvl attempts to set it to 1

An example custom implementation:
   A custom vsetvli2 and vsetcl2 (in a custom op-code mapping, any name can be used) would provide the window into vtype. vsetvli2 would set custom bits from the immediate field with defaults or unchanged behaviour in the standard bits that are not mapped to the immediate field. A symmetric layout of all the bits to match vsetvli2 encoding is possible (but not required) for the vsetvl2 instruction.

Of course, any custom instruction could modify any of the vtype bits, but to be complaint vill must be set if unsupported (including the custom bits) combinations of bits is attempted.