riscvarchive/riscv-v-spec #479: Clarify Vector Integer Extension source EMUL constraint

JamesKenneyImperas, 2020-05-17:
For Vector Integer Extension instructions, the specification states:

`If the source EEW is not a supported width or the source EMUL is not a supported LMUL, an illegal instruction exception is raised.`

I can understand the source EEW constraint, but can you please clarify why the source EMUL constraint is necessary? It means (for example) that these instructions can never be used when LMUL=1/8.

David-Horner, 2020-05-18:
Thank you very much for expressing this concern.

> For Vector Integer Extension instructions, the specification states:
>
> |If the source EEW is not a supported width or the source EMUL is not
> a supported LMUL, an illegal instruction exception is [...]
You are correct this constraint is bogus.
>
> It means (for example) that these instructions can never be used when 
> LMUL=1/8.
The typical case would be that these instructions would not be used when 
LMUL=1/8.
Rather the 1/8 LMUL would typically be used to prep the smaller EW 
arguements for expansion.
Then the desired EW at a higher LMUL would be set and the expand (by 2,4 
or 8) would happen in the higher LMUL.

However, the constraint is not EMUL (effective LMUL), that is irrelevant.
I believe the intended constraint is that vl * SEW does not exceed LMUL 
* VLEN.
However, that is automatically satisfied by vsetvl[i], so there is no 
further test on EMUL required.
It just happens that in many of the use cases the calculated EMUL is 
valid, but it is not required.

Residual aspects of the previous LMUL constraints would induce the 
expectation that EMUL would have to be valid.
But with the v0.9 horizontal interleave model, LMUL no longer reflects a 
vertical structural aspect and so is irrelevant in the current situation.

This could change if, as I have proposed, fractional LMUL is associated 
with an interleave pattern.
If that pattern were determined by each fractional LMUL level, EMUL 
would be relevant.
However, Since my actual proposal determines that pattern independently 
of LMUL (via CLSTR but need only be relevant for fractional LMUL) the 
problem does not arise in my full proposal either.

JamesKenneyImperas, 2020-05-18:
To clarify this further for others who may be confused, LMUL and SEW are constrained by the (somewhat mind-bending) statement:

`Implementations must support fractional LMUL settings for LMUL = SEW/ELEN, for the ELEN value at LMUL=1. An attempt to set an unsupported SEW and LMUL configuration sets the vill bit in vtype.`

For an example implementation with ELEN=64, I think this means that these settings would be legal:

```
        vsetvli     t0, t0, e8,mf8      // LMUL=1/8, SEW/ELEN=1/8
        vsetvli     t0, t0, e8,mf4      // LMUL=1/4, SEW/ELEN=1/8
        vsetvli     t0, t0, e8,mf2      // LMUL=1/2, SEW/ELEN=1/8
        vsetvli     t0, t0, e16,mf4     // LMUL=1/4, SEW/ELEN=1/4
        vsetvli     t0, t0, e16,mf2     // LMUL=1/2, SEW/ELEN=1/4
        vsetvli     t0, t0, e32,mf2     // LMUL=1/2, SEW/ELEN=1/2
```
Whereas these settings would be illegal, and set vtype.vill=1:

```
        vsetvli     t0, t0, e16,mf8     // LMUL=1/8, SEW/ELEN=1/4
        vsetvli     t0, t0, e32,mf4     // LMUL=1/4, SEW/ELEN=1/2
        vsetvli     t0, t0, e32,mf8     // LMUL=1/8, SEW/ELEN=1/2
        vsetvli     t0, t0, e64,mf2     // LMUL=1/2, SEW/ELEN=1
        vsetvli     t0, t0, e64,mf4     // LMUL=1/4, SEW/ELEN=1
        vsetvli     t0, t0, e64,mf8     // LMUL=1/8, SEW/ELEN=1
```
In the above implementation, when SEW=64, the minimum legal LMUL is 1, meaning that source EEW of 1/2, 1/4 and 1/8 are all legal.

Similarly, when SEW=32, the minimum legal LMUL is 1/2. vzext.vf8/vsext.vf8 would violate the source EEW constraint and are therefore illegal. vzext.vf4/vsext.vf4 have source EMUL of 1/2 * 1/4 = 1/8, which is legal; vzext.vf2/vsext.vf2 have source EMUL of 1/2 * 1/2 = 1/4, which is legal.

When SEW=16, the minimum legal LMUL is 1/4. vzext.vf8/vsext.vf8 and vzext.vf4/vsext.vf4 would violate the source EEW constraint and are therefore illegal. vzext.vf2/vsext.vf2 have source EMUL of 1/4 * 1/2 = 1/8, which is legal.

Assuming that the EMUL constraint is indeed redundant, I think it might be better to remove it, and leave just the EEW constraint.

David-Horner, 2020-05-18:
On 2020-05-18 5:44 a.m., JamesKenneyImperas wrote:
> Assuming that the EMUL constraint is indeed redundant, I think it 
> might be better to remove it, and leave just the EEW constraint.
Of course, I agree with you.
As you have pointed out it is an invalid requirement .
EMUL is only incidentally applicable, and in this sense redundant.

David-Horner, 2020-05-18:
On 2020-05-18 5:44 a.m., JamesKenneyImperas wrote:

I read that as two distinct ideas.
> |Implementations must support fractional LMUL settings for LMUL = 
> SEW/ELEN, for the ELEN value at LMUL=1.|

|However, additional fractional LMUL settings are allowed.|

> |An attempt to set an unsupported SEW and LMUL configuration sets the 
> vill bit in vtype.|

This is a statement that unsupported fractional LMUL will set the vill bit.

I have thoughts on the reasonableness of even this, and whether other 
bounds are possible and preferred, but still mulling them.

However, I don't agree that the allowed LMUL SEW combinations for 
vsetvl[i] are directly relevant to the EMUL calculations for integer 
expand or for other mixed EW instructions.

I do continue to agree that integer expand instruction should not be 
restricted by the provided EMUL calculation and that I continue to 
assert no such calculation is necessary or valuable. I don't even see it 
meaningful in the current design.

<blather>
But I've been told "reasonable people can hold reasonable contrary 
ideas" (or words to that effect).
I believe, that most often, there is an underlying unifying idea that 
can be embraced by all reasonable people that drastically reduces the 
contrary ideas those reasonable people will hold.
However, I don't consider "flat earther"s reasonable when it comes to 
that subject, even though they might be very reasonable otherwise.
</blather>

solomatnikov, 2020-05-18:
> Whereas these settings would be illegal, and set vtype.vill=1:
>
> ```
>         vsetvli     t0, t0, e16,mf8     // LMUL=1/8, SEW/ELEN=1/4
>         vsetvli     t0, t0, e32,mf4     // LMUL=1/4, [...]

If `VLEN==SEW==64`, then setting `LMUL` to 1/2, 1/4 or 1/8 doesn't make sense.

David-Horner, 2020-05-18:
> Whereas these settings would be illegal, and set vtype.vill=1:
>
>         vsetvli     t0, t0, e16,mf8     // LMUL=1/8, SEW/ELEN=1/4
>         vsetvli     t0, t0, e32,mf4     // LMUL=1/4, [...]
Conversely when vlen equals 128 256 and 512 then 1/2 1/4 & 1/8 do make
sense.
The original limitation doesn't mention VLEN.

solomatnikov, 2020-05-18:
> > > In the above implementation, when SEW=64, the minimum legal LMUL is 1,
 meaning that source EEW of 1/2, 1/4 and 1/8 are all legal.
> >
> > If VLEN==SEW==64, then setting LMUL to 1/2, 1/4 or 1/8 doesn't make sense.
>
> Conversely when vlen equals 128 256 and 512 then 1/2 1/4 & 1/8 do make sense. The original limitation doesn't mention VLEN.
> The original limitation doesn't mention VLEN.

For SW portability reasons it makes sense to make these constraints conservative, i.e. to make sure SW works independently of `VLEN`.

@kasanovic it would be good to add a note explaining reasoning behind these constraints.

aswaterman, 2020-05-23:
I would support turning the following sentence into a bijection, forbidding implementations to be more permissive about which SEWs they allow for a given fractional LMUL: "Implementations must support fractional LMUL settings for LMUL ≥ SEW/ELEN, for the ELEN value at LMUL=1"

JamesKenneyImperas, 2020-05-23:
I agree with Andrew that this statement is more accurate. Assuming my concrete examples in the third comment here are correct, I think it would help general understanding to add them to the specification to clarify what the sentence means.

kasanovic, 2020-06-28:
Resolved in task group.

commit 7088c9e7 (2020-10-25): Cleaned up wording of constraints on fractional LMUL settings.
Closes #479.
closed 2020-10-25 by kasanovic