riscvarchive/riscv-v-spec #481: Note on MLEN=1 (reduced functionality compared to 0.8)

JamesKenneyImperas, 2020-05-18:
(This isn't really an issue, but a supplemental note for people like me who are consumers of this specification rather than contributors to it. Please let me know if there is a better place for this kind of thing.)

Be aware that the MLEN=1 change in the 0.9 specification, although simpler, loses some functionality compared to 0.8 when LMUL<=1. Before this change, Section 5.3 (Vector Masking) said this:

_The destination vector register group for a masked vector instruction can only overlap the source mask register (v0) when LMUL<=1. Otherwise, an illegal instruction exception is raised._

Now, it says this:

_The destination vector register group for a masked vector instruction cannot overlap the source mask register (v0), unless the destination vector register is being written with a mask value (e.g., comparisons) or the scalar result of a reduction. Otherwise, an illegal instruction exception is raised_

In practice, this means that some masked instructions that target v0 which were legal in 0.8 are illegal in 0.9. For example, `vadd.vv v0, v1, v2, v0.m` is now always illegal; previously, it was legal for LMUL=1.

JamesKenneyImperas, 2020-05-19:
On reflection, perhaps it would be most useful to explicitly state this restriction as a Note in section 5.3 - then this issue is redundant and can be closed.

kasanovic, 2020-07-20:
The normative text you quote states this restriction.  What would the content of the additional note be?

JamesKenneyImperas, 2020-07-20:
I am primarily thinking here of users who migrate from 0.8 to 1.0 implementations: that process is easier if incompatible functionality _reductions_ are explicitly stated. You are correct that the normative text clearly states the new behavior, but I think it would help to highlight it. The note itself could be just the text I wrote:

NOTE: this means that some masked instructions that target v0 which were legal in 0.8 are illegal in 1.0. For example, `vadd.vv v0, v1, v2, v0.m` is now always illegal; previously, it was legal for `LMUL=1`.

commit 29f06412 (2020-07-21): Note regarding 0.8->1.0 change.
Closes #481.
closed 2020-07-21 by kasanovic