riscvarchive/riscv-v-spec #485: Add information on changes between versions to Introduction

JamesKenneyImperas, 2020-05-21:
Many of the other RISC-V manuals contain information in the introduction about changes between versions. I think it would be very useful to do this or the Vector Extension Specification now, particularly since there is hardware under development using version 0.8, and some of the users of that will certainly want to migrate to 0.9 (or a later version) at some point. My first stab at this is:

- mstatus.VS and sstatus.VS fields moved to bits 10:9;
- new CSR vcsr added and fields VXSAT and VXRM relocated there from CSR fcsr;
- vfcvt.rtz.xu.f.v, vfcvt.rtz.x.f.v, vfwcvt.rtz.xu.f.v, vfwcvt.rtz.x.f.v, vfncvt.rtz.xu.f.v, vfncvt.rtz.x.f.v, vzext.vf2, vsext.vf2, vzext.vf4, vsext.vf4, vzext.vf8 and vsext.vf8 instructions added;
- fractional LMUL support added, controlled by an extended vtype.vlmul CSR field;
- vector tail agnostic and vector mask agnostic fields added to the vtype CSR;
- all vector load/store instructions replaced with new instructions that explicitly encode EEW of data or index;
- whole register load and store operation encodings changed;
- VFUNARY0 and VFUNARY1 encodings changed;
- vfredsum implementation details have changed;
- MLEN is always 1, which changes the operation of all masked instructions and makes some classes of instruction that were legal under 0.8 illegal under 0.9.

Has this missed anything significant?

David-Horner, 2020-05-21:
This is an excellent suggestion.
 It would be very helpful to include the issue or patch reference numbers.
Another major change is SLEN to apply horizontally rather than previous
vertical striping.

JamesKenneyImperas, 2020-05-21:
(I've edited the original message to include David's point to keep everything in one place.)

kasanovic, 2020-07-03:
Trying to do this for larger changes, but the GitHub record is the ultimate authority and it's hard/error-prone work to replicate this in text.

closed 2020-07-03 by kasanovic