riscvarchive/riscv-v-spec #488: the overlap constraint of segment load store

jovireal, 2020-05-28:
In the spec, the destination register overlap constraint is only applied to indexed segment loads, but  no constraint for other segment load/store. But it seems not proper for v0 used as both mask register and dest register for segment load/stores as it affects the result of multiple dest registers?

David-Horner, 2020-07-15:
I see this as only an issue for loads as stores do not have destination registers.
Section 5.3 does what you want :

> The destination vector register group for a masked vector instruction cannot overlap the source mask register
> (v0), unless the destination vector register is being written with a mask value (e.g., comparisons) or the scalar
> result of a reduction. Otherwise, an illegal instruction exception is raised.

However, there is a potential restart issue with both indexed load types if the index vector is part of the destination register (or register set)..

aswaterman, 2020-07-16:
I don't think we need to take any action here.  As David points out, we already have text disallowing overlap with v0.

The potential restart issues David mentions are already handled.  We have text disallowing overlap entirely for indexed segment loads.  For regular indexed loads, overlap is allowed in some circumstances (following the usual rules in https://github.com/riscv/riscv-v-spec/blob/c427c1033414e43b8c435a4295f3beabf48dae2e/v-spec.adoc#52-vector-operands), but that's not a problem for restartability, since the elements that were overwritten correspond to indices already processed, so they aren't necessary for restartability.

closed 2020-12-18 by kasanovic