riscvarchive/riscv-v-spec #489: why riscv-v-spec-0.9.html just list vle32 and vle64 instructions

luxufan, 2020-05-28:
in the riscv-v-spec-0.9.html, it just lists vle32.v and vle64.v instructions
[image]
is there has some instructions like vle8.v vle128.v ...?

jim-wilson, 2020-05-28:
If you look at section 7.2, you will see it is vle<EEW> and vse<EEW>.  These are just two examples of the instructions that exist.  The actual set of instructions depends on what the valid values are for EEW on the target.  For unit stride instructions, EEW is SEW, and values are defined for SEW up to 1028.  But listing vle1028.v would be potentially misleading, as most implementations probably won't support that.  An implementation is only required to implement loads and stores up to ELEN.

kasanovic, 2020-06-28:
text can be expanded.

kasanovic, 2020-07-20:
Fixed in update to 1.0-draft.

closed 2020-07-20 by kasanovic