riscvarchive/riscv-v-spec #492: Is it possible to add more load/store whole register instructions?

monkchiang, 2020-06-04:
The RVV-0.9 spec has VL1R.V and VS1R.V to access memory.
But it’s only useful, when LMUL is one.
When LMUL is more than 1, the compiler needs to use N-times load/store instructions to access instructions.  Also, the compiler needs to reserve 2 general-purpose registers to calculate address. So, the compiler generates several ADD instructions to calculate address.

If  there are instructions to deal with the LMUL 2, 4, 8.
And assembly like this:
```
  vl (2 | 4 | 8) r	 v0, (a0)
  vs (2 | 4 | 8) r	 v0, (a0)
```
But without these instructions, I need to use vl1r and vs1r  instructions to expand it. For example, a load instruction with LMUL is 4.
The assemble like this:
```
   csrr tmp1, vlenb
   vl1r v0 (a0)
   add tmp2, tmp1, a0
   vl1r v1, (tmp2)
   add tmp2, tmp1, a0
   vl1r v2, (tmp2)
   add tmp2, tmp1, a0
   vl1r v3, (tmp2)
```

To make the program better, I suggest adding more load and store instructions for LMUL is 2, 4, 8.

Rolfy360, 2020-06-04:
Maybe introduce new lumop/sumop for embedded EMUL in the encoding, like the embedded EEW?

kito-cheng, 2020-06-23:
FYI, there is whole register load/store with 1+ vector register exist in the spec for a week and than removed:
https://github.com/riscv/riscv-v-spec/commit/49cbd95cce7907f5a74b3709bf1801a162a43619

kito-cheng, 2020-06-23:
From the compiler implementation view, I would like to have whole register load/store with 2, 4, 8 registers to simplify the spilling code gen too.

kasanovic, 2020-06-28:
The task group debated this.  I view this as straightforward and useful addition given that microarchitectures must support the underlying operations in any case, as these are unit-stride vector register group load/stores except with fixed vl/emul in instruction encoding instead of variable vl/emul from vl/vtype CSR.

Some people are opposed.

David-Horner, 2020-06-29:
@kasanovic from #449 
> Spec will have blanket statement that behavior of the unsupported width/operation combinations is "reserved" to allow future expansion.

The <nf> also has such language, perhaps it will be added to the blanket statement as well.

If multi whole register (<nf> greater than 1) load/stores are included post V1.0, adoption will occur for performance reasons (a strong motivator) assuming the inclusion was for this reason. However, whatever the reason the new facility will be incorporated wherever it provides value as there is little impediment to software inclusion. 

The downside is hardware space fragmentation. If multi whole register load/stores provide a substantial win, early hardware implementers will be penalized over V1.0+ newcomers that can easily include it.

We can consider making <nf> optional in V1.0 for whole register  load/stores.
-Some implementers will make it hardware selectable.
-Some embedded will be <nf>=0 only as they will only run such code 
-Some will be <nf>=0  and emulate. With run time patching they can replace with a sequence of whole register (either in place or millicode) 
This flexibility itself has cost in verification  and software eco system support.

Note however, that if <nf> greater than zero is supported post V1.0 the "optional" cost is not avoided, only delayed.

kasanovic, 2020-08-07:
Someone has to make a stronger case for leaving these out - they don't seem to add much in implementation/verification cost, and are useful to software.

commit 6dc34f62 (2020-10-26): Clarified whole register load/store.
Allowed implementations to report misaligned for smallest supported EEW.
Closes #492.
closed 2020-10-26 by kasanovic