riscvarchive/riscv-v-spec #493: unbind vstart from element index

jnk0le, 2020-06-06:
As `vstart` is not intended to be used by regular code. I propose to unbind prestart condition from (memory) element indexing so the microarchitectures can read/execute/write elements in preferred order.

Currently `vstart` holds the element index at which the execution can be resumed. This most probably limits the granurality at which interrupts can be taken, as the execution units might not map to striping pattern easily. In 0.8 there was a concern about spatial datapath width larger than SLEN. 0.9 now scatters consecutive elements across all sections when SLEN<VLEN.

`vstart` will now hold an architecture dependent (currently it can raise illegal exceptions in the same manner), or somehow standardized prestart condition that can be imprecise within a single instruction.
It can also be further extended for handling simpler imprecise exceptions.

Some overlap constraints will be messed up again though.

kasanovic, 2020-06-28:
We are cleaning up exception handling text.  Thismight beismpler now. with SLEN=VLEN decision.

David-Horner, 2020-07-02:
I agree that vstart should be considered opaque by the application.
Other than on "Bare Metal", an Execution Environment will have to 
request through an EEI to reliably determine meaningful information from 
vstart.
An EE can establish that a write of data read from vstart will 
re-establish the prior state of vstart.
Platforms may require this behaviour.

On simple machines there could be a mapping

jnk0le, 2020-07-23:
One more usefull thing could be traps taken within elements in case of SEW wider than datapath (ediv/crypto/etc.)

kasanovic, 2020-10-26:
Decided not to do this, as it complicates OS/debugger interface to determine fault cause.

closed 2020-10-26 by kasanovic