riscvarchive/riscv-v-spec #496: Is there an I in mnemonics of Vector Indexed Instruction?

Rolfy360, 2020-06-12:
In table 7 and table 8,
index load and store and described as VLXEI<EEW> and VSXEI<EEW>,
so are example code in 7.6. Vector Indexed Instructions.
But in old ver. 0.8, ver. 0.9 and vmem-format.adoc,
they are "VLX* indexed" and "VSX*" indexed
The I is very abrupt comparing to other 2 types 
VLE8/VLSE8/VLXEI8
VLE8/VLSE8/VLXE8

Is it a typo?

knightsifive, 2020-06-13:
I'm not sure I understand your question, or what you mean by "abrupt". I believe there is intended to be an `i` in the instruction mnemonics for indexed loads and stores. The `8` in `vlxei8.v`, indicates that the _index_ EEW (effective element width) equals 8, whereas the _data_ EEW equals the SEW (standard element width). Please contrast this to `vle8.v`, where the data EEW equals 8 and there is no "index EEW" in this context.

I think you spotted a typo in one of the examples in Section 7.8.3; I went ahead and submitted PR https://github.com/riscv/riscv-v-spec/pull/497.

I agree that the "opcodes" in the table in Section 7.3 are confusing since they do not match the mnemonics for indexed loads and stores. I'm hoping someone else can step in and suggest a fix for this.

Rolfy360, 2020-06-13:
Thanks for the reply.
I think I misunderstood the words and thought all load/store use EEW as data width.
Since vector indexed instructions use SEW as data width and use EEW as index width, "ei" in mnemonics is a better choice for differentiation.

closed 2020-06-13 by Rolfy360