riscvarchive/riscv-v-spec #499: incompatibility between ELEN==64 and ELEN==32 ?

solomatnikov, 2020-06-16:
According to the discussion in https://github.com/riscv/riscv-v-spec/issues/479 and https://github.com/riscv/riscv-v-spec/blob/17d7293f5fa81076fe3cc8f0b46419971ecc5c5e/v-spec.adoc#332-vector-register-grouping-vlmul20 `mf8` is illegal if `ELEN==32`.

This means that the code written for `SEW==32` or smaller for HW with `ELEN==64` can use `mf8` but it cannot be executed on HW with `ELEN=32`.

One solution is to mandate `VLEN >= 64` for implementations with `ELEN==32`.

Another solution is make `mf8` behave as `mf4` for implementations with `ELEN==32`.

aswaterman, 2020-06-16:
I don't think fractional LMUL is a good reason to mandate a new constraint on VLEN.

The spec already says: "Implementations must support fractional LMUL settings for LMUL ≥ SEW/ELEN, for the ELEN value at LMUL=1"

I propose we turn that statement into a bijection: "Implementations must support fractional LMUL settings for, and only for, LMUL ≥ SEW/ELEN, for the ELEN value at LMUL=1"

solomatnikov, 2020-06-16:
This code:
```
vsetvli t0, t0, e8, mf8, tu, mu
vle8.v  v1, (t1)
vadd.vi v2, v1, 1
vse8.v  v2, (t2)
```
is legal for an implementation with `ELEN==64`.

I think most people would expect this to work on implementations with `ELEN==32`.

zakk0610, 2020-06-18:
I think people would not expect that, for example, `setvli t0, e32, m1`  with widening instructions are supported in ELEN==64, but they can not work on ELEN==32 platform.

so it's what ELEN should be known in compiler time https://github.com/riscv/riscv-v-spec/issues/302#issuecomment-540621841 and toolchain would encode ELEN info https://github.com/riscv/riscv-v-spec/issues/360 in object to avoid ELEN inconsistent problem.

ebahapo, 2020-06-18:
Since ELEN≥`max(XLEN, FLEN)`, perhaps it suffices to record the ABI for compatibility.  Specifically, ELEN≥32 for rv32f and ELEN≥64 for rv{32,64}d and rv64f.

solomatnikov, 2020-06-18:
> I think people would not expect that, for example, `setvli t0, e32, m1` with widening instructions are supported in ELEN==64, but they can not work on ELEN==32 platform.

Yes, if you set SEW to `e32` and then do widening, it can only work on `ELEN==64` HW, regardless of fractional LMUL.

However, in my example SEW is set to `e8` and one would expect it to work on `ELEN==32` HW with or without widening.

solomatnikov, 2020-06-18:
Another option is to get rid of `mf8`. It's not clear why `mf8` is needed other than symmetry with `m8`.

jnk0le, 2020-06-19:
> 
> 
> It's not clear why `mf8` is needed other than symmetry with `m8`.

extending 8bit-> 64bit elements while keepeng at `m1`

commit 7529c273 (2020-06-22): Define behavior of LMUL < SEW/ELEN to be reserved.
Closes #499.
closed 2020-06-22 by kasanovic