riscvarchive/riscv-v-spec #501: Unordered Indexed Load

billhuffman, 2020-06-26:
The brief discussion of memory ordering this morning leads me to think that we need indexed loads that are unordered just like we have indexed stores that are unordered.  It doesn't matter for a single hart, but memory ordering means it matters for multiple harts.  Otherwise, unless the memory being accessed is private to a single hart, the indexed loads can't be done out of order.

solomatnikov, 2020-06-30:
I think unordered indexed/strided loads are a good way to relax memory consistency model further for high performance implementations.

@daniellustig

David-Horner, 2020-07-09:
Depending upon the application of the RVWMO model, whether elements n 
and n+1 are considered independent when they have no syntactic data, 
address or control dependency between them, from the perspective of 
other harts, the loads can be reordered and hence, can be reordered 
explicitly in the executing hart as long as the locally perceived 
program order (defined by the syntactic dependencies) is maintained.

I have been trying to get a worthwhile first thoughts on this to 
present. Also, considering the possibility of further relaxing the model 
for vector to vector operations from syntactic to allow certain classes 
of "dynamic actual" control and data( and possibly address) dependencies.

I believe it will be better to constrain initial implementations, extend 
the memory model for RVVWMO and only if later needed, claw it back and 
introduce explicit relaxation.
This allows for relaxing the actual implementations and meeting in the 
middle with a solid physical implementation and an implementable, 
efficient, effective and verifiable memory model.

On 2020-06-26 1:40 p.m., billhuffman wrote:
> The brief discussion of memory ordering this morning leads me to think
> that we need indexed loads that are unordered just like we have
> indexed stores that are unordered. It doesn't matter for a [...]

On 2020-06-30 11:51 a.m., Alex Solomatnikov wrote:
>
> I think unordered indexed/strided loads are a good way to relax memory 
> consistency model further for high performance implementations.
>
>

billhuffman, 2020-07-09:
It seems like the rule that vector memory ordering is the same as a scalar loop would be, plus Rule #2 in the memory ordering section (Coherence for Read-Read pairs) requires indexed loads to order accesses to the same address in element order.

David-Horner, 2020-07-10:
>
> It seems like the rule that vector memory ordering is the same as a 
> scalar loop would be, plus Rule #2 
> <https://github.com/riscv/riscv-v-spec/issues/2> in the memory 
> ordering section (Coherence for Read-Read pairs) requires indexed 
> loads to order accesses to the same address in element order.

That would be
*• Overlapping-Address Orderings:*
1. b is a store, and a and b access overlapping memory addresses
*2. a and b are loads, x is a byte read by both a and b, there is no 
store to x between a**
**and b in program order, and a and b return values for x written by 
different memory**
**operations*

So, yes if the addresses overlap. But no if they do not.

In  that  the usual case involves only the index values that are 
multiples of the eew,
a) the addresses will not overlap and we can read them in ascending 
order (or any other that is performant),
or b) those that are the same we can read immediately after one another
(which in effect means in either order as an additional  physical read 
will not occur in the tRVWMO domain of "regular main memory".
For indexes that are not multiples of eew the fall back to in order will 
not be wrong, but even there some of the reads can likely be optimized 
that don't overlap.

Depending upon how concurrence is defined in vector ops (what it really 
means to loop over the elements in order), they may be further 
relaxation of "global memory order".
Note as an equivalence, effectively vector operations (and load in 
particular) are unrolled scalar loops with a massive number of available 
registers.
(in general less constrained than renamed scalar registers in a 
massively ooo implementation).

What especially complicates load and concurrence is restart and the 
definition of vrestart as a transparent index for resumption (vs an 
opaque parameterized directive to resume).

billhuffman, 2020-07-10:
Since the index vector contains byte offsets, any sort of overlap is possible.  I expect the most common overlap will be identical loads.  It is those identical loads or other overlaps that I think are the whole subject of discussion.

David-Horner, 2020-07-10:
>
> Since the index vector contains byte offsets, any sort of overlap is 
> possible. I expect the most common overlap will be identical loads. It 
> is those identical loads or other overlaps that I think are the whole 
> subject of discussion.

Agreed it is possible.

  It can be avoided by scaling index offset by EEW.
Perhaps this should be seriously considered as it also
  helps avoid misaligned addresses and
  helps avoid the shift of the index by log2(EEW) which I have 
frequently seen in code.
Programmers most conveniently work with indexes to elements rather than 
offsets in memory. .

Strided loads can be easily checked to see that abs(rs2) >= EEW before 
optimizing.

However, even for indexed a check that log2(EEW) low order bits are 
clear in each index element may not be overly onerous for an instruction 
that, when optimized for the typical case, can provide substantial 
performance benefit.

A pessimistic implementation that processes from 0 to vl can still 
optimize by passes through that load from present cache entries, locally 
assessing address overlap.

In other words, the benefits from and mechanisms used to out-of-order 
process loads can be seen and applied by RVVWMO ordering VLXEI and not 
require VLUXEI.

Further, what does unordered loads really mean from a concurrency 
management stance?
  What guarantees does it still give, or actually relax/remove/ignor?
In the most extreme case it could mean that reads are allowed from 
before or after a full fence.
Is that what is envisioned? Is that what could be implemented under the 
current loose definition?
Will we have to provide new fence formulations to provide a 
<unordered>vector to scalar to <unordered>vector  fence?
Will it be so heavy handed that the unordered loads will be effectively 
unusable?

aswaterman, 2020-07-10:
We should not change the definition of the instructions to scale by EEW,
because one use case is for the index vector to be a vector of pointers,
with a scalar base address of x0 (or a small offset).

billhuffman, 2020-07-10:
I agree with Andrew indexes need to be bytes.  And it would not solve the problem if they were scaled.  Overlap would still be there as identical indexes.  And if hardware finds identical indexes (or different but overlapping) in order for indexed loads, it might as well do so for indexed stores, which are less common in my experience.

This issue is considering whether the unordered concept to indexed stores.  There is another about unordered and exceptions and a third about yet looser ordering for indexed loads and stores.

David-Horner, 2020-07-10:
>
> I agree with Andrew indexes need to be bytes.
I agree with Andrew that pointer based indexes are an important use case 
and thus not to be eliminated.
The question I intended to pose is whether VLXEI<EEW> is the correct 
vehicle for that.
If indexes should rather be array indexes rather than memory offsets for 
this instruction (and related store) and a different formulation 
specifically deal with XLEN pointer addressing.

> And it would not solve the problem if they were scaled. Overlap would 
> still be there as identical indexes.
Identical indexes is not a problem in the RVWMO model. There is no 
intervening write from another hart to the identical index loads as they 
are allowed to occur concurrently in RVWMO absent data, address and 
control dependencies. Control and address dependencies can be and in 
most cases need to be resolved before loads can occur. There are no data 
dependencies per se as the operation is a load.
>
> And if hardware finds identical indexes (or different but overlapping) 
> in order for indexed loads,
I don't follow the connection to stores
>
> it might as well do so for indexed stores,
I have an issue open about clarifying index store behaviour and 
explicitly naming the ordered store as Ordered.

> which are less common in my experience.

> This issue is considering whether the unordered concept to indexed 
> stores.
As we have discovered, the unordered load and  RVWMO are intimately 
linked. A request for an Unordered load variant has to consider the two 
issues you mention.
>
> There is another about unordered and exceptions and a third about yet 
> looser ordering for indexed loads and stores.
What is the issue number for "unordered and exceptions" ?
The issue #504 raises the "yet looser ordering for indexed loads and 
stores", now hopefully github linked to this issue.

#504 appears to suggest the heavy weight fence to allow weaker than 
RVWMO memory access explicitly for indexed load/stores.
I agree that we should express RVV concurrency in a relaxed manner to 
allow for a RVV specific RVVWMO definition that allows for the 
parallelism requested there.
However, even within the current RVWMO model and a process model that 
stipulates "as if processed consecutively from 0 to vl" there is 
sufficient latitude to provide significant parallelism and respect the 
currently provided fencing instructions.

billhuffman, 2020-07-10:
I believe that identical indexes has exactly the same memory ordering issue that overlapping but not identical has.  A store to that location is very unlikely, but the fact that it can occur requires implementation constraints on how the overlapping or identical accesses happen.

Unordered and exceptions is issue #502.  Sorry I didn't include the issue numbers.

daniellustig, 2020-07-13:
I'm not fully up to date with the status of the vector spec, so please forgive me if I'm missing any context, but since @solomatnikov cc'ed me in I'll add my two cents (for #501 #502 #504):

I think the natural thing to do from the memory model POV would be to stick with what the vector spec draft already suggests, or to mimic what the RVWMO spec says for misaligned loads or stores being decomposed into multiple memory operations, e.g.:

"A vector load or store instruction is decomposed into a set of component memory operations.  These memory operations are ordered normally with respect to the memory operations generated by preceding and subsequent instructions in program order.  Unless the instruction is indexed-unordered, the program order on those memory operations also respects element order.  For indexed-unordered instructions, the memory operations are not ordered with respect to each other in program order.  The memory operations then follow the standard RVWMO rules."

If that's acceptable from an implementation perspective, then that small snippet should fully cover the memory model implications.  However, if you want to relax vector-to-scalar ordering like #504 proposes, for example, then you'll need to build a lot more new memory model machinery.

Allowing repeated writes as proposed in #502 seems like a bad idea to me from a theory perspective.

David-Horner, 2020-07-13:
>"A vector load or store instruction is decomposed into a set of component memory operations. 

>If that's acceptable from an implementation perspective, 

1) What RVWMO does not state is the order in which the components are to be process.
The implication is that the order is indeterminate and any physical order is allowed. 

2) Some RVV instructions require elements to be processed in order from 0 (or vstart if not zero) to vl (vector length). 
3) Indeed, for instruction restart, vstart is explicitly premised on this assumption.

3 .b) Notably, POR synchronous traps definition anticipates element set processing that is not strictly processed from 0 to vl.
       I hope to, instead of the "exception definition" of synchronous used  RVV, expand the concurrency process model which naturally allows the current trap allowance as synchronous with respect to it. I have proposed that this extend process model "fuses" consecutive RVV instructions further relaxing visible process architectural state. #364  

4) As with RVWMO's extensive analysis of RVI instruction dependencies, RVV instructions will also need to be profiled to ensure correct data, address and control dependencies are observed. Changes to vmul and vsew rearrange register dependencies. The related case of mixed width, narrowing and widening operations, introduce further complications to this mapping. Fortunately we have abandoned visible SLEN affects with an in-register matching in-memory approach, so things are much better than they would have been.

4.b) Most RVV instructions operate on elements in vertical sets (where their "indexes" and widths are equal.).
      This is a substantial change from RVWMO in which all registers are considered collectively for their dependency impacts.      
      Instead, RVV can consider that not only memory operations that do not overlap, but also , elements sets that do not overlap can generate dependency rules that even though syntactically defined allow greater relaxation than expressed in RVWMO.

5) Whereas the syntactic dependency tracing was sufficient for RVI, I believe RVV could benefit from "actual" execution dependency results. e.g. interaction of operation under mask with operation under complementary mask.  

I believe each of these is distinct. e.g. vstart need not be premised on process order from 0 to vl.

These are the components that I believe need to be addressed in a RVVWMO "extension" to RVWMO., issues that are unique or at least dominant in RVV.
However, it is only item 5 that I believe may need extensive new mechanisms in place to fully support, whereas, a subset (e.g. the example I mentioned) may be easily accommodated. In other words , incremental application may be possible.

The rest are, to my way of thinking, refinements of existing infrastructure. Leveraging RVWMO model, with judiciously defined process constraints, rather than adopting a radically different RVVWMO approach, is my goal.

billhuffman, 2020-07-13:
From what I can see, I agree with @daniellustig that the natural thing to do is to stick with the ordering already described.  I believe that all stride-1 and stride-N vector loads and stores actually have no RVWMO ordering requirements among the various elements themselves.  Indexed loads and stores, however, can have identical/overlapping accesses. Therefore, RVWMO rule #1 will apply to vector indexed stores and rule #2 will apply to vector indexed loads.  I believe that the remaining RVWMO rules do not apply to indexed loads and stores either.

We saw very early that vector indexed stores had an issue even for a single hart that meant having an unordered version was valuable.  At the single hart level, unordered indexed loads provide no value.  But when we consider memory ordering, I think we find that unordered versions have just as much value for vector indexed loads as they do for stores.  Thus issue #501.

As for @daniellustig's comment on #502, allowing repeated writes looks pretty ugly to me as well.  But the alternative seems to me to require translation and protection checks to occur in order for vector indexed loads and stores.  Probably further discussion should be under that issue.

The reason for issue #504 is, again, the difficulty of peering ahead for vector indexed operations.  Requiring other operations to be ordered with them is very restrictive and essentially forbids overlap with succeeding vector or scalar memory operations.  If there's not a standard way of allowing unordered indexed loads and stores to be out of order wrt succeeding instructions, there will certainly be a non-standard one.  Again, further discussion probably should be under that issue.

David-Horner, 2020-07-15:
https://github.com/riscv/riscv-v-spec/issues/501#issuecomment-656438224
>A store to that location is very unlikely, but the fact that it can occur 

I was not sufficiently specific on what environment I considered for my assertion that identical memory overlap means the reads need not be ordered.

RVWMO nicely sidesteps the hardware issue by stipulating order by memory events; even when the memory model is intended to address the “standard” memory hierarchy and specifically allows features of this model (notably a write buffer) it still obscures the hardware nature of “regular main memory operations” on which it depends and to which it explicitly states is limited.

I was assuming a power of 2 cache line greater than or equal to the element size, cache coherent, non-distributed, single-copy atomic physical memory.

This specification  , with a write buffer, is the current “standard” memory configuration.

I did a little cheat because RVWMO does not guarantee atomicity to unaligned elements and expresses uncertainty about mixed width simultaneous operations (specifically with writes).
It is the fringe cases of elements spanning cache lines that are problematic, but even then the problem for read is one of cache thrashing while attempting to acquire the lock on all required adjacent pages.
So, the requirement that “log2(EEW) low order bits are clear” provides a sufficient condition (although not a necessary one) to allow reordering of all such reads.

Although it is instructive to consider the overlapping cases, it is sufficient to apply RVWMO rules to  non-overlapping loads to demonstrate no intervening write can occur with this physical configuration.

First consider a set of element loads with only one non-overlapping memory reference each,
RVWMO allows these reads to be in any global memory order, and as there is no dependence (address, data or control) between load elements in a single RVV load instruction the local hart physical read order (for “main memory” without I/O related cache rules, e.g. serializing byte order on half/word/double address ports) is allowed to vary from the “as if processed from 0 to v” rule.

Consider then adding into a RVV load an additional element read from one of the previous locations. (judicious masking of a RVV load can create the first state and unmasking one of the duplicates create this extant case).
As before, all the “independent memory reads, including the first of the duplicate reads can be reorganized so that the duplicate read is at the end of that set and thus the new duplicate read will immediately follow. As the cache system will have  a read lock on the cache line, and now that two consecutive read to the same location would otherwise be presented to the cache subsystem, there need be no intervening write to the memory location as a single read from cache can suffice for both load requests.
QED?

As the non-overlapping load is a very common use case, I don’t see a need for an unordered load variant.

(I will describe what I consider necessary criteria for  non-overlap and the difficulty of overlapping memory reads later and perhaps elsewhere, not wanting to hijack this thread).

billhuffman, 2020-07-15:
@David-Horner: I think you're suggesting a fully associative cache that holds all values read by an indexed load but not yet fully in order.

I think it probably solves the memory ordering issue within the instruction but requires a lot of gates that are fundamentally unnecessary as the order of those loads will "never" really matter.  I expect actual high-performance implementations simply to violate the spec instead.  For the same reasons we have an unordered indexed store, it would be valuable to have an unordered indexed load with looser rules.

If we don't have an unordered indexed load, we are saying that memory ordering is a second class requirement - that we're willing to add an instruction for the final value issue (in indexed store) but not for memory ordering (in indexed load).

The fully associative cache also doesn't solve the additional difficulties discussed in Issue #504.  I expect implementations there simply to violate the spec.

These expected violations of the spec may be done under a mode bit.  But it would be far better to have ordered and unordered versions with loose enough order requirements on the unordered to work for high-performance implementations.

David-Horner, 2020-07-15:
No. I'm not suggesting a fully associative cache, nor one that holds all required values for the load.
The cache configuration I specified is sufficient and n-way associative is acceptable.
For the aligned non-overlapping of my "cheat", the cache does not have to simultaneously accommodate all element memory locations, indeed only one cache line of the memory working set needs to be resident at a time. 
Although I did not present a justification for this, for the not aligned but none overlapping duplicates allowed case, only the two adjacent cache lines of a split element need to be simultaneously present in the cache.

Nor am I suggesting in this thread that we relax the "process from 0 to vl" requirement. (I make that suggestion elsewhere.) 

Rather I am saying:
Even when the processing is from 0 to vl, the RVWMO memory model allows the resultant values read to be different than what may appear as the global memory order as seen by other harts.
As a result, in practice an implementation can, for non-overlapping accesses with duplicates avoid a second read for the duplicates and allow the cache subsystem to provide "stale" data. 
Multiple concurrent requests can be submitted to the cache subsystem.
However, ordering the index values to provide ordered addresses ( thus allowing a simple cache system to optimize multiple requests to a given cache latch), essentially out of 0 to vl order is perfectly legal and not a relaxation of the 0 to vl processing order per se.

billhuffman, 2020-07-15:
I don't understand how one cache line is sufficient.  Suppose an implementation accesses 128 elements in any order before going on to further elements.  Suppose that the last one looked at is element 7.  The other 127 elements are different from each other but element 7 happens to match one of the later elements.  There are 120 later elements that it could match.  Element 7 could match any one of them, and whichever one it matches, I need to have kept that value so that element 7 does not get a newer value than an identical, higher numbered element.  It also seems to mean 128^2/2 comparitors to find matching elements.  And that's before accounting for overlapping elements, which would need to be detected as well, even if they were handled by starting over in order.

David-Horner, 2020-07-15:
>
> I don't understand how one cache line is sufficient.
to be resident at a time....
>
> Suppose an implementation accesses 128 elements in any order before 
> going on to further elements.
>
> Suppose that the last one looked at is element 7.
of the initial 128.
>
> The other 127 elements are different from each other but element 7 
> happens to match one of the later elements.
(OK. only one duplicate in the first 128)
>
> There are 120 later elements that it could match. Element 7 could 
> match any one of them, and whichever one it matches, I need to have 
> kept that value so that element 7 does not get a newer value than an 
> identical, higher numbered element.
That is not a requirement. RVWMO allows the simplification of providing 
the later element load request to be filled by the previous load, but it 
does not require the two values to be the same. Some may want the RVV 
model to ensure matching values for matching addresses, but as currently 
formulated we do not guarantee that outcome.

For example a load of stride zero to a strongly ordered non-idempotent 
I/O location (potentially with other read side-effects) may be required 
to return differing values each read and must also be read the vl times.

Back to the "main memory" case.
As RVV allows load and store operations to be interrupted and restarted 
there can be no reasonable expectation of matching values for matching 
addresses. (consider that a task context switch has occurred).

Performance aside, I hope it is clear that a single resident cache line 
is sufficient for the non-overlap duplicate case..

So, to be clear, that "consistency" is not what I am advocating.
Rather that we execute the RVV load as (if) a set of vl scalar loads 
presented to the ooo superscalar's instruction decoder in 0 to vl order 
and apply the RVWO rules.
The overhead in RVV for non-overlapping memory locations is much less 
than it would require for this hypothetical superscalar ooo and still 
provide comparable performance (with a comparable cache subsystem) if 
the hart were to first check in-cache lines and order its requests.

> It also seems to mean 128^2/2 comparitors to find matching elements. 
> And that's before accounting for overlapping elements, which would 
> need to be detected as well, even if they were handled by starting 
> over in order.

Agreed, overlapping elements are more challenging. I'm working on that.

Given that the typical use case will be no or identical overlap, there 
is a clear win for RVV loads with RVWMO.
Even with some non-identical overlap we could see substantial benefits 
to accommodate subsets of those cases as well.
Even the pathological case has no need for comparitors; in order 
processing suffices.

billhuffman, 2020-07-15:
I wasn't asking for the same values.  That was just to keep from doing the accesses in the wrong order and getting a newer value in the smaller index access and an older value in the larger index access.

I don't seem to understand what you're suggesting at all.  Could you explain what happens when there's a low index that matches a high index but they get accessed in the reverse order?  And the hardware doesn't know what will match until late in the game?

David-Horner, 2020-07-19:
> I wasn't asking for the same values. That was just to keep from doing the accesses in the wrong order and getting a newer value in the smaller index access and an older value in the larger index access.
>
> I don't seem to understand what you're suggesting at all.

> Could you explain what happens when there's a low index that matches a high index but they get accessed in the reverse order?

Certainly.

I will assume aligned and same size access to simplify the narrative.

 Up to now I have described what RVWMO allows and therefore what legal relaxations are possible.

This question moves to the realm of what happens when an implementation potentially violates RVWMO rules by a certain behaviour.

As a result, "what happens" needs further explanation than either

    1) a simplistic detailing of the behaviour of differing cache subsystems or

    2) a reduction to a statement of correctness.

What happens is dependent upon details not present in the question.

Especially important is the reason for the reads being accessed in the reverse order.

A) If the cache subsystem consolidates concurrent identical reads, then the right thing happens even if the subsystem does its reads in reverse order. Both (all) elements can be updated with the identical results provided by the cache.

Note: the read scheduling process can instead consolidate pending identical address reads thus avoiding unnecessary cache read requests and the complication of reverse order reads.

B) If the cache subsystem does not consolidate concurrent identical read requests and does not maintain order of read requests, the result allocation process is allowed to pick one of the results and allocate that value to all outstanding elements with the same memory address (even if those requests have not yet been handed to the cache subsystem).

The result allocation process may mark all such allocations as complete to avoid any further request to the cache subsystem and further work if and when any outstanding cache read requests are signaled.

Alternatively, it may mark none of the elements complete, but apply the same process of updating all the "outstanding" processed elements with the most recent value and only when that set of read requests in complete mark the set complete.

However, knowing that the subsystem does not consolidate and does not maintain order it may be preferred that the scheduler do the consolidation. There are many approaches that can be used, but a reasonable strategy is to augment the bookkeeping of the first read request to the address to add to the list the current request.

C) if the cache subsystem does not consolidate concurrent identical read requests but does maintain order of at least identical address read requests, the reverse order only occurs because the scheduler made an out of order request. This can be handled as in B) above. Hopefully there is a good design reason to shift the onus to after the cache subsystem response. 

However, this does violate the premise I established that RVWMO provides meaningful relaxation even in the presence of 0 to vl processing order.
The significant idea here is that the RVWMO model gives considerable latitude to implementer because
within an RVV load there is 
- not explicit memory synchronization ( rules 4 through 8),
- no syntactic dependencies ( 9 through 11) and
- no pipeline dependencies (12 and 13).
- Overlapping address orderings 1 and 3 do not apply, and
- as all elements are of the same size, **if all reads are memory size-aligned, non-overlapping or identical addresses**, then all identical reads can be consecutive without an intervening write eliminating rule # 2.
      (see previous comment for demonstration by induction)

As a result an unordered read is unnecessary in the common **bolded** above use case,
and in many other cases as well.
(e.g. unaligned depending upon how such unaligned memory fields are allowed to be decomposed and/or whether there is cache-line crossing).

It is worth noting that the results that can occur may be unexpected .

On completion of the RVV load the values in elements 0 through vl of vd can have any values consistent with the global memory order within a window.
That window is bounded by the Program Order Rules which as we see can provide an extremely large range. The opportunity for "unexpected" results is large, but worse, because the reads are stipulated to occur in 0 to vl there is an expectation that "strict" memory ordering is enforced. What RVWMO may actually allow is not anticipated, may be thought to be superseded by process order, or is completely absent in consideration.

The values could  be in reverse chronological order, or in reverse dependency order as seen by the other hart(s) doing the writes.
There is currently no provision to provide fences between successive reads.
So, even though order 0 through vl may be stipulated the returned values in many use cases are not at all dependent upon that order and, in fact, the stipulation creates an unnecessarily complex and potentially misleading conceptual process model.

> And the hardware doesn't know what will match until late in the game?

There are many approaches that are consistent with out of order read processing and identical address consolidation that do not need to know what will match until the element is reached. More aggressive ooo may read further ahead, but even simple opportunistic approaches will benefit from judiciously identifying non-overlapping address scenarios, identifying duplicates and consolidating as they arise.

kasanovic, 2020-12-18:
Resolved in https://github.com/riscv/riscv-v-spec/commit/511d0b84a3848de437fd01990d078feaa2871b11

closed 2020-12-18 by kasanovic