riscvarchive/riscv-v-spec #504: Further Ordering Relaxation of Unordered Indexed Stores/Loads

billhuffman, 2020-06-26:
In high performance implementations, it is difficult to order an indexed store with following scalar operations.  I propose that the unordered indexed store and unordered indexed load (see issue #501) not be required to be ordered, even on the same hart, with any following memory operation except other unordered indexed loads and stores up until some particular instruction is executed.

I don't think we want a special instruction that waits, but rather an instruction that already exists, is easy to use, but won't happen at the wrong time.  I suggest the CSR read of the vtype register.

Any memory operation before the unordered indexed store or load will be ordered before it.  Any memory operation after the CSR read will be ordered after any unordered indexed store or load that was before the CSR read.

I think with a definition like this, it doesn't need to be an extension or a mode but can be in the spec.   The unordered versions can, of course, be implemented as the ordered ones for simpler hardware and all ordering will be automatic.  For cases where it is not know that the delayed operation is OK, software can pick the ordered versions.  This is another reason why unordered indexed loads should be included along with the stores.

David-Horner, 2020-07-10:
I agree that we should express RVV concurrency in a relaxed manner. this will allow for a RVV specific RVVWMO definition that allows for the parallelism requested..
However, even within the current RVWMO model and a process model that stipulates "as if processed consecutively from 0 to vl" there is sufficient latitude to provide significant parallelism and respect the currently provided fencing instructions.
I do not believe there is need to provide a heavier handed mechanism for <unordered>vector to scalar to <unordered>vector  fencing. And thus I disagree with a mechanism that specifically uses a csr to do the job.

kasanovic, 2020-12-18:
Group decided to stay with RV MCM model for inter-instruction ordering.

closed 2020-12-18 by kasanovic