riscvarchive/riscv-v-spec #510: add element index to vstart for segment loads and stores

solomatnikov, 2020-06-30:
Different elements of the segment might end up in different pages, causing more than 1 page fault.

The proposal is to to use 3 MSBs of `vstart` as an element index within the segment.

jnk0le, 2020-06-30:
related to #493

David-Horner, 2020-07-02:
Both indexed and strided versions of segmented loads and stores can cause multi-page faults at a given element index.
Unit-stride segmented load/stores can also cause multiple synchronous exceptions within a segment. Consider PMP and PMA exceptions.

This issue would not be problematic if vstart were in units of elements rather than  in units of structures.

As I see it, either we use units of elements or omit offset/index  within structure.

 As I commented in 493, this would be the "bare metal" visible info, that may be (and should be considered) opaque at any less privileged level.

solomatnikov, 2020-07-06:
> This issue would not be problematic if vstart were in units of elements rather than in units of structures.

If vstart were in units of elements, then divide is required to determine element index in vector register, which is bad for implementation.

aswaterman, 2020-07-16:
This feature isn't necessary, for the same reason we don't need to have special handling for misaligned scalar memory accesses that straddle page boundaries.

If either page is unmapped, you take that fault and restart from the beginning of the segment.  If both pages are unmapped, you'll handle one then handle the other.

The fact that the result may be partially written back to the regfile isn't problematic, since for indexed accesses we disallow overlap with the index vector altogether.  So there's always enough information to restart.

David-Horner, 2020-07-17:
What this does not provide is guaranteed forward progress.

Restarting at the beginning of the segment can reintroduce the fault, even if each fault is independently dealt with.
Consider e1024 with 8 fields. This can potentially span three 4K pages, thus potentiality 2 faults within the same restart group. Not identical to the misaligned scalar case. Ensuring 2 pages are resident for the spanned scalar operation is stringent enough. Ensuring 3 pages are resident is more onerous but is avoided if element rather than segment restart can be specified.

Similarly, as I mentioned,  PMP and PMA exceptions can occur in addition to page faults.
Even without the complication of page faults, if the policy is for the OS to perform the operation for that element, resumption at the segment level does not allow forward progress as those affected elements will fault when once again reached.

Granted, it can be stipulated that the execution environment has to first handle all the elements in the segment and then restart at the next segment. But this, too, is substantially different than the scalar case. Element resumption simplifies the EEs requirements.

aswaterman, 2020-07-17:
e1024 is 128 bytes, not 1024 bytes. So eight of them can only span two pages. This is the same as the existing misaligned scalar case, which requires two data pages (and two code pages) to be mapped in in order to guarantee forward progress. Since we aren’t exacerbating the forward progress onus on software, that isn’t a suitable justification for this feature.

David-Horner, 2020-07-17:
Oops.
However, are PMP and PMA exceptions irrelevant?
Do they not combine with the page faults to complicate EE handling? These exceptions would have to be handled collectively to ensure forward progress. Scalar only deals width a single element at a time whereas segments deal with multiple. This touches on the same issue that caused multiple register load/restore to be rejected in RVC.

aswaterman, 2020-07-17:
Not irrelevant, but also not problematic AFAICS.  PMP and PMA exceptions are usually not meant to be restarted.  Even when they are restartable, the same sort of forward progress issues you get with paging don't manifest, since they're only protection mechanisms, not translation mechanisms.  You can contrive EEs that would break under these assumptions (e.g., one that gives user programs access to only two adjacent 512-byte regions at any given time), but I've yet to see a realistic example.

David-Horner, 2020-07-17:
One last area of concern and a closing "rejoin".

We are potentially going to have cache and cache-line challenges with e1024 (and worse when misaligned) and that compounded by segments of 8 elements.
I believe segment rather than element restart increases the challenges for these extreme cases to have deterministic forward progress. Cache size and associativity complicating the analysis. Given that even simple machines may have crypto support that uses e1024 it is not only the HPC with huge cache that are affected. The RVC argument remains relevant to this corner case as well.

rejoin; e1024 and pages: 
What, no 4K bit pages?
(would be custom of course) ;-)

aswaterman, 2020-07-17:
Forward progress with respect to cache hits is different, in the sense that cache misses don't cause architecturally visible exceptions.  Microarchitectures are obviously free to devise whatever partial completion and restart mechanisms they need to guarantee forward progress with respect to cache hits.

David-Horner, 2020-07-19:
Regarding cache behaviour, substantial consideration was given LR/SC, and especially SC success conditions with respect to cache live-lock and forward process guarantees.  
Although processor lock-up may not be architecturally visible, repetitive time-out interrupts on successive restarts of the same instruction certainly is.
As with LR/SC, can we not give guidance and even provide a mechanism that will in all but pathological cases succeed.
Why should segmented load/stores not be afforded the same attention?

aswaterman, 2020-07-19:
I guess to me it’s obvious how to guarantee forward progress for these instructions, whereas for LR/SC it is more subtle. But I’m not opposed to adding a remark that the assumption is that a single segment must eventually be able to execute to completion, irrespective of cache or TLB misses, in order to guarantee forward progress. 

“All but pathological cases” doesn’t cut it - if we recommend a microarchitecture, it must suffice for the pathological cases, too.

David-Horner, 2020-07-19:
>“All but pathological cases” doesn’t cut it - 

It was for the LR/SC that I suggested such cases do exist.
Not wanting to be pedantic, but an obvious pathological situation would be a DMA periodically taking the lock on that page in cycles just under the execution time of the allowed 16 instructions.
Similarly, although debug triggers set between LR and SC are specifically called out as a situation (pathological as it is), running with debugged enabled has the potential to disrupt LR/SC timing and thus cache-line lock loss due to other concurrent harts/devices even if triggers are outside the critical 16 instruction section.

Perhaps, LR/SC's get-out-of-gaol-free-card is the fourth bullet:
> The LR and SC addresses must lie within a memory region with the LR/SC eventuality property.
Is this "R/SC eventuality property" definable as specifications outside LR/SC non-failure or is this just circular reasoning?

The suggested "assumption" remark also appears comperable to LR/SC's fourth bullet in lack of guidance, and no where close to the specific constraints of the other 4 bullets.

LR/SC and its constraints were carefully crafted to ensure usability and reliability.
The granularity of restart afforded segmented load/store does not appear to have been granted the same consideration.

2henwei, 2020-08-06:
> If either page is unmapped, you take that fault and restart from the beginning of the segment. If both pages are unmapped, you'll handle one then handle the other.

Would kinda thrashing happen in this case? I know this may rarely happen though.

aswaterman, 2020-08-06:
@2henwei My point is that the situation is the same as for misaligned instruction fetches, scalar loads, and scalar stores.  Thrashing is theoretically possible, but because in all of these cases, only two adjacent pages need be mapped, OSes are capable of guaranteeing forward progress.

closed 2021-01-29 by kasanovic