riscvarchive/riscv-v-spec #511: [question] how to build the vtype for vsetvl instruction?

JerryShih, 2020-06-30:
How to generate the rs2 vtype for vsetvl instruction?

`vsetvl  rd, rs1, rs2`

Does it just use the calculator to find the vtype value according to the following table?
https://github.com/riscv/riscv-v-spec/blob/master/vtype-format.adoc

Could we have a helper function to get the vtype value?
e.g.
```
vgetvtype t0, e32,m8
vsetvl xx, xx, t0
```

knightsifive, 2020-06-30:
I see two approaches. The most efficient might be simply precomputing the `vtype` value using the table you referenced, and loading it into an X-register with `li`. An alternative is to use `vsetvli` and then `csrr` the `vtype` register. This alternative has the disadvantage of overwriting `vtype`, but this is harmless in your example.

jim-wilson, 2020-06-30:
We can define some constants, in C macros or asm equates, put them in a header file, and then you can do for instance
	li	t0, RVV_E32|RVV_M8
because gas handles simple C operators on constants.  So this works for instance
	.equ	RVV_E32, 0x2<<2
	.equ	RVV_M8, 0x3
	li	t0, RVV_E32|RVV_M8
and this works as a .S file
#define V_E32 (0x2<<2)
#define V_M8 0x3
	li	t0, V_E32|V_M8

JerryShih, 2020-07-02:
Thank you for the examples.
My opinion is that if we have a particular instruction(e.g. **vgetvtype t0, e32,m8,ta,tu**) to construct the vtype, the asm code could be more portable if the vtype format changes in the future.

knightsifive, 2020-07-02:
Can you give any applications of using `vsetvl` (non-immediate form) with a static `vtype`? The only time I've ever used `vsetvl` is in context-restore code, and in that case `vtype` was "dynamic" (and previously saved via `csrr`).

JerryShih, 2020-07-02:
@knightsifive 
Here is an example:
https://github.com/riscv/riscv-v-spec/blob/master/example/strcpy.s#L9

I would like change the lmul from m1 to m8 during the loop(it's similar to the tcp packet size increasing).
When the string data is small, it might wastes a lot of time for the long m8 data loading. If I could increase the lmul for every loop, it might be good for both small and large string data.

kasanovic, 2020-07-10:
For this example, if you start with LMUL=0, then can just increment the vtype value in register to bump up LMUL while leaving other fields unchanged.

kasanovic, 2021-03-19:
This is a software concern rather than issues with vector spec per se, so closing here.

closed 2021-03-19 by kasanovic