riscvarchive/riscv-v-spec #514: How to combine/split vectors?

joy2myself, 2020-07-03:
Hi,

I want to use rvv to combine/split vectors.
Combining operation is like the vcombine instruction in NEON, to contact two short vectors.
And splitting operation is something like vget_low and vget_high instructions in NEON, to get halves of the vector.
But I didn't find instructions to combine/split vectors in the reference manual.
I think maybe I can use shift operation and some arithmetic operations to realize these two functions, but it is too complicated. Is there an easy way to do this? Or should we add such instructions?

Sincerely,
Yin

kasanovic, 2020-07-03:
"vslideup.vi vdest, vsrc, 64"  with vl set to 128b (e.g., SEW=8, vl=16) will move the lower 64b of vsrc into bits 64-127 of vdest.

kasanovic, 2020-07-03:
In general, look at the slide instructions for this kind of operation in Section 17.3.

closed 2020-07-03 by kasanovic

jim-wilson, 2020-07-03:
I would suggest the best way to combine two LMUL=1 vectors is to treat them as a single LMUL=2 vector.  Then the combination is just a register allocation problem to put the two vectors in a valid register for an LMUL=2 vector.  Similarly, the best way to split an LMUL=2 vector is to treat it as two LMUL=1 vectors.  This doesn't work for fractional lmul, but you can use slide instructions for that as Krste mentioned.

joy2myself, 2020-07-05:
Thanks a lot for the guidance! But I still have some questions.

As Krste mentioned, I can combine two vectors as bellow:

```
vsetvli  t0, 8, e8,m1
vle8.v  v1 src1
vle8.v  v2 src2
vsetvli  t0, 16, e8,m1
vslideup.vi  v1, v2, 64
```

But now I want to use rvv intrinsics in C++ to implement this function.
And the intrinsic of `vslideup` instruction is defined as bellow:

`vint8m1_t vslideup_vx_i8m1 (vint8m1_t src, size_t offset);`

I can not pass the dest vector (v1) to `vslideup`, so they are not combined. How can I use intrinsics to implement this assembly code?

Sincerely,
Yin

kito-cheng, 2020-07-06:
@jim-wilson I thinkg it might not work as expect if VL < VLMAX even with non-fractional LMUL?