riscvarchive/riscv-v-spec #515: Possible conflict between vector register grouping (3.3.2) and ELEN > VLEN?

JamesKenneyImperas, 2020-07-03:
Section 3.3.2 contains this statement and note:

_Implementations must support fractional LMUL settings for LMUL ≥ SEW/ELEN, for the ELEN value at LMUL=1. An attempt to set an unsupported SEW and LMUL configuration sets the vill bit in vtype._

Note | Requiring LMUL ≥ SEW/ELEN allows software operating on mixed-width elements to only use a single vector register to hold the widest (ELEN) elements, with fractional LMUL used to hold narrower elements. When LMUL < SEW/ELEN, there is no guarantee an implementation would have enough bits in the fractional vector register to store at least one element, as VLEN=ELEN is a valid implementation choice.
-- | --

Is this statement and the following note strictly true when ELEN > VLEN?

For example, if ELEN=64 and VLEN=32 then when SEW=8 the first statement implies that a fractional LMUL of 1/8 must be supported, but in this case the wider value will occupy _two_ registers.

Thanks.

JamesKenneyImperas, 2020-07-03:
(Perhaps if section 3.3.2 is enhanced with examples as suggested in #479 then this case could be included as well.)

kasanovic, 2020-07-03:
The text explicitly says "for the ELEN value at LMUL=1".    If ELEN=64 and VLEN=32, then ELEN at LMUL=1 is 32 and only fractional LMUL of 1/4 must be supported.

JamesKenneyImperas, 2020-07-04:
The specification states in section 2 that ELEN is an implementation-defined constant parameter; here you are saying its meaning varies with LMUL in some circumstances. I think the clause _for the ELEN value at LMUL=1_ should be clarified because it currently isn't obvious (to me at least) what this means.

Perhaps it would help to introduce another derived parameter here, e.g. FRLEN=min(VLEN,ELEN), and simplify the statement to something like:

_Implementations must support fractional LMUL settings for LMUL ≥ SEW/FRLEN_. An attempt to set an unsupported SEW and LMUL configuration sets the vill bit in vtype.

(I think this is what you are saying, but I'm not sure.)

kasanovic, 2020-10-26:
Clarified in https://github.com/riscv/riscv-v-spec/commit/38e37429c26f1538dd23c5f14c02d9f628928fbc#diff-ea57dd7a8daf0aa62f553688c1970c8e6608945d25597f8661c5ea6670fb509c

closed 2020-10-26 by kasanovic