riscvarchive/riscv-v-spec #517: Vector Load/Store Whole Register Instruction formats

JamesKenneyImperas, 2020-07-06:
Section 7.9 now shows the formats for Vector Load/Store Whole Register Instructions as:

```
Format for Vector Load Whole Register Instructions under LOAD-FP major opcode
31 29  28  27 26  25 24   20 19       15 14   12 11      7 6     0
 nf  | mew| mop  | 1| 01000 |    rs1    | width |    vd   |0000111| VL<nf>R

Format for Vector Store Whole Register Instructions under STORE-FP major opcode
31 29  28  27 26 25  24   20 19       15 14   12 11      7 6     0
 nf  |  0 | mop |  1| 01000 |    rs1    |  000  |   vs3   |0100111| VS<nf>R
```

Is this correct, or should the `mop` fields actually be `00` (i.e. unit-stride only)?

Rolfy360, 2020-07-07:
In Table 7, 8
mop=00 is for unit-stride
and in Table 9, 10 
lumop/sumop fields are 01000 for unit-stride, whole registers
so I think the mop here should be 00.

also in vmem-format.adoc
mop in unit-stride and strided column
should use 00 and 10 directly.

Format for Vector Load Instructions under LOAD-FP major opcode
31 29  28  27 26  25  24      20 19       15 14   12 11      7 6     0
 nf  | mew| mop | vm |  lumop   |    rs1    | width |    vd   |0000111| VL*  unit-stride
 nf  | mew| mop | vm |   rs2    |    rs1    | width |    vd   |0000111| VLS* strided
 nf  | mew| mop | vm |   vs2    |    rs1    | width |    vd   |0000111| VLX* indexed

Format for Vector Store Instructions under STORE-FP major opcode
31 29  28  27 26  25  24      20 19       15 14   12 11      7 6     0
 nf  | mew| mop | vm |  sumop   |    rs1    | width |   vs3   |0100111| VS*  unit-stride
 nf  | mew| mop | vm |   rs2    |    rs1    | width |   vs3   |0100111| VSS* strided
 nf  | mew| mop | vm |   vs2    |    rs1    | width |   vs3   |0100111| VSX* indexed

kasanovic, 2020-07-10:
Fixed with #526

closed 2020-07-10 by kasanovic