riscvarchive/riscv-v-spec #520: vrgather SEW=8b only works up to 256 elements

respasa, 2020-07-07:
The spec already warns about SEW=8 in the text saying:

> When SEW=8, vrgather.vv can only reference vector elements 0-255.

However, this effectively makes 256 x 8 = 2048b the largest VLEN where the ISA "fully/completely" works. Past that size, vrgather.vv does not fullfil its role. 

One proposal would be to define EMUL and EEW for the "index" source of vrgather based on the implementation VLEN of the machine for SEW=8, 

```
 if ( SEW == 8 && VLMAX > 256 ) {
   if ( LMUL == 8 ) { illegal instruction }
   else             { EEW = 2*SEW;  EMUL = 2*LMUL }
}
```
This would now "fix" vrgather for SEW=8 and LMUL=1,2,4. Obviously, this can't work for LMUL=8. And, yes, this only moves the goalpoast from VLEN=2048 to VLEN=131072, but that feels more satisfactory...

David-Horner, 2020-07-10:
I believe the current formulation is useful even for eew=8, so eliminating it is problematic.
It is possible to have another variant of vrgather that works on mixed vector for vs1 and vs2 if this use case is warranted.This could be added post v1.0 hopefully with the expedited process in place.

I see this as an after version V1.0 issue.

kasanovic, 2020-07-10:
Software would have to know VLMAX to be able to use this form, so this would break software portability across different VLEN machines.

kasanovic, 2020-07-10:
Proposal from task group is to add a vrgatherei16, which has fixed EEW of 16 for the index vector.  This in particular would support vrgather of >256 8b elements. Another benefit is lower register usage for the index vector for SEW>16b.

commit a6792503 (2020-07-17): Added vrgatherei16 instruction.
Closes #520.
closed 2020-07-17 by kasanovic