riscvarchive/riscv-v-spec #521: vcompress --> vdecompress

HKLee2040, 2020-07-07:
Since there is a "vcompress" to pack the non-zero value together, why not a "vdecompress" to unpack the non-zero value to original form?

David-Horner, 2020-07-10:
vcompress can be processed from 0 to vl without back tracking.
The proposed vdecompress would have to first determine the last element to move to a higher location (scan the mask register) then work backwards through both mask and data to copy the masked element to the location designated by the mask bit.
This would be an exception to normal order processing, which is also the reason why vfind-last-set-mask-bit is not included.

Such overhead is taken seriously for v1.0 as exemplified by the note for this instruction.

> Although possible, vcompress is one of the more difficult instructions to restart with a non-zero vstart, so assumption is implementations will choose not do that but will instead restart from element 0. This does mean elements in destination register after
> vstart will already have been updated.

The main cost/complexity for vcompress restart is determining the number of set mask bits up to the restart point.
Just the mask bit counting for vdecomress to determine start point is more complex/costly. So it definitely exceeds a threshold applied elsewhere, and thus is not included.,

HKLee2040, 2020-07-10:
Thanks for your information.

There is one interesting paper just for your reference:
"ZCOMP: Reducing DNN Cross-Layer Memory Footprint Using Vector Extensions"
Publication:MICRO '52: Proceedings of the 52nd Annual IEEE/ACM International Symposium on MicroarchitectureOctober 2019 Pages 126–138
https://doi.org/10.1145/3352460.3358305

They define two new ZCOMP instructions: zcompl and zcomps. 
zcompl is used to "load" and decompress the elements, whereas zcomps compresses and "stores" them.
And the base address is updated according to non-zero value count.

closed 2020-07-14 by HKLee2040