riscvarchive/riscv-v-spec #522: overlap of destination register with sources on VRED operations

respasa, 2020-07-07:
In section 15 of the spec on reductions, we have the following sentence:

`The destination vector register can overlap the source operands, including the mask register.
`
Why are we making this so "lax"? I can see how the vs1 and vd registers want to be the same, so that you can keep accumulating into the same logical register. However, overlapping the vs2 register (the vector one) complicates the operation for renaming. If LMUL > 1, and vs2 has been assigned, say, 4 physical registers, having vd overlap vs2 AND allowing vd to not follow the LMUL rules requires patching up the rename tables. In fact, I would probably force vs1 and vd to follow the LMUL rules.

David-Horner, 2020-07-10:
I see this as similar to allowing EEW > VLEN.
It complicates the LMUL=1 normal case.
However, an implementation can avoid this by fixing EEW <= VLEN.
There is no global setting that restricts the VRED register overlap.

Of course, an implementation can trap on such overlaps, and emulate using another RVV scalar (element [0]) or even use in the trap a set of custom VRED ops that write to a RVI xn register (or custom xn set) instead. Then load it to a RVV scalar.
So, an implementation could keep its internal engine simple at the expense of these overlapping trapping variants taking longer.
The chosen domain software will avoid the pain and devise code sequence that does not overlap if the pain is sufficiently large.

Perhaps for V1.0 we should address the larger question about how to handle:
Stipulating in a given "use domain profile" that a feature (like this register overlap) is disallowed and restrictions/failure anticipated. 
Establishing a generalized method for defining these and perhaps informing operating environment that it is present.

kasanovic, 2020-07-20:
Changing LMUL in-between instructions results in the same problem of having to rename portions of vector register groups.  I don't see how you get away from renaming individual vector registers in any renamed system, or what the advantage would be.

closed 2020-10-23 by kasanovic