riscvarchive/riscv-v-spec #531: Mask layout complicates parallel operations

bbougenot, 2020-07-13:
Consider an architecture where the vector register file is divided accross lanes. Vector elements are interleaved accross lanes. Each lane processes Its corresponding group of elements in parallel. Since all vector elements are independent from one another, no communication between lanes is required (except for a few instructions such as slides). However, the new mask register layout resuts in mask information being scattered accross lanes since we have bit level granularity. In addition to this several logical operations require results to be in mask register format. As a result special interlane communication which could lead to routing congestion is required just for mask and logical operations. This was not an issue when mask information was stored in the least significant bit of Its corresponding element. Is there a clever way to map mask information accross lanes? It seems the new mask configuration adds significant complexity since every register requires two completely different access patterns.

David-Horner, 2020-07-15:
I certainly agree that this model simplicity increased hardware complexity.
However, there are various problems with the old format, not the least of which was it breaking with fractional lmul #443 .

I agree with what I see as  a bias towards simplifying the RVV ISA model at the expense of moderate hardware complexity.
But I also agree with you that specific decisions made with that bias need to be justified.

Simplifying the RVV ISA model at the expense of software benefits is also counter productive.
It appears harder to identify software benefits as they are more subjective. 
How data is organized (e.g. complex numbers in adjacent pair or separate arrays) has a potentially substantial impact on what facilities are present in the ISA.
In this case, for mask values, does the programming community prefer  a linear bit string of mask values to elements  or an element to byte correlation?
This is the fundamental question behind issue #435 and  #448.

I proposed byte correlation to element as far back as #399.

#481 (#485)  is another issues about mask bit string concerns.

The proposal in #448 partially addresses the alignment problem.
It does not eliminate.

However, the impact to hardware for either MLEN=1 proposal is minimized for short VLEN (minimal systems 32 or 64 have minimal wiring issues).

For larger VLEN similar tricks for mask as for SLEN are possible.
As it is frequently the case that setting the mask and using it are at the same EEW, it makes sense to tag the mask register with the EEW used to set the values.

kasanovic, 2020-07-20:
There are hardware/software tradeoffs with any mask design.  We rejected separate predicate registers as that would require a whole different set of architectural state and associated opcodes to operate on that state.  We tried various predicate interleaving formats, but these all optimize for some assumed microarchitectural design point or another at detriment to others.  The simple MLEN=1 design is simple to describe and helps some software use cases, such as using/generating packed predicate bit vectors in memory.  The ISA also helps microarchitecture optimizations by only using v0 as the source of a mask in an instruction, which can be handled specially in microarchitectures to provide the additional read port.   Mask-mask operations can be trivially bit-sliced. Comparison and vmadc operations require cross-lane communication of mask values, but the bandwidth of results is much lower than for data values (1/8 at most).  The vfirst/vpopc and vms?f instructions require special functional units, but need cross-lane communication with any interleaving - arguably the simple MLEN=1 packing is the most efficient for these instructions.    After considering many options in the past, we are sticking with this simplest packing in v1.0.

closed 2020-07-20 by kasanovic

David-Horner, 2020-07-20:
I gather "we" means ratification at a TG meeting.

kasanovic, 2020-07-20:
Ratification happens after public review and approval by TSC.

Before that, the TG has to agree on what to put up for public review, which is what v1.0 is.   We will run a vote within TG to approve putting the proposal up as v1.0.

bbougenot, 2020-07-21:
Thanks for your responses. I guess for the time being there's no alternative to interlane communication. Are there plans to further change the mask design after v1.0 or is this part of the ISA starting to stabilize?