riscvarchive/riscv-v-spec #532: define vstart as an opaque value (some values will have specific meanings)

David-Horner, 2020-07-17:
To future proof vstart I propose vstart be defined as an opaque value that in general is not the restart element index, but rather some encoding that provides the necessary information to restart at either the hardware provided element/condition or modified to restart at the next element. 

Future datatypes may need to be restartable to allow reasonable interrupt response (e.g. crypto graphic operations) with information that does not correlate to an element number/location.

Affected definitions and documentation:

1) The definition of prestart need to also be changed to state "vstart determines for the hart the elements to be ignored as already processed". ( e.g. a minimal RVV64 system may provide a bit map that identifies the 64 bytes that are the prestart.)

2) As previously implied ,make explicit that a zero vstart value means no restart is attempted for the next restartable RVV instruction. 
    (This is implicit throughout the current documentation, especially in defining trapping behaviour in terms of non-zero vstart, which  is logically equivalent and where implied as a zero length prestart in section 5.4)
  
3) Further, although implicit in the current definition, adding 1 to the hart generated non-zero vstart will provide to a restarting instruction a request to restart at the next element. This is technically new functionality that is easily assumed to work (but according to current definition is allowed to trap). Note, as the hart generated vstart can be generated as always even, the add of on becomes a one bit flag to the provided opaque value. (as in the minimal RVV64 system, the lsb can identify the most significant byte in the 8th register in the register group. As a result it would not be filled in if a restart is pending).

4) vsrtart csr is redefined to be as many bits are required up to including XLEN. 

5) The rule that the instruction is not processed if vrestart > vl must be rephrased as not processed if only tail and prestart (elements) are present/exist. Alternately, body is the null set.
 
6) Each implementation may provide an algorithm or a custom csr to derive information on the restart element location. (we may reserve a standard csr for this purpose).   

Add a  note that the usage as currently described (vstart as an index to the relevant element) is a valid implementation of the opaque vstart.

David-Horner, 2020-07-18:
An obvious extension to this proposal is that a bit be reserved in vstart (XLEN-1, the sign bit, the most likely candidate) to differentiate the PoR model, plain text element pointer implementation from the opaque class of implementations.
Benefits:
1) If sign bit of zero means the PoR model, no change to existing implementations is required.   
2) no out of processes checks need to be made. Situations that need to manipulate or compare the vstart value will most likely be reading the value anyway. Its sign is thus readily determined and no other extraordinary process need to be followed.
3) Initial implementations are going to be the simple model anyway, so KISS, Keep It Simple "Simple".

I will address these in reverse order:
# Counter arguments:
# 3 KISS it.
>"It can scarcely be denied that the supreme goal of all theory is to make the irreducible basic elements as simple and as few as possible without having to surrender the adequate representation of a single datum of experience." - Albert Einstein (see **)

 The vstart concept is far from simple. As implementations become more sophisticated, both the recovery mechanism and determination of the point for restart can be simplified, and restart process be more efficient if more information than just the element restart location is available to represent the needed restart information. If sufficient flexibility is not provided in the vstart representation, then custom restart/recovery methods/techniques and facilities will be invented and re-invented adding an unnecessary burden to EEs and OS. Fragmentation is inevitable.

Settling prematurely on or even biasing towards an oversimplified model is short sighted, may need flag day for successor, and sets up for future cost. 

 However, I believe the characteristics stipulated are 
- of low or no additional burden to hardware
- functional: all the facilities required by the OS/EE are available (affected element identification, restart request at current and next element)
- extensible: multiple generations of an implementation can enhance operations/performance/etc without having to change support environment software. If an implementer decides to introduce a new procedure for restart (e.g. an new algorithm to determine affected element) most if not all the infrastructure will be in place for it. 
- be well supported: There will, in any case, be pressure to conform to existing standards and methods. To the extent in which those are already robust there is little need to deviate. We can anticipate that brilliant minds are ready to and already consider the issues related to an opaque restart identifier.

# 2 Simplicity – use existing as per PoR

Using the high bit which will naturally be low in PoR formulation will allow in channel distinction of PoR vs an opaque representation. It provides no benefit to opaque nor “PoR plus” representations. It biases towards PoR and thus disadvantages other representation.
Note: PoR plus is a representation that appears to be a simple PoR design, but as the implementation has substantial latitude to determine and thus provide restricted restart values, it also has the opportunity to imbed in the encoding valuable restart information. This enhancement would need out of channel checking to determine and thereby utilize any additional “user visible” information.

The vstart encoding format is expected to be consistent throughout the up time of a hart. There is no need for deriving the nature of the encoding on each use. Rather at start up the appropriate determination can be made by an out of channel process and appropriate run time routines loaded that will correctly operate on the identified vstart formulation. 
Software managing many exception types do not need to knowing the current element. In such situations there is no need for the hart to provide a mechanism to determine the actual current element, rather, encoding can be specific to the event allowing increased restart data by foregoing a consistent method to derive a “current element”. Indeed, hardware designers can differentiate; applying their secret sauce without exposing its nature.

Current element identification  is only required for emulation ; emulation of an environment, feature or functionality.  It is following such an emulation that advancing to next element would be desirable. Emulation requires a non-trivial determination of the circumstances and understanding of the consequences of modifying the current element in software. To (also) enable extensive bit encoding in these cases, an implementation can stipulate which if any instructions can request restart at the next element. EEs that still wish to do emulation can in many/most cases emulate the whole of the instruction and safely advance to the next instruction. e.g. allowing access to PMP and PMA protected regions during RVV load/store execution ( according to authorization controls; i.e. environment emulation).

 Note: requesting restart at the “current” element and at the next element are the same for all implementations, so common code can be reused, refined and perfected, but where differentiating is needed, it can be done within an envelope the allows extensive diversification, still with common case  handling. {e.g. context switching only requires vector register and vstart save; with restore upon switch back.)

[... cut ...]
Given the small benefit realized by using special encoding within vstart, I strongly recommend against this being even a partially authoritative source for vstart format determination.
- vstart format determination should be performed as part of hart configuration (not dynamically on each exception).
- embeded format information is of little informative value, cannot be determined prior to an exception and hobbles hart restart encoding.

**
This is Einstein's exemplification of the aphorism "Everything should be as simple as possible, but no simpler".

David-Horner, 2020-09-17:
related to #493 
actually could be exactly the same issue.
@jnk0le first tied the issue to indexed load/stores but then also to determining "element" with ediv/crypto, etc.
And so he also was considering the need to future proof vstart.

David-Horner, 2020-09-17:
If plain text index of restart element is to be provided by the implementation, I recommend this approach, suggested above:

> in the alternative, zeros in XLEN-1 to log2(8*vlen) could be used as the flag, 

I favour this approach as it is subsumes POR; is compatible with POR and indistinguishable when enabled.

kasanovic, 2020-10-26:
For v1.0, we're only specifying plain vstart, that points to faulting element when possible.  In general, imprecise restartable/swappable exceptions require considerably more micro architectural state than can fit in a single XLEN-bit register.  There is a placeholder in priv specs for adding support for swappable exceptions in coprocessors with opaque state save/restore, but having a standard format for vstart on imprecise exceptions is valuable for debug tools.

closed 2020-10-26 by kasanovic