riscvarchive/riscv-v-spec #533: Mandate VLEN>=128 for application processor profile

kasanovic, 2020-07-17:
Having a lower bound on VLEN simplifies portable code.  Discussion in task group considered VLEN>=256, but 128 includes common design points for application processors.

David-Horner, 2020-07-17:
Also relevant is  the discussion that RVV base will address or be synonymous with "application processor profile".
Should the title be changed to reflect the RVV base relevance?

phkahler, 2020-08-19:
There are so many applications that use 3 and 4-element vectors I don't think 128 is enough. Sure there are applications using single precision FP vectors of those sizes, but that doesn't make it an appropriate minimum size for hardware. With 256 one can write a whole lot of 3D code without having to support hardware variations of vlen that are too short. If you have double precision I think it's fair to also mandate actual vectors in the ISA rather than just pairs of numbers. People can do serial operations in hardware if they want to skimp on gates.

kasanovic, 2020-08-19:
Mandating 256b VLEN would double the minimum size of the vector register file which is one of the largest components of a vector unit. 

With a guarantee of VLEN>=128, code that needs 64b vectors can use LMUL=2 to guarantee space for 4-element vectors of 64b/element.  LMUL=2 does halve the number of vector register groups from 32 to 16, but this is the cost of providing portability to smaller implementations.

closed 2020-10-26 by kasanovic

brucehoult, 2021-04-11:
> There are so many applications that use 3 and 4-element vectors I don't think 128 is enough.

It seems a huge waste to use a machine that has vector registers holding 16 or 32 or 64 or more elements to load problem domain vectors of only 3 or 4 elements into a vector register.

I believe most applications with such short and fixed vector lengths would be better off using a vector register for each element of the application vector and processing a large number of problem domain vectors in parallel. Either the segmented load/store instructions (preferably) or strided load/store instructions can be used to convert an array of short vectors in memory into 3 or 4 vector registers holding element-wise arrays.

If code is written this way then VLEN = XLEN works fine and simply processes one problem domain vector at a time -- the same as loading just one 3 or 4 element vector into a huge vector register would.

I don't think widening instructions are a problem with VLEN = XLEN either, as I believe it's been decided to allow ELEN > VLEN.

commit 6ce90efe (2020-10-26): Specified that base "V" extension requires VLEN>=128.
Closes #533.