riscvarchive/riscv-v-spec #536: Eliminate inactive (masked out) elements from participating in RVWMO determinations.

David-Horner, 2020-07-19:
The only direction PoR - https://github.com/riscv/riscv-v-spec/commit/a67925038b653a75cd7eadb68b9915449941745d gives to load instructions is:
> Masked vector loads do not update inactive elements in the destination vector register group.

and thus is silent on masked out (inactive) elements participation in RVWMO memory ordering.

Consider a strided load:
vlse32 v8,(rs1), rs2, v0.t  where rs2 has value of 16 and rs1 is 32 bit aligned.

If v0 has even bits set, every other element has values loaded from aligned non-overlapping addresses .
RVWMO allows non-overlapping reads to proceed in parallel, allowing opportunistic unordered loading form any relevant present cache-line. 

If the odd elements must also have their data read (but not transfered into the vector register inactive elements), then RVWMO rule # 2 kicks in and opportunistic reading of present cache-lines is hampered until all required preceding cachelines are processed. In that interval cache-line may have been ejected to allow the predecessor entries to be loaded (or competing writes to occur), and would then need to be reacquired. 
Not only that, but as these odd access are not aligned a simplistic approach of making all ordered requests to the cache subsystem could yield pathetic performance. 

I am in favour of interpreting masked out (inactive) elements of a load as not needing to be read, thus not participating in RVWMO rule # 2 determinations.

The indexed case is also affected by decision.
(as are the segmented variants of strided and indexed).

aswaterman, 2020-07-19:
Going further, I had planned to write that mask dependencies are control dependencies under RVWMO. That is, a masked element operation is like a branch around the corresponding operation.

This is usually consistent with how the scalar code would’ve looked prior to vectorization.

It also subsumes your suggestion, since if the mask bit is off, then the corresponding element operation isn’t actually performed, so it doesn’t affect the PPO.

David-Horner, 2020-07-19:
I believe this issue should be addressed on its own even if later subsumed.

Even as a control flow dependency, the branch around could be defined as the "set in the register" part analogous to conditional move following the relevant calculation in transient working registers.

I agree with this mask as control dependencies idea, and also mentioned it in the context of a RVVWMO.

The problem is, that even though the possibility of control dependency can be determined syntactically, the actual dependency flow cannot, but is dependent upon current register values (in V0 specifically).

Do you have a solution for determining and tracing the specifics of this control dependency?
Are there any mature tools (graphing or analytic) to support this dependency type?

It would make consideration of the whole process model (including modifying vsew and/or vlmul that change register dependency allocation) much easier to formalize and develop a relaxed synchronous trap definition.

I hope you have something you can share, even if off list.
(I mention this here for transparency. The specific request I am making is for information on modeling/analysis tools and not any RISCV specific suggestion or decision vetting/avocacy).

kasanovic, 2021-06-07:
Closed by 0c7f5f7b52d0c92586aabddaf6090603ea7d0834

closed 2021-06-07 by kasanovic