riscvarchive/riscv-v-spec #539: Support only v[z|s]ext.vf2 integer extension instructions in base vector extension

michael-platzer, 2020-07-26:
The vector integer extension instructions that were added to the spec in version 0.9 (partially as a substitute for the removed byte, half and word vector loads and stores) effectively perform a (double-)widening, quad-widening or octuple-widening vector move, where a source vector register with an EEW of SEW/2, SEW/4 or SEW/8 respectively gets widened into a destination register with an EEW of SEW.

I propose requiring only the (double)-widening version (`v[z|s]ext.vf2`) in the base vector extension and introducing a new specialized ISA extension for the quad- and octuple-widening variants (`v[z|s]ext.vf4` and `v[z|s]ext.vf8`), similar to how the quad-widening multiply-add instruction got its own extension (Zvqmac).

When quad-widening multiply-add was first added to the spec, a note said:

    NOTE: It is currently unclear if quad-widening will be part of
    standard base, or part of an extension.  Quad-widening would add a new
    operand access pattern to the standard.

The quad-widening multiply-add instructions now has a dedicated ISA extension (Zvqmac) and is not required in the base vector extension, presumably to avoid introducing a new access pattern (possibly among other reasons).

However, requiring all implementations of the base vector extension to support the current integer extension instructions means that the quad-widening access pattern must be supported anyway and that an octuple-widening access pattern is required on top of that.

AFAIK, the byte, half and word vector loads and stores were removed to reduce the complexity of the memory interface, but requiring all implementations to support quad- and octuple-widening access patterns would essentially move that complexity to the integer arithmetic unit instead.

Quad- and octuple-widening integer extension (`v[z|s]ext.vf4` and `v[z|s]ext.vf8`) can be implemented in software by using two or three double-widening extension instructions (`v[z|s]ext.vf2`) in series (a `vsetvl[i]` would be required between every two of them to update the SEW). Hence, it should be sufficient for base implementations to support the `v[z|s]ext.vf2` variant only. Some implementations may then choose to support the quad- and octuple-widening integer extension instructions for better performance.

kasanovic, 2020-07-27:
The main concern is not about relative width, but about expected performance needed.
The extension instructions do not have to run at 2x, 4x, 8x normal bandwidth.  The source can be read at 1/2, 1/4, 1/8 of the normal destination rate.   These instructions were added to enable the widening loads to be dropped, and are much simpler to implement than the widening loads (which simultaneously have to shuffle data due to address alignment, and which receive data in possibly irregular chunks from the memory system). 

Implementations of the quad-widening macs would have been motivated (though obviously not required) to run at 4x rate to keep multipliers busy.

kasanovic, 2020-08-07:
This is really about which are in standard ISA subsets - implementers are always free to use a reduced set - however, for portable software the more generic ISA subsets  will include these.

closed 2020-08-07 by kasanovic