riscvarchive/riscv-v-spec #543: Is it possible to detect vector extensions in runtime?

joy2myself, 2020-07-31:
I need a runtime check for rvv support. Is it possible to check it in runtime? If it is, what should I do for a runtime check?

sincerely,
Yin

aswaterman, 2020-07-31:
In RISC-V, this sort of information is generally passed through the ABI, e.g., with a mechanism like https://man7.org/linux/man-pages/man3/getauxval.3.html

The RISC-V Linux port will provide a `getauxval`-based mechanism to detect the vector extension.

jim-wilson, 2020-07-31:
On Fri, Jul 31, 2020 at 12:40 AM Andrew Waterman <notifications@github.com>
wrote:

> In RISC-V, this sort of information is generally passed through the ABI,
> e.g., with a mechanism like
> https://man7.org/linux/man-pages/man3/getauxval.3.html
>
> The RISC-V Linux port will provide a getauxval-based mechanism to detect
> the vector extension.
>

Expanding on that a little, AT_HWCAP should have all arch extensions listed
as a bit vector like misa.  This assumes the kernel has vector support.  So
code like this should work in a linux application, run on a machine with
multiply support but not vector support.

hifiveu017:1029$ cat tmp.c
#include <stdio.h>
#include <sys/auxv.h>

int
main (void)
{
  unsigned long hwcap = getauxval (AT_HWCAP);
  if (hwcap & (1 << ('m' - 'a')))
    printf ("has multiply support\n");
  if (hwcap & (1 << ('v' - 'a')))
    printf ("has vector support\n");
  return 0;
}
hifiveu017:1029$ gcc tmp.c
hifiveu017:1030$ ./a.out
has multiply support
hifiveu017:1031$

If your target isn't running linux, and you have an embedded application
running in machine mode, then you should just check misa.  If you are
adding code to the linux kernel, then you should be calling an opensbi
routine to read misa.

Jim

closed 2020-08-14 by joy2myself

am11, 2022-07-28:
@jim-wilson, thank you for the explanation. I found your similar answer at https://lists.riscv.org/g/tech/message/70, from where I gathered that each hart can have different features / supported extensions on a given system.

Assuming this mixed-capability harts is going to be a(n unavoidable) trend in the future hardware; if the binary was compiled for riscv64g and running on a system with, say, four harts; two with `G` extension and two simple ones with `I`-only capability, will the OS scheduler (particularly Linux) seamlessly do the right thing for the executing process; i.e. OS will never schedule the execution on incompatible harts? Or is it considered the responsibility of (every multi-threaded) application to find the capable CPU and explicitly set affinity (using `pthread_setaffinity_np` programmatically or `taskset` command at launch time)?