riscvarchive/riscv-v-spec #544: Adding "addvl" instruction for efficient stack pointer calculation

chuanhua, 2020-07-31:
In order to manipulate RVV stack (prologue, epilogue, stack reference, register alignment, function argument in stack) efficiently, we need a scalar "addvl" instruction to add an integer multiple of vlenb value to stack pointer. Without this instruction, we need several instructions to calculate the stack pointer value and need to allocate several scalar tmp registers for this calculation which may lead to extra register save/restore code.

Example 1: allocate RVV stack in prologue:
`csrr  t0, vlenb`
`addi  t1, zero, 7`
`mul t0, t0, t1`
`sub sp, sp, t0`

Example 2: reference vector stack object, offset #7
`addi a0,sp,#unscaled_offset`
`csrr  a1, vlenb`
`addi  a2, zero, 7`
`mul a2, a1, a2`
`add a0, a0, a2`
`vl1r.v  v24, (a0)`

Is it possible to add such an instruction for efficient RVV stack support?

kito-cheng, 2020-08-03:
+1, that will significantly simplify the code gen for spilling vector register.

chuanhua, 2020-08-03:
The basic form of "addvl" instruction should use the vlenb CSR as an implied source operand. And I think whether LMUL needs to be considered as another implied source operand can be discussed.

David-Horner, 2020-08-03:
>The basic form of "addvl" instruction should use the vlenb CSR as an implied source operand. 

I disagree.
Since the operation is defined as sp+=( VLEN / 8) * (imm or rs1), and vlenb is defined as VLEN /8 there is a correlation, but the Uarch can determine if it wants to read vlenb or pull the bits from an internal VLEN representation.
> And I think whether LMUL needs to be considered as another implied source operand can be discussed.

I don't see any need for LMUL to be factored into the operation . The code needs to know LMUL to operate. It can then dynamically scale rs1 or at compile time the imm.

>we need a scalar "addvl" instruction to add an integer multiple of vlenb value to stack pointer.

Adding to sp does not help with Example 2.

The problem you are stating is the difficulty deriving a multiple of vlenb to add/subtract from a register, especially sp.

Since vlenb is a Read Only CSR, the csrrw and csrrwi are unused.

These variants could be used to request a multiplier for vlenb.

Thus examples 1 and 2 become.

Example 1: allocate RVV stack in prologue:
csrrwi t0, vlenb, 7
sub sp, sp, t0

Example 2: reference vector stack object, offset #7
csrrwi a0, vlenb, 7
addi a0,a0,#unscaled_offset
add a0,a0,sp,
vl1r.v v24, (a0)

Note: csrrwi on vlenb will likely be assigned an alias (maybe vlenxi and vlenx?)

Some may object to this unorthodox use of csrrw, however, such use is planned for fast interrupt TG and been floated for other uses as well.

chuanhua, 2020-08-06:
I am sorry that I did not describe the "addvl" instruction envisioned clearly. What I am actually thinking about the addvl instruction has the following syntax and functionality.

"addvl  rd, rs1, signed_imm_6bit"
"rd = rs1 + (signed_imm_6bit * vlenb)"

So it should be useful for example 2 as well. But it seems that I cannot find an existing instruction format with 6-bits of immediate values.

Your proposal of using "csrrwi rd, vlenb, unsigned_imm_5bit" with the function of "rd = unsigned_imm_5bit * vlenb" is a good idea.

aswaterman, 2020-08-06:
I'm not much of a fan of the "magic CSR" approach.  It creates an awkward special case, since many styles of implementation perform CSR accesses at commit, whereas here we want to execute the instruction earlier in the pipeline.  It also adds a special case to the illegal-instruction detection logic, since currently all CSRs with address bits 11:10=3 trap on writes.

chuanhua, 2020-08-06:
Then we can define 2 instructions in the OP opcode space as follows:

"addvl rd, rs1, uimm5" --> "rd = rs1 + (uimm5 * vlenb_const)" // when rs1=x0, rd = uimm5 * vlenb_const
"subvl rd, rs1, uimm5" --> "rd = rs1 - (uimm5 * vlenb_const)" // when rs1=x0, rd = - (uimm5 * vlenb_const)

aswaterman, 2020-08-06:
TG should discuss whether to add such instructions, but if they are approved, I think your signed-immediate version is fine.

David-Horner, 2020-08-06:
> I'm not much of a fan of the "magic CSR" approach.
See below

> It creates an awkward special case, since many styles of implementation
> perform CSR accesses at commit, whereas here we want to execute the
> instruction earlier in the pipeline.
For vlenb we also want early CSR pipeline access. No stall if we want ooo
and vector :vl, vtype, etc. to interleave.

It also adds a special case to the illegal-instruction detection logic,
> since currently all CSRs with address bits 11:10=3 trap on writes.
I had thought that this issue could be deferred until after v1.0. But your
concern is valid and we may best carve out a section on 11:10=3 for csrrw
csrc/s allowed variants.
The original scheme is valid but as with debug in S and H it must evolve.

Magic case.
There is a camp that is adverse to  creating awkward special case
instructions with multiple ways of accessing CSRs, registers that exist
because of special handling raising such issues as already mentioned,
including synchronization. Better to keep the complexity out of the
"normal" instructions.

vsetlv is an intentional exception. It is so prevalant that it needs to be
understood and corner cases addressed.
 I fear addvl would not get that scruitiny.
But we are otherwise not in a rush to resolve it. V1.1 s ok.
But the vector CSR mapping we want to get right for v1.0

vbx-glemieux, 2020-08-06:
Have you worked out what this would look like with a larger example?

This code snippet is to save or load a single vector register. Normally,
the code would be saving/restoring multiple registers. This takes a long
time to execute. For each vector register saved/loaded, a new base address
needs to be computed. I don't think you would use addvl to compute the
address from scratch each time, when you can simply do a subi/addi to
modify sp each time.

Instead, perhaps we need vload/vstore instructions that compute the
required effective address using the register number? For example:

vl1r.v vX,(a0)

where EffectiveAddress = a0 + (vlenb * X)

The "*X" is a variable shift.

Guy

kasanovic, 2021-03-19:
Moved to post v1.0.